esphome/issues #1388: Compilation fails after update platform

StevenBL, 2020-07-31:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

PS C:\Users\Steven\meetupTTNIOT\esphome> platformio platform update
Platform Espressif 8266
--------
Updating espressif8266                   @ 2.6.1          [Up-to-date]
Updating framework-arduinoespressif8266  @ 3.20703.0      [Up-to-date]
Updating tool-esptool                    @ 1.413.0        [Up-to-date]
Updating tool-esptoolpy                  @ 1.20800.0      [Up-to-date]

Platform Espressif 8266
--------
Updating espressif8266                   @ 2.2.3          [Detached]
Updating toolchain-xtensa                @ 2.100100.200706[Up-to-date]
Updating framework-arduinoespressif8266  @ 2.20502.0      [Up-to-date]
Updating tool-esptool                    @ 1.413.0        [Up-to-date]
Updating tool-esptoolpy                  @ 1.20600.0      [Up-to-date]
**ESP (ESP32/ESP8266, Board/Sonoff):**

**ESPHome version (latest production, beta, dev branch)**

**Affected component:**

**Description of problem:**
Yaml complied OK, but not anaymore after updating the ESPHOME to the status as given above

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: esptemp0box1xyz
  platform: ESP8266
  board: nodemcuv2
# 14-12-2019   nu 2e versie met box en sensoren eraan.
# met crimp in elkaar gezet. Crimptool PA09.
# ook de captive mode erin
# werkt op 14-12
# nu display erbij op de i2c bus 
#  ook CO2 sesnsor erbij. Werkt
# Op 15-12-2019 in my-2e-box04.yaml  file 
# Er lijken thermische problemen te zijn. Foor MHZ19 lijkt dat geen issue. BME280 eruitgehaald na testen.
# zelfs op breadboard een issue.
# 27-12-2019 sensor ver buiten de box, dus T BME is OK
# met druk wel iets doen
# 28-12  druk aangepast op buienradar.
# 12-1-2020  cal van CO2 op true
# 14-01-2020 heeft gewerkt s'avonds 13-1-2020 toen de lucht erg weinig CO2
# bevatte in de logeerkamer. Nu deze functie als extra awitch erin zodat je hem aan kunt zetten
# en met beleid toepassen. Ook de zero cal fie eventueel.

wifi:
  networks:
  # set te WIFI acces
  # have used phoen to start off nov 5 2019
  # comment  line 
  #  het is mogelijk meer SSID's op te geven. Het werkt op 16-11-19
  - ssid: "CATOMARINE_DESK"
    password: "vleermuis01"
  - ssid: "CATOMARINE"
    password: "vleermuis01"
  - ssid: "QuietWillowS2"
    password: "GKP9XaPVoX"
  - ssid: "AndroidAPKEV"
    password: "ccmv4036"
  

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "ESP CO2 meter"
    password: "1234abcd"
    

i2c:
  sda:  D7  # op 3-12-2019 als in instructie meetup, is de gele draad in de box
  scl:  D6 # idem als boven  is de groene draad in de box
  scan:  True
  
uart:
  rx_pin:  D1  # D1  green wire of MHZ19
  tx_pin:  D2  # D2  blue  wire of MHZ19
  baud_rate:  9600
  # lijkt niet goed te werken op 3-12-2019
  
sensor:
  - platform:  bme280
    temperature:
      id:  temp
      filters: 
      - offset:  -0.0
      # for T calibration this box hardware is reference (27-12-2019)
      # for presure -0.85  hPa
      name:  "BME280 temperature"
    pressure:
      id:  press
      filters:
      - offset: -0.25 
      name:  "BME280 pressure"
    humidity:
      id: hum
      name:  "BME280 humidity"
    address:  0x76
    update_interval:  60s
    
  - platform: mhz19  # wel of geen B -geen b en geen B
    id: my_mhz19_id
    co2:
      name: "MH-Z19 CO2 Value"
      id:  co2value
    temperature:
      name: "MH-Z19 Temperature"
    update_interval: 60s
    automatic_baseline_calibration: false
    #  dit is nu de testwaarde, true 
   
font:
  - file: "COMIC.ttf"
    id: my_font
    size: 15
    
display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    address: 0x3C
  #  rotation: 180  in box niet nodig
  # in volgende regel drie regel op display met temp, pressure en RH
    lambda: |-
      //it.print(0, 0, id(my_font), "Hello World!");   
      it.printf(0, 7, id(my_font), "T : %.1f C", id(temp).state);
      it.printf(0, 22, id(my_font), "P : %.1f hPa", id(press).state);
      it.printf(0, 37, id(my_font), "CO2 : %.1f ppm", id(co2value).state);
      // If the sensor has the value 30.02, the result will be: "The sensor value is: 30.0";
  
  
  # mqtt public broker ota er uit
mqtt:
  broker: broker.hivemq.com
  port:  1883
  
web_server:
  port: 80
  
switch:
  - platform: gpio
    name: "LED-BOX1-KEV"
    pin:
       number:  16
       inverted:  yes
  - platform: gpio
    name:  "Calibrate MH16 box"
    pin:
      number:  16 # zelfde blauwe lED als eerdere switch
      inverted: yes
    on_turn_on:
      then:
        - mhz19.abc_enable: my_mhz19_id
       # - mhz19.calibrate_zero: my_mhz19_id
    on_turn_off:
      then:
        - mhz19.abc_disable: my_mhz19_id 

# captive_portal:
# captve eruit omdat het niet goed compileerde (28-12-2019)

# Enable logging
logger:

# Enable Home Assistant API
# api:
# previous line commented out for MQTT

# ota:
# commented out to keep it simple with MQTT

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

ahawks, 2020-08-01:
I'm having this same issue on a windows 10 install.

ahawks, 2020-08-01:
I think I got it to work by running this first:

platformio update

ashirley, 2020-08-01:
Same.

The first compile failure was using:

> espressif8266 2.2.3:
> - framework-arduinoespressif8266 2.20502.0 (2.5.2)
> - tool-esptool 1.413.0 (4.13)
> - tool-esptoolpy 1.20600.0 (2.6.0)
> - toolchain-xtensa 2.100100.200706 (10.1.0)

running `platformio platform update` gave these:

> espressif8266 2.6.1:
> - framework-arduinoespressif8266 3.20703.0
> - tool-esptool 1.413.0
> - tool-esptoolpy 1.20800.0
> - toolchain-xtensa 2.40802.191122

Note that this seems to be a downgrade of toolchain-xtensa. re-running the compile then used:

> espressif8266 2.2.3:
> - framework-arduinoespressif8266 2.20502.0 (2.5.2)
> - tool-esptool 1.413.0 (4.13)
> - tool-esptoolpy 1.20600.0 (2.6.0)
> - toolchain-xtensa 2.40802.200502 (4.8.2)

I'm not really sure what that means :-P but I hope the toolchain-xtensa downgrade points to the root cause.

StevenBL, 2020-08-03:
Did retry after update on August 3 of espressif8266 from 2.6.1.  to 2.6.2 see below.  No cahnge, Problem persists. According to the error log (see above) the definition of  **isinf** and **isnan** is missing in the std file. Not sure what that means and how to resolve.

PS C:\Users\Steven> platformio platform  update --dry-run
Platform Espressif 8266
--------
Checking espressif8266                   @ 2.6.2          [Up-to-date]
Checking framework-arduinoespressif8266  @ 3.20704.0      [Up-to-date]
Checking tool-esptool                    @ 1.413.0        [Up-to-date]
Checking tool-esptoolpy                  @ 1.20800.0      [Up-to-date]

Platform Espressif 8266
--------
Checking espressif8266                   @ 2.2.3          [Detached]
Checking toolchain-xtensa                @ 2.100100.200706[Up-to-date]
Checking framework-arduinoespressif8266  @ 2.20502.0      [Up-to-date]
Checking tool-esptool                    @ 1.413.0        [Up-to-date]
Checking tool-esptoolpy                  @ 1.20600.0      [Up-to-date]

J3n50m4t, 2020-08-04:
@StevenBL if you got some new important infos please don't edit. almost overlooked the last edit. 

what do you mean by limited update?

J3n50m4t, 2020-08-04:
Running on MacOSX it starts working after reinstalling everything today.
`pip3 install esphome` followed by `platformio platform update`
gives 
```
Platform Espressif 8266
--------
Updating espressif8266                   @ 2.2.3          [2.6.2]
Uninstalling espressif8266 @ 2.2.3: 	[OK]
PlatformManager: Installing espressif8266 @ 2.6.2
espressif8266 @ 2.6.2 has been successfully installed!
Updating tool-esptool                    @ 1.413.0        [Up-to-date]
Uninstalling framework-arduinoespressif8266 @ 2.20502.0: 	[OK]
Uninstalling tool-esptoolpy @ 1.20600.0: 	[OK]
Uninstalling toolchain-xtensa @ 2.100100.200706: 	[OK]
PackageManager: Installing toolchain-xtensa @ ~2.40802.191122
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
toolchain-xtensa @ 2.40802.200502 has been successfully installed!
PackageManager: Installing framework-arduinoespressif8266 @ ~3.20704.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
framework-arduinoespressif8266 @ 3.20704.0 has been successfully installed!
PackageManager: Installing tool-esptoolpy @ ~1.20800.0
tool-esptoolpy @ 1.20800.0 has been successfully installed!
```
Now I'm able to compile the yaml created by `esphome test.yaml wizarrd`

StevenBL, 2020-08-04:
> 
> 
> @StevenBL if you got some new important infos please don't edit. almost overlooked the last edit.
> 
> what do you mean by limited update?

This means I did noy carry  out the last step to update till all elements where up-to-date. That would bring back the strange  version of the xtensa toolchain. Thanks for the tip about editing.

StevenBL, 2020-08-06:
The action 2 days ago seems to solve the problem in a way which allows me to continue building IOT apps.
@StevenBL 

Action was: Unistalled esphome and platformio. After reinstall and limited!! update original system seems to work again.
Issue with updating in my configuration with two times the platform??

closed 2020-08-06 by StevenBL

ahawks, 2020-08-07:
I just followed those instructions on WIndows 10 and have the exact same problem still. I would not consider this issue closed. It's also my first and only experiene with ESPHome, so it's a huge bummer not to be able to compile and get going on it.

`pip3 uninstall esphome`
`pip3 uninstall platformio`
`pip3 install esphome`
`platformio platform update`
`esphome test.yaml compile`

Same errors as before.

ahawks, 2020-08-07:
I also re-did it, but first used platformio to uninstall all installed frameworks first.
`platformio platform uninstall espressif8266`
and anything else listed in `platformio platform list`
Then `pip3 uninstall esphome platformio`
Then re-installed and compiled. Still getting:

```
Compiling .pioenvs\test\src\esphome\components\api\subscribe_state.cpp.o
Compiling .pioenvs\test\src\esphome\components\api\user_services.cpp.o
In file included from src/esphome/core/component.h:5,
                 from src\esphome\components\api\user_services.h:3,
                 from src\esphome\components\api\user_services.cpp:1:
C:\Users\ahawk\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:270:12: error: 'isinf' has not been declared in 'std'
  270 | using std::isinf;
      |            ^~~~~
C:\Users\ahawk\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:271:12: error: 'isnan' has not been declared in 'std'
  271 | using std::isnan;
      |            ^~~~~
In file included from src/esphome/core/component.h:5,
                 from src\esphome\components\api\proto.h:3,
                 from src\esphome\components\api\proto.cpp:1:
```

ahawks, 2020-08-07:
UPDATE: I GOT IT WORKING TOO.

```
rm -rf ~/.platformio
rm -rf ./.esphome
platformio platform list # uninstall anything listed
pip3 uninstall esphome platformio
pip3 install esphome platformio
esphome test.yaml wizard
platformio platform update
esphome test.yaml compile
```

StevenBL, 2020-08-07:
@ahawks  Great that the system is working now. @StevenBL

syphernl, 2020-08-08:
> UPDATE: I GOT IT WORKING TOO.
> 
> ```
> rm -rf ~/.platformio
> rm -rf ./.esphome
> platformio platform list # uninstall anything listed
> pip3 uninstall esphome platformio
> pip3 install esphome platformio
> esphome test.yaml wizard
> platformio platform update
> esphome test.yaml compile
> ```

I followed these same instructions only to get the same problem yet again. 
Maybe you were just lucky getting a different version of one of the dependencies?

This is the output of my `platformio platform list`:
```bash
espressif8266 ~ Espressif 8266
==============================
Espressif Systems is a privately held fabless semiconductor company. They provide wireless communications and Wi-Fi chips which are widely used in mobile devices and the Internet of Things applications.

Home: https://espressif.com/
Frameworks: arduino, esp8266-nonos-sdk, esp8266-rtos-sdk, simba
Packages: toolchain-xtensa, framework-arduinoespressif8266, framework-esp8266-rtos-sdk, framework-esp8266-nonos-sdk, framework-simba, tool-esptool, tool-esptoolpy, tool-mkspiffs
Version: 2.2.3
```