esphome/issues #1389: TuyaMCU: TUYA_CMD_WIFI_SELECT is not handled

ryrzy, 2020-07-31:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

tywe3s

**ESPHome version (latest production, beta, dev branch)**

1.16.0-dev

**Affected component:**

https://next.esphome.io/components/tuya.html

**Description of problem:**
i would like to test my wifi battery door sensor with tuya mcu and i cant get it working, in log is only error: TUYA_CMD_WIFI_SELECT is not handled. it shows always when i open/close sensor. on tasmota it works ok.
It is exactly this sensor: https://templates.blakadder.com/TY01.html

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: wifi_door_1
  platform: ESP8266
  board: esp01_1m
  #board: d1_mini

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_psswd

# Make sure logging is not using the serial port
logger:
  baud_rate: 0

# Enable Home Assistant API
api:

# Make sure you can upload new firmware OTA
ota:

uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600

# Register the Tuya MCU connection
tuya:

```

**Logs (if applicable):**

```
[19:14:22][I][app:105]: ESPHome version 1.16.0-dev compiled on Jul 31 2020, 19:13:23
[19:14:22][C][wifi:443]: WiFi:
[19:14:22][C][wifi:303]:   SSID: [redacted]
[19:14:22][C][wifi:304]:   IP Address: 192.168.x.x
[19:14:22][C][wifi:306]:   BSSID: [redacted]
[19:14:22][C][wifi:307]:   Hostname: 'wifi_door_1'
[19:14:22][C][wifi:311]:   Signal strength: -44 dB ▂▄▆█
[19:14:22][C][wifi:315]:   Channel: 11
[19:14:22][C][wifi:316]:   Subnet: 255.255.255.0
[19:14:22][C][wifi:317]:   Gateway: 192.168.1.1
[19:14:22][C][wifi:318]:   DNS1: 192.168.1.1
[19:14:22][C][wifi:319]:   DNS2: (IP unset)
[19:14:22][C][uart_esp8266:075]: UART Bus:
[19:14:22][C][uart_esp8266:077]:   TX Pin: GPIO1
[19:14:22][C][uart_esp8266:080]:   RX Pin: GPIO3
[19:14:22][C][uart_esp8266:081]:   RX Buffer Size: 256
[19:14:22][C][uart_esp8266:083]:   Baud Rate: 9600 baud
[19:14:22][C][uart_esp8266:084]:   Data Bits: 8
[19:14:22][C][uart_esp8266:085]:   Parity: NONE
[19:14:22][C][uart_esp8266:086]:   Stop bits: 1
[19:14:22][C][uart_esp8266:088]:   Using hardware serial interface.
[19:14:22][C][logger:185]: Logger:
[19:14:22][C][logger:186]:   Level: DEBUG
[19:14:22][C][logger:187]:   Log Baud Rate: 0
[19:14:22][C][logger:188]:   Hardware UART: UART0
[19:14:22][C][ota:029]: Over-The-Air Updates:
[19:14:22][C][ota:030]:   Address: wifi_door_1.local:8266
[19:14:22][C][api:095]: API Server:
[19:14:22][C][api:096]:   Address: wifi_door_1.local:6053
[19:14:22][C][tuya:023]: Tuya:
[19:14:22][C][tuya:026]:   Configuration will be reported when setup is complete. Current init_state: 3
[19:14:22][C][tuya:027]:   If no further output is received, confirm that this is a supported Tuya device.
[19:14:23][E][tuya:170]: TUYA_CMD_WIFI_SELECT is not handled
[19:14:27][E][tuya:170]: TUYA_CMD_WIFI_SELECT is not handled
[19:14:28][E][tuya:170]: TUYA_CMD_WIFI_SELECT is not handled
[19:14:32][E][tuya:170]: TUYA_CMD_WIFI_SELECT is not handled
[19:14:37][E][tuya:170]: TUYA_CMD_WIFI_SELECT is not handled

```

**Additional information and things you've tried:**

OttoWinter, 2020-08-06:
CC @ssieb (https://github.com/esphome/esphome/pull/743)

Why is the "`TUYA_CMD_WIFI_SELECT` not handled" an error message? As far as I understand it's not required for devices to respond to that command, no?

Also, error log level is reserved for things that prevent the ESP from functioning all together (and don't just affect one integration).

Can we reduce the log level for this command safely?

ssieb, 2020-08-06:
What log level should be used for something that causes a component to fail?  Isn't that what ERROR is for?
I don't remember exactly why I used error.  Either it was because I wanted to make sure to find out if it was actually used or maybe because it is an error not to handle it.  But it turned out to be a valid use in this case.  This device is a low power one and the protocol docs are only in Chinese.  Using Google Translate, it looks like we *have* to handle this message.  "`This command is a synchronous command. After the MCU data is reported, it is necessary to wait for the module to return the report result and wait for the timeout period of 7 seconds.`"  It's a little confusing, but "MCU" refers to the external MCU and the ESP is the "module".  Further on, it appears to say that if there is no response for too long, it will forcibly turn off the ESP.  With more reading, it appears that protocol version 0 and version 3 use the same values for different commands.  Protocol 0 is for battery powered devices and this command is a datapoint push.  The response from the ESP is to say whether or not the data was successfully sent.  I think we need fully separate handling for these devices since the commands are different and the control flow is different.
This is the flowchart:
[image]
It's too low resolution for good automatic translation.  If someone knows where an English version is (the page says it's only available in Chinese) or can translate it, that would be really helpful.

ssieb, 2020-08-06:
@ryrzy are you able to serial flash this device?  Testing this change could make it impossible to OTA flash the device and serial logs would be very useful to understand what's happening.
@OttoWinter is there any way to tell when data has been successfully sent?  And some way needs to be found to keep the device on for OTA updates.

ryrzy, 2020-08-07:
@ssieb no problem with serial flash, but will there not be problems with logs tuyamcu is using tx and rx too? To flash it with serial i must disable tuyamcu always :)

ssieb, 2020-08-07:
If the Tuya mcu is using the standard uart0 pins, then logs won't work unless you can put the logger on some other pins.  I don't know how accessible your device is.  But as long as you can do serial flashing, I'll be happy to get you to test it.  The OTA logs might provide enough info.  Would you know how to test the code from my branch?

ryrzy, 2020-08-07:
@ssieb hmm i think i will be able :) 
my door sensor look exactly like this one: https://templates.blakadder.com/TY01.html
soldering is no problem if mean this ;)

ssieb, 2020-08-07:
If you could connect a uart to alternate pins on that device, that would be amazing.  I could order one myself, but then there won't be any fixes for at least a month. :grin:  Either way, set the logger level to verbose.  My branch is at https://github.com/ssieb/esphome/tree/tuya.  It's not a full implementation yet, but you should at least get the dump of datapoints even over OTA,

ryrzy, 2020-08-09:
@ssieb 
sorry for delay :) i have so much work bleeee
something is wrong with tuya code, or i have made something not right?

> INFO Reading configuration wifi_door_1.yaml...
> 
> INFO Generating C++ source...
> INFO Compiling app...
> INFO Running:  platformio run -d wifi_door_1
> Processing wifi_door_1 (board: esp01_1m; framework: arduino; platform: espressif8266@2.6.1)
> --------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
> HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
> PACKAGES:
>  - framework-arduinoespressif8266 3.20703.0 (2.7.3)
>  - tool-esptool 1.413.0 (4.13)
>  - tool-esptoolpy 1.20800.0 (2.8.0)
>  - toolchain-xtensa 2.40802.200502 (4.8.2)
> Dependency Graph
> |-- <ESPAsyncTCP-esphome> 1.2.3
> |   |-- <ESP8266WiFi> 1.0
> |-- <ESP8266WiFi> 1.0
> |-- <ESP8266mDNS> 1.2
> |   |-- <ESP8266WiFi> 1.0
> |-- <ESPAsyncWebServer-esphome> 1.2.7
> |   |-- <ESPAsyncTCP-esphome> 1.2.3
> |   |   |-- <ESP8266WiFi> 1.0
> |   |-- <Hash> 1.0
> |   |-- <ESP8266WiFi> 1.0
> |   |-- <ArduinoJson-esphomelib> 5.13.3
> |-- <ArduinoJson-esphomelib> 5.13.3
> Compiling .pioenvs/wifi_door_1/src/esphome/components/tuya/tuya.cpp.o
> Compiling .pioenvs/wifi_door_1/src/main.cpp.o
> In file included from src/esphome/components/tuya/tuya.cpp:2:0:
> src/esphome/components/tuya/tuya.cpp: In member function 'void esphome::tuya::Tuya::handle_command_(esphome::tuya::TuyaCommandTypeV0, uint8_t, const uint8_t*, size_t)':
> src/esphome/components/tuya/tuya.cpp:143:66: error: expected ')' before ':' token
>        ESP_LOGV(TAG, "MCU requested %s wifi mode", buffer[0] == 0 : "smartconfig", "AP");
>                                                                   ^
> src/esphome/core/log.h:79:91: note: in definition of macro 'esph_log_v'
>    esp_log_printf_(ESPHOME_LOG_LEVEL_VERBOSE, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
>                                                                                            ^
> src/esphome/components/tuya/tuya.cpp:143:7: note: in expansion of macro 'ESP_LOGV'
>        ESP_LOGV(TAG, "MCU requested %s wifi mode", buffer[0] == 0 : "smartconfig", "AP");
>        ^
> *** [.pioenvs/wifi_door_1/src/esphome/components/tuya/tuya.cpp.o] Error 1
> ====================================================================================== [FAILED] Took 1.25 seconds ======================================================================================
> (venv) xxx@yyy:~/esphome/Testing$

ssieb, 2020-08-09:
I don't understand why that's there.  The file on my computer has the right code and git says it's up to date with github.  The correct line is:
```
ESP_LOGD(TAG, "MCU request %s Wifi mode", buffer[0] == 0 ? "smartconfig" : "AP");
```
Either change those characters or comment out the line.  Hopefully it will get corrected when I make another commit.

ssieb, 2020-08-09:
Sorry, I was looking at the wrong line.  I've pushed a fix.  I was wondering why it let me compile, but then I realized I probably didn't install the updated code before doing the compile.

ryrzy, 2020-08-09:
ok, thanks, will look at this maybe tomorow

ryrzy, 2020-08-10:
@ssieb 
now it looks ok :) i didnt change uart pins, online log shows everything :)

```
INFO Reading configuration /config/esphome/wifi_door_1.yaml...
INFO Starting log output from wifi_door_1.local using esphome API
INFO Connecting to wifi_door_1.local:6053 (xx.xx.xx.xx)
INFO Successfully connected to wifi_door_1.local
[08:20:40][I][app:105]: ESPHome version 1.16.0-dev compiled on Aug 10 2020, 08:14:09
[08:20:40][C][wifi:443]: WiFi:
[08:20:40][C][wifi:303]:   SSID: [redacted]
[08:20:40][C][wifi:304]:   IP Address: xx.xx.xx.xx
[08:20:40][C][wifi:306]:   BSSID: [redacted]
[08:20:40][C][wifi:307]:   Hostname: 'wifi_door_1'
[08:20:40][C][wifi:311]:   Signal strength: -40 dB ▂▄▆█
[08:20:40][V][wifi:313]:   Priority: 0.0
[08:20:40][C][wifi:315]:   Channel: 11
[08:20:40][C][wifi:316]:   Subnet: 255.255.255.0
[08:20:40][C][wifi:317]:   Gateway: 192.168.1.1
[08:20:40][C][wifi:318]:   DNS1: 192.168.1.1
[08:20:40][C][wifi:319]:   DNS2: (IP unset)
[08:20:40][C][uart_esp8266:075]: UART Bus:
[08:20:40][C][uart_esp8266:077]:   TX Pin: GPIO1
[08:20:40][C][uart_esp8266:080]:   RX Pin: GPIO3
[08:20:40][C][uart_esp8266:081]:   RX Buffer Size: 256
[08:20:40][C][uart_esp8266:083]:   Baud Rate: 9600 baud
[08:20:40][C][uart_esp8266:084]:   Data Bits: 8
[08:20:40][C][uart_esp8266:085]:   Parity: NONE
[08:20:40][C][uart_esp8266:086]:   Stop bits: 1
[08:20:40][C][uart_esp8266:088]:   Using hardware serial interface.
[08:20:40][C][logger:185]: Logger:
[08:20:40][C][logger:186]:   Level: VERBOSE
[08:20:40][C][logger:187]:   Log Baud Rate: 0
[08:20:40][C][logger:188]:   Hardware UART: UART0
[08:20:40][C][web_server:131]: Web Server:
[08:20:40][C][web_server:132]:   Address: wifi_door_1.local:80
[08:20:40][C][web_server:134]:   Basic authentication enabled
[08:20:40][C][ota:029]: Over-The-Air Updates:
[08:20:40][C][ota:030]:   Address: wifi_door_1.local:8266
[08:20:40][W][ota:036]: Last Boot was an unhandled reset, will proceed to safe mode in 4 restarts
[08:20:40][C][api:095]: API Server:
[08:20:40][C][api:096]:   Address: wifi_door_1.local:6053
[08:20:40][C][tuya:043]: Tuya:
[08:20:40][C][tuya:051]:   Battery powered device detected
[08:20:40][C][tuya:068]:   Product: '{"p":"3slpirlnlqp0bqo1","v":"1.0.0"}'
[08:20:40][V][tuya:314]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=5
[08:20:40][V][tuya:120]: Received Tuya: CMD=0x00 VERSION=0 DATA=[01 (1)] INIT_STATE=5
[08:20:41][V][tuya:314]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=5
[08:20:41][V][tuya:120]: Received Tuya: CMD=0x00 VERSION=0 DATA=[01 (1)] INIT_STATE=5
[08:20:42][V][tuya:314]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=5
[08:20:42][V][tuya:120]: Received Tuya: CMD=0x00 VERSION=0 DATA=[01 (1)] INIT_STATE=5
[08:20:43][V][tuya:314]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=5
[08:20:44][V][tuya:314]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=5
[08:20:45][V][tuya:314]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=5
[08:20:46][V][tuya:314]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=5
[08:20:47][V][tuya:314]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=5
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to wifi_door_1.local:6053 (xx.xx.xx.xx)
INFO Successfully connected to wifi_door_1.local
[08:20:56][V][tuya:314]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=5
[08:20:56][V][tuya:120]: Received Tuya: CMD=0x00 VERSION=0 DATA=[01 (1)] INIT_STATE=5
[08:20:57][C][tuya:043]: Tuya:
[08:20:57][C][tuya:051]:   Battery powered device detected
[08:20:57][C][tuya:054]:   Datapoint 1: switch (value: ON)
[08:20:57][C][tuya:068]:   Product: '{"p":"3slpirlnlqp0bqo1","v":"1.0.0"}'
[08:20:57][V][tuya:314]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=5
[08:20:57][V][tuya:120]: Received Tuya: CMD=0x00 VERSION=0 DATA=[01 (1)] INIT_STATE=5
[08:20:58][V][tuya:120]: Received Tuya: CMD=0x05 VERSION=0 DATA=[01.01.00.01.00 (5)] INIT_STATE=5
[08:20:58][V][tuya:284]: Datapoint 1 update to 0
[08:20:58][V][tuya:314]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=5
[... 156 lines cut ...]
[08:36:00][D][sensor:092]: 'Door sensor': Sending state 1.00000  with 0 decimals of accuracy
[08:36:00][D][tuya.sensor:013]: MCU reported sensor is: ON
[08:36:01][V][tuya:314]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=5
[08:36:01][V][tuya:120]: Received Tuya: CMD=0x00 VERSION=0 DATA=[01 (1)] INIT_STATE=5
[08:36:01][C][tuya:043]: Tuya:
[08:36:01][C][tuya:051]:   Battery powered device detected
[08:36:01][C][tuya:054]:   Datapoint 1: switch (value: ON)
[08:36:01][C][tuya:068]:   Product: '{"p":"3slpirlnlqp0bqo1","v":"1.0.0"}'
[08:36:02][V][tuya:314]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=5
[08:36:02][V][tuya:120]: Received Tuya: CMD=0x00 VERSION=0 DATA=[01 (1)] INIT_STATE=5
[08:36:03][V][tuya:314]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=5
[08:36:03][V][tuya:120]: Received Tuya: CMD=0x00 VERSION=0 DATA=[01 (1)] INIT_STATE=5
[08:36:04][V][tuya:314]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=5
[08:36:04][V][tuya:120]: Received Tuya: CMD=0x00 VERSION=0 DATA=[01 (1)] INIT_STATE=5
[08:36:05][V][tuya:314]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=5
[08:36:05][V][tuya:120]: Received Tuya: CMD=0x00 VERSION=0 DATA=[01 (1)] INIT_STATE=5
[08:36:06][V][tuya:314]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=5
[... cut ...]
[08:36:07][V][tuya:314]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=5
[08:36:07][V][tuya:120]: Received Tuya: CMD=0x00 VERSION=0 DATA=[01 (1)] INIT_STATE=5
[08:36:08][V][tuya:314]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=5
[08:36:08][V][tuya:120]: Received Tuya: CMD=0x00 VERSION=0 DATA=[01 (1)] INIT_STATE=5
[08:36:09][V][tuya:314]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=5
[08:36:09][V][tuya:120]: Received Tuya: CMD=0x00 VERSION=0 DATA=[01 (1)] INIT_STATE=5
```

ssieb, 2020-08-11:
That's definitely improved.  How long have you let it run?  Does it ever turn off?  It shouldn't be dumping the config every time the value changes, although it should really be turning off before the value changes.  My code doesn't acknowledge the data yet, so it would have to go to the maximum timeout.

ryrzy, 2020-08-11:
yes it is powering off after x seconds  (didn't mesure it). but i see sometimes it takes realy much time to wake up (it wakes up, connect disconnect and connect one more time).

ssieb, 2020-12-14:
I forgot about this one.  I have hardware now, but no time. :-(

closed 2021-09-06 by stale[bot]