esphome/issues #1392: After transmit any command ESP32 constantly restarts

ekorsanov01, 2020-08-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

homeassistant v0.113.2
docker_version v18.09.7
hassio - true
host_os | Ubuntu 18.04.3 LTS
installation_type | Home Assistant Supervised
os_name | Linux
os_version | 4.19.63-sunxi64
python_version | 3.8.3
supervisor | 229
virtualenv | false

**ESP (ESP32/ESP8266, Board/Sonoff):**

esp32 (single core))

[https://yadi.sk/i/d2eHc5Y8M4oFxw](url)
[https://yadi.sk/i/9WWosrqvQklFQA](url)

**ESPHome version (latest production, beta, dev branch)**

ESPHome version like v1.14.5

**Affected component:**

https://esphome.io/components/remote_transmitter.html

**Description of problem:**

After transmit any command ESP constantly restarts,
but the command is successfully sent once before restarting the Board and the TV turns on!
After sending the command, the IR led remains at a voltage of 1.4 V, can this be a problem?

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  devicename: xiaomi_ir
  upper_devicename: Xiaomi IR Remote

esphome:
  name: $devicename
  platform: ESP32
  board: esp32doit-devkit-v1
  platformio_options:
    platform: espressif32@1.11.0
    platform_packages: |-4
      
          framework-arduinoespressif32 @ https://github.com/pauln/arduino-esp32.git#solo-no-mac-crc/1.0.4
  
wifi:
  ssid: "your_SSID_here"
  password: "your_Password_here"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Xiaomi Ir Fallback Hotspot"
    password: "your_Password_here"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "api_password_here"

ota:
  password: "ota_password_here"
[... 227 lines cut ...]
[D][binary_sensor:034]: 'Xiaomi IR Remote Button D': Sending initial state OFF
[C][remote_receiver.esp32:012]: Setting up Remote Receiver...
[C][wifi:033]: Setting up WiFi...

.....

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
[I][logger:156]: Log initialized
[C][status_led:014]: Setting up Status LED...
[C][ota:364]: There have been 10 suspected unsuccessful boot attempts.
[E][ota:369]: Boot loop detected. Proceeding to safe mode.
[I][app:028]: Running through setup()...
[C][wifi:033]: Setting up WiFi...

```

**Additional information and things you've tried:**

glmnet, 2020-08-02:
Why carrier duty 100%?

Try 50. Also try without the receiver just to try if it is related.

ekorsanov01, 2020-08-02:
with 50% and without the receiver 
problem reproduced again (((

```
[VV][scheduler:131]: Running timeout '' with interval=0 last_execution=59449 (now=59457)
[D][switch:029]: 'Xiaomi IR Remote AIR Power On' Toggling ON.
[VV][remote_base:040]: Sending times=1 wait=0ms: 9000, -4500, 560, -560, 560, -560, 560, -1690, 560, -560, 560, -560, 560, -560, 560, -560, 560, -560, 560, -1690, 560, -1690, 560, -560, 560, -1690, 560, -1690, 560, -1690, 560, -1690, 560, -1690, 560, -560, 560, -560, 560, 
[VV][remote_base:053]:   -560, 560, -1690, 560, -560, 560, -560, 560, -560, 560, -560, 560, -1690, 560, -1690, 560, -1690, 560, -560, 560, -1690, 560, -1690, 560, -1690, 560, -1690, 560

Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
[I][logger:156]: Log initialized
[C][status_led:014]: Setting up Status LED...
[VV][preferences:038]: LOAD 0: valid=YES, 0=0x00000001 1=0x42F4583F (Type=233825507, CRC=0x42F4583F)
[C][ota:364]: There have been 1 suspected unsuccessful boot attempts.
[VV][preferences:051]: SAVE 0: 0=0x00000002 1=0x13D89D5A (Type=233825507, CRC=0x13D89D5A)
[I][app:028]: Running through setup()...
[V][app:029]: Sorting components by setup priority...
[D][binary_sensor:034]: 'Xiaomi IR Remote Button A': Sending initial state OFF
[D][binary_sensor:034]: 'Xiaomi IR Remote Button B': Sending initial state OFF
[D][binary_sensor:034]: 'Xiaomi IR Remote Button C': Sending initial state OFF
[D][binary_sensor:034]: 'Xiaomi IR Remote Button D': Sending initial state OFF
[VV][scheduler:056]: set_interval(name='update', interval=60000, offset=29646)
[VV][preferences:038]: LOAD 1: valid=YES, 0=0x00000000 1=0x000041C0 (Type=2722369216, CRC=0xFD85AF20)
[D][climate:172]: 'dantex_ac' - Sending state:
[D][climate:175]:   Mode: OFF
[D][climate:186]:   Target Temperature: 24.00°C
[VV][preferences:051]: SAVE 1: 0=0x00000000 1=0x000041C0 (Type=2722369216, CRC=0xFD85AF20)
[C][wifi:033]: Setting up WiFi...
[V][wifi_esp32:033]: Enabling STA.
[VV][esp-idf:000]: I (428) wifi: wifi driver task: 3ffb5ec8, prio:23, stack:3584, core=0
[VV][esp-idf:000]: I (440) wifi: wifi firmware version: 7997e4b
[VV][esp-idf:000]: I (440) wifi: config NVS flash: enabled
[VV][esp-idf:000]: I (446) wifi: config nano formating: disabled
[VV][esp-idf:000]: I (453) wifi: Init dynamic tx buffer num: 32
[VV][esp-idf:000]: I (459) wifi: Init data frame dynamic rx buffer num: 32
[VV][esp-idf:000]: I (467) wifi: Init management frame dynamic rx buffer num: 32
[VV][esp-idf:000]: I (475) wifi: Init management short buffer num: 32
[VV][esp-idf:000]: I (482) wifi: Init static rx buffer size: 1600
[VV][esp-idf:000]: I (489) wifi: Init static rx buffer num: 16
[VV][esp-idf:000]: I (495) wifi: Init dynamic rx buffer num: 32

Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
[I][logger:156]: Log initialized
[C][status_led:014]: Setting up Status LED...
[VV][preferences:038]: LOAD 0: valid=YES, 0=0x00000002 1=0x13D89D5A (Type=233825507, CRC=0x13D89D5A)
[C][ota:364]: There have been 2 suspected unsuccessful boot attempts.
[VV][preferences:051]: SAVE 0: 0=0x00000003 1=0x60BCD276 (Type=233825507, CRC=0x60BCD276)
[I][app:028]: Running through setup()...
[V][app:029]: Sorting components by setup priority...
[D][binary_sensor:034]: 'Xiaomi IR Remote Button A': Sending initial state OFF
[D][binary_sensor:034]: 'Xiaomi IR Remote Button B': Sending initial state OFF
[D][binary_sensor:034]: 'Xiaomi IR Remote Button C': Sending initial state OFF
[D][binary_sensor:034]: 'Xiaomi IR Remote Button D': Sending initial state OFF
[VV][scheduler:056]: set_interval(name='update', interval=60000, offset=13869)
[VV][preferences:038]: LOAD 1: valid=YES, 0=0x00000000 1=0x000041C0 (Type=2722369216, CRC=0xFD85AF20)
[D][climate:172]: 'dantex_ac' - Sending state:
[D][climate:175]:   Mode: OFF
[D][climate:186]:   Target Temperature: 24.00°C
[VV][preferences:051]: SAVE 1: 0=0x00000000 1=0x000041C0 (Type=2722369216, CRC=0xFD85AF20)
[C][wifi:033]: Setting up WiFi...
[V][wifi_esp32:033]: Enabling STA.
[VV][esp-idf:000]: I (428) wifi: wifi driver task: 3ffb5ec8, prio:23, stack:3584, core=0
[VV][esp-idf:000]: I (440) wifi: wifi firmware version: 7997e4b
[VV][esp-idf:000]: I (440) wifi: config NVS flash: enabled
[VV][esp-idf:000]: I (446) wifi: config nano formating: disabled
[VV][esp-idf:000]: I (453) wifi: Init dynamic tx buffer num: 32
[VV][esp-idf:000]: I (459) wifi: Init data frame dynamic rx buffer num: 32
[VV][esp-idf:000]: I (467) wifi: Init management frame dynamic rx buffer num: 32
[VV][esp-idf:000]: I (475) wifi: Init management short buffer num: 32
[VV][esp-idf:000]: I (482) wifi: Init static rx buffer size: 1600
[VV][esp-idf:000]: I (489) wifi: Init static rx buffer num: 16
[VV][esp-idf:000]: I (495) wifi: Init dynamic rx buffer num: 32

Brownout detector was triggered
```

glmnet, 2020-08-04:
You shouldn't ignore the "brownout detector triggered" that means the esp is not powered properly. Like the usb cable is damaged or something else. 
May be the IR LED is drawing too much power and the esp reboots due to this 

Can you give info how this is assembled?

ekorsanov01, 2020-08-07:
After sending data with remote_transmitter, the IR LED stays ON 
even after restart "brownout detector triggered".

[image]

closed 2020-12-28 by stale[bot]