esphome/issues #1393: ESPHome crashes on repeatedly setup SSD1306 display commands

htvekov, 2020-08-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io

**ESP (ESP8266, Board/Sonoff):**
ESP8266 Wemos D1/D1 Mini, NodeMCU 

**ESPHome version (latest production, beta, dev branch)**
ESPHome (dev)  
v1.16.0
**Affected component:**
SSD1306 and SSD1309 displays
https://esphome.io/components/display/ssd1306.html

**Description of problem:**
I use direct id(display_name),setup() calls in lambda to update display with new brightness setting for both SSD1306 and SSD1309 display. For the SSD1309 display I also use same command to turn display on again after turning it off pulling RST pin low. I've tested with several ESP8266, and three different displays, so it not faulty hardware causing this.

Cant figure out any other way to accomplish both dynamic brightness setting  and turning on SSD1309 display, without calling id(display_name),setup() to init new settings. And this setup works absolutely perfectly - but unfortunately not for long... 😖

As the repeated calls allocates more and more RAM  on each call and after appx. some 15 calls, ESP runs out of memory and crashes 💀☠

Don't really know if this is a memory leak bug or I'm just using (abusing) service call the wrong way ?
Are repeated id(display_name).setup() calls meant to reallocate new memory 1kb block from heap on every call ?

**Problem-relevant YAML-configuration entries:**
YAML below is just a quick test YAML. No conditions to test if display if already off (or on), before issuing new on/off commands from PIR (binary_sensor). This is just to quickly demonstrate ESP's memory drain.
```yaml
esphome:
  name: ssd1306
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "MY SSID"
  password: "MY PASSWORD"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Ssd1306 Fallback Hotspot"
    password: "AUTOGENERATED PASSWORD"

captive_portal:

# Enable logging
logger:
debug:
  
# Enable Home Assistant API
api:

ota:

time:
  - platform: homeassistant
    id: esptime

binary_sensor:
  - platform: gpio
    pin: D7
    name: pir1306
    device_class: motion
    on_press:
      then:
        - binary_sensor.template.publish:
            id: backly
            state: ON
        - binary_sensor.template.publish:
            id: backly
            state: OFF
  - platform: template
    id: backly
    filters:
      - delayed_off: 20s
    on_press:
      then:
        - lambda: |-
            id(test).set_brightness(100);
            id(test).setup();
    on_release:
      then:
        - lambda: |-
            id(test).set_brightness(0);
            id(test).setup();

switch:
  - platform: gpio
    pin: D0
    name: "juhu1306"
    on_turn_on:
      then:
      - lambda: |-
          id(test).set_brightness(100);
          id(test).setup();
    on_turn_off:
      then:
      - lambda: |-
          id(test).set_brightness(0);
          id(test).setup();

i2c:
  scl: D1
  sda: D2
interval:
  - interval: 10s
    then:
      - display.page.show_next: test
      - component.update: test

display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    reset_pin: D0
    address: 0x3C
    brightness: 100%
    id: test
    pages:
      - id: page1
        lambda: |- bla bla bla bla very long lambda

```

**Logs (if applicable):**

```
[21:17:41][D][debug:023]: ESPHome version 1.16.0-dev
[21:17:41][D][debug:025]: Free Heap Size: 3824 bytes
[21:17:41][D][debug:053]: Flash Chip: Size=4096kB Speed=40MHz Mode=DOUT
[21:17:41][D][debug:190]: Chip ID: 0x00B2C38E
[21:17:41][D][debug:191]: SDK Version: 2.2.2-dev(38a443e)
[21:17:41][D][debug:192]: Core Version: 2_7_3
[21:17:41][D][debug:193]: Boot Version=31 Mode=1
[21:17:41][D][debug:194]: CPU Frequency: 80
[21:17:41][D][debug:195]: Flash Chip ID=0x001640EF
[21:17:41][D][debug:196]: Reset Reason: Software/System restart
[21:17:41][D][debug:197]: Reset Info: Software/System restart
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to ssd1306.local:6053 (192.168.0.124)
WARNING Couldn't connect to API (Error connecting to 192.168.0.124: timed out). Trying to reconnect in 1 seconds
INFO Connecting to ssd1306.local:6053 (192.168.0.124)
WARNING Couldn't connect to API (Error connecting to 192.168.0.124: [Errno 113] No route to host). Trying to reconnect in 1 seconds
INFO Connecting to ssd1306.local:6053 (192.168.0.124)
WARNING Couldn't connect to API (Error connecting to 192.168.0.124: [Errno 113] No route to host). Trying to reconnect in 2 seconds
INFO Connecting to ssd1306.local:6053 (192.168.0.124)
INFO Successfully connected to ssd1306.local
[21:18:57][D][binary_sensor:036]: 'pir1306': Sending state ON
[21:18:59][D][binary_sensor:036]: 'pir1306': Sending state OFF
[21:19:03][D][binary_sensor:036]: 'pir1306': Sending state ON
[21:19:05][D][switch:021]: 'juhu1306' Turning ON.
[21:19:05][D][switch:045]: 'juhu1306': Sending state ON
[21:19:05][C][ssd1306_i2c:010]: Setting up I2C SSD1306...
[21:19:05][D][binary_sensor:036]: 'pir1306': Sending state OFF
[21:19:06][D][switch:025]: 'juhu1306' Turning OFF.
[21:19:06][D][switch:045]: 'juhu1306': Sending state OFF
[21:19:06][C][ssd1306_i2c:010]: Setting up I2C SSD1306...
[21:19:06][D][debug:204]: Free Heap Size: 8824 bytes
[21:19:09][D][switch:021]: 'juhu1306' Turning ON.
[21:19:09][D][switch:045]: 'juhu1306': Sending state ON
[21:19:09][C][ssd1306_i2c:010]: Setting up I2C SSD1306...
[21:19:10][D][switch:025]: 'juhu1306' Turning OFF.
[21:19:10][D][switch:045]: 'juhu1306': Sending state OFF
[21:19:10][C][ssd1306_i2c:010]: Setting up I2C SSD1306...
[21:19:15][D][switch:021]: 'juhu1306' Turning ON.
[21:19:15][D][switch:045]: 'juhu1306': Sending state ON
[21:19:15][C][ssd1306_i2c:010]: Setting up I2C SSD1306...
[21:19:16][D][switch:025]: 'juhu1306' Turning OFF.
[21:19:16][D][switch:045]: 'juhu1306': Sending state OFF
[21:19:16][C][ssd1306_i2c:010]: Setting up I2C SSD1306...
[21:19:19][D][switch:021]: 'juhu1306' Turning ON.
[21:19:19][D][switch:045]: 'juhu1306': Sending state ON
[21:19:19][C][ssd1306_i2c:010]: Setting up I2C SSD1306...
[21:19:19][D][debug:204]: Free Heap Size: 3816 bytes
[21:19:20][D][switch:025]: 'juhu1306' Turning OFF.
[21:19:20][D][switch:045]: 'juhu1306': Sending state OFF
[21:19:20][C][ssd1306_i2c:010]: Setting up I2C SSD1306...
[21:19:22][D][switch:021]: 'juhu1306' Turning ON.
[21:19:22][D][switch:045]: 'juhu1306': Sending state ON
[21:19:22][C][ssd1306_i2c:010]: Setting up I2C SSD1306...
[21:19:22][D][debug:204]: Free Heap Size: 1752 bytes
[21:19:23][D][switch:025]: 'juhu1306' Turning OFF.
[21:19:23][D][switch:045]: 'juhu1306': Sending state OFF
[21:19:23][C][ssd1306_i2c:010]: Setting up I2C SSD1306...
[21:19:23][D][binary_sensor:036]: 'backly': Sending state OFF
[21:19:23][C][ssd1306_i2c:010]: Setting up I2C SSD1306...
[21:19:25][D][switch:025]: 'juhu1306' Turning OFF.
[21:19:27][D][binary_sensor:036]: 'pir1306': Sending state ON
[21:19:27][D][binary_sensor:036]: 'backly': Sending state ON
[21:19:27][C][ssd1306_i2c:010]: Setting up I2C SSD1306...
[21:19:27][E][display:017]: Could not allocate buffer for display!
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to ssd1306.local:6053 (192.168.0.124)
INFO Successfully connected to ssd1306.local
[21:19:55][D][api.connection:604]: Client 'Home Assistant 0.112.3 (192.168.0.19)' connected successfully!

```

**Additional information and things you've tried:**

kbx81, 2020-08-03:
See https://github.com/esphome/issues/issues/1235 and https://github.com/esphome/esphome/pull/1090

kbx81, 2020-08-03:
General rule of thumb -- do not repeatedly call `setup()` as it can create a variety of issues such as what you are seeing with memory being allocated repeatedly, eventually consuming it all and crashing.

htvekov, 2020-08-04:
Hi' @kbx81 

Thank you so much for your swift reply - much appreciated 🙂
I can see that all my needs are in your PR. Set brightness and turn on/off display completely 👍🤗

Tried to copy the three files from your PR to a custom component folder to test it. I can't get it to work though, ESP crashes on init and won't start up. Perhaps I've didn't get the exact right files or missed something essential in the process to get non merged PR to work in dev version copying files to custom component ?
I'm quite a newbie when it comes to Github, so I'll fiddle some more with it.
If I don't succeed, I'll wait for the PR to go be merged.

Ciao !

htvekov, 2020-08-11:
Hi' @kbx81 

Have tried numerous times, without succes,  to implement your PR files as a SSD1306_base custom component.
ESP crashes on startup, so I must be doing something wrong.

I'll wait for PR to be merged, before I'll implement these new display on/off and brightness commands.
Any idea when PR will be merged ?

Ciao !

kbx81, 2020-08-11:
I believe there are some issues with having a custom component that had the same name as an existing component--that could be why you're having issues there. 

Merging is entirely up to @glmnet and @OttoWinter 😄

htvekov, 2020-08-11:
Can't believe your PR hasn't been cherry picked for immediate merging yet 😆😁😉

I'll keep an eye on your PR.
Thanks a million for these upcoming new display commmands 👍👍😎

glmnet, 2020-08-14:
Yes, I believe if the PR has positive feedback already and is a bugfix this should be included in v.15, I'm gonna tag it

kbx81, 2020-08-24:
https://github.com/esphome/esphome/pull/1090 merged -- I believe we can close this issue.

closed 2020-08-24 by kbx81