esphome/issues #1395: fast_led clockles partition withs effects cause boot loop

Sangoku, 2020-08-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hostname | homeassistant
System | HassOS 4.11
Deployment | production
Runing on a hyperV in windows 10 

**ESP (ESP32/ESP8266, Board/Sonoff):**

https://www.aliexpress.com/item/4000089372031.html

esp01 with the rgb module on it. flashed with the flasher.

**ESPHome version (latest production, beta, dev branch)**

release 1.14.5  

**Affected component:**

https://esphome.io/components/light/fastled.html 
paired with the component platform partition
https://esphome.io/components/light/partition.html

**Description of problem:**

When you split 1 WS2812B into more then one partition you should be able to assign every partition a subset of light effects. When you use any other effect then the color transition the esp gets stuck in a endless BOOT loop.  This does not happen until you try to turn on the given effects for first time for more then one segment. After that it is always stuck in a loop. until manual usb flashing is done. 

**Problem-relevant YAML-configuration entries:**
```yaml
light:
   - platform: fastled_clockless
     chipset: WS2812B
     pin: 2
     num_leds: 277
     name: stepenice RGB
     id: sve_stepenice
     rgb_order: GRB
     max_refresh_rate: 16ms
     default_transition_length: 1s
     effects: 
      - random:
          name: Random Color
          transition_length: 4s
          update_interval: 5s
    #   - addressable_scan:
    #       name: Scan Effect
    #       move_interval: 100ms
    #       scan_width: 32
    #   - addressable_twinkle:
    #       name: Twinkle Effect  
    #       twinkle_probability: 50%
    #       progress_interval: 8ms
    #   - addressable_fireworks:
    #       name: Fireworks Effect
    #       update_interval: 32ms
    #       spark_probability: 60%
    #       use_random_color: true
    #       fade_out_rate: 120
    #   - flicker:
    #       name: Flicker 
    #       alpha: 95%
    #       intensity: 30% 
    #   - addressable_rainbow:
    #       name: Rainbow 
    #       speed: 60
    #       width: 32
          
     #sve stepenice.       
   - platform: partition
     id: step1 #32
     name: step1
     segments:
     - id: sve_stepenice
       from: 0
       to:   31  
           - random:
          name: Random Color
          transition_length: 4s
          update_interval: 5s
    #   - addressable_scan:
    #       name: Scan Effect
    #       move_interval: 100ms
    #       scan_width: 32
    #   - addressable_twinkle:
    #       name: Twinkle Effect  
    #       twinkle_probability: 50%
    #       progress_interval: 8ms
    #   - addressable_fireworks:
    #       name: Fireworks Effect
    #       update_interval: 32ms
    #       spark_probability: 60%
    #       use_random_color: true
    #       fade_out_rate: 120
    #   - flicker:
    #       name: Flicker 
    #       alpha: 95%
    #       intensity: 30% 
    #   - addressable_rainbow:
    #       name: Rainbow 
    #       speed: 60
    #       width: 32
     
   - platform: partition
     id: step2 #32
     name: step2
     segments:
     - id: sve_stepenice
       from: 32
       to:   64
     effects: 
      - random:
          name: Random Color
          transition_length: 4s
          update_interval: 5s
    #   - addressable_scan:
    #       name: Scan Effect
    #       move_interval: 100ms
    #       scan_width: 32
    #   - addressable_twinkle:
    #       name: Twinkle Effect  
    #       twinkle_probability: 50%
    #       progress_interval: 8ms
    #   - addressable_fireworks:
    #       name: Fireworks Effect
    #       update_interval: 32ms
    #       spark_probability: 60%
    #       use_random_color: true
    #       fade_out_rate: 120
    #   - flicker:
    #       name: Flicker 
    #       alpha: 95%
    #       intensity: 30% 
    #   - addressable_rainbow:
    #       name: Rainbow 
    #       speed: 60
    #       width: 32
        

```

**Logs (if applicable):**

```
 There have been 3 suspected unsuccessful boot attempts.
[23:45:44]
[23:45:44]Exception (28):
[23:45:44]epc1=0x402253ad epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000008 depc=0x00000000
[23:45:44]
[23:45:44]>>>stack>>>
[23:45:44]
[23:45:44]ctx: cont
[23:45:44]sp: 3ffffc90 end: 3fffffc0 offset: 01a0
[23:45:44]3ffffe30:  3fff4c8c 3fff4c04 00000008 40224d62  
[23:45:44]3ffffe40:  3fffff10 3fffff00 3ffffea0 402019be  
[23:45:44]3ffffe50:  3fff2ddc 3ffffebc 3ffffeb8 40224f0a  
[23:45:44]3ffffe60:  3fff02a8 3fff4c04 3fff2ddc 3fff3ba4  
[23:45:44]3ffffe70:  4021b950 3fff4c04 3fff3b9c 40201b77  
[23:45:44]3ffffe80:  4021b950 3fff2d90 667004ff 3fff2d60  
[23:45:44]3ffffe90:  3fff4900 00000101 00000000 00000000  
[23:45:45]3ffffea0:  3fff4ccc 3fff4cd0 3fff4cd0 00000101  
[23:45:45]3ffffeb0:  00000000 00000000 3fff02a8 3fff4c98  
[23:45:45]3ffffec0:  40248d77 4bc6a7f0 3fff0b3c 40212c88  
[23:45:45]3ffffed0:  3fff4414 4020b5a8 4bc6a7f0 3fff15a4  
[23:45:45]3ffffee0:  3fff1378 3fff2ddc 3fff4c58 402055a8  
[23:45:45]3ffffef0:  00000000 3fff1378 3fff4c04 40205830  
[23:45:45]3fffff00:  3fff1378 3fff1378 3fff4c04 40201d36  
[23:45:45]3fffff10:  00000000 00000000 4bc6a7f0 00000000  
[23:45:45]3fffff20:  3fff3004 00000002 3fff15a4 40205458  
[23:45:45]3fffff30:  3fff3008 00000002 3ffefe5c 4021adf0  
[23:45:45]3fffff40:  3fff3008 00000002 3ffefe5c 4021ae74  
[23:45:45]3fffff50:  3fffdad0 00000002 3ffefe5c 4020be6f  
[23:45:45]3fffff60:  00000000 00000000 3fff2cac 3fff18a4  
[23:45:45]3fffff70:  3fff300c feefeffe feefeffe feefeffe  
[23:45:45]3fffff80:  3ffefee8 3fff3364 3ffeff50 3fff01d8  
[23:45:45]3fffff90:  3fffdad0 00000000 3fff01a8 4020e700  
[23:45:45]3fffffa0:  3fffdad0 00000000 3fff01a8 402184d8  
[23:45:45]3fffffb0:  feefeffe feefeffe 3ffe8538 401009b9  
[23:45:45]<<<stack<<<

```

**Additional information and things you've tried:**

every possible hw. related test. Condensators, resistors, everything.. untill i realized it happens even when it is just sitting on the usb port when i try to turn on the lights.

OttoWinter, 2020-08-06:
Could you check if this also happens when you use `fastled` directly (with no partitions)? (Sorry if I misunderstood the description, but I couldn't find it there).

Sangoku, 2020-08-06:
The same Hardware is working 100% ok with the same strip without the partitions. 
First when you add the partitions it causes the boot loop. 

Any other effect then random color is causing the issue.

closed 2020-12-18 by stale[bot]