esphome/issues #1396: ESP 8266 user-defined services with remote transmitter

Enzokot, 2020-08-02:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 esp_wroom_02
**ESPHome version (latest production, beta, dev branch)**

1.16.0-dev
**Affected component:**

https://next.esphome.io/components/api.html#user-defined-services
https://next.esphome.io/components/remote_transmitter.html#remote-transmitter-transmit-samsung-action

**Description of problem:**

When I call service ``` esphome.s06_send_power ``` it's work fine
When I call service
``` 
service: esphome.s06_send_samsung_command
data_template:
 command: 0xE0E040BF
``` 
the service return error ```Value out of range: 3772793023``` May be this is limit of ```int``` type for ESP8266?
**Problem-relevant YAML-configuration entries:**
```yaml

api:
  services:
    - service: send_samsung_command
      variables:
        command: int
      then:
        - remote_transmitter.transmit_samsung:
            data: !lambda 'return command;'
    - service: send_power
      then:
        - remote_transmitter.transmit_samsung:
            data: 0xE0E040BF

remote_transmitter:
  pin: 
    number: GPIO14
    inverted: false
  carrier_duty_percent: 50%
  id: "transmitter_s06"
```

**Logs (if applicable):**

```
Logger: homeassistant.components.websocket_api.http.connection.2827980104
Source: components/esphome/__init__.py:292
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 9:05:11 (1 occurrences)
Last logged: 9:05:11

Value out of range: 3772793023
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 125, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1281, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1316, in _execute_service
    await handler.func(service_call)
  File "/usr/src/homeassistant/homeassistant/components/esphome/__init__.py", line 292, in execute_service
    await entry_data.client.execute_service(service, call.data)
  File "/usr/local/lib/python3.8/site-packages/aioesphomeapi/client.py", line 387, in execute_service
    setattr(arg, map_single[arg_desc.type_], val)
  File "/usr/local/lib/python3.8/site-packages/google/protobuf/internal/python_message.py", line 662, in field_setter
    new_value = type_checker.CheckValue(new_value)
  File "/usr/local/lib/python3.8/site-packages/google/protobuf/internal/type_checkers.py", line 135, in CheckValue
    raise ValueError('Value out of range: %d' % proposed_value)
ValueError: Value out of range: 3772793023

```

**Additional information and things you've tried:**

OttoWinter, 2020-08-06:
Hmm, true the native API services use 32-bit integers for values (which is kind of correct because `int` is 32-bit in C).

We might want to add 64-bit integer support for native API using a new `long long` datatype, but I'm not sure if we want to bring C-isms like the `long long` datatype into ESPHome.

Enzokot, 2020-12-04:
Keep alive

Enzokot, 2021-04-03:
Keep alive

Enzokot, 2021-08-02:
Keep alive