esphome/issues #1402: Not compiling on latest version

dclobato, 2020-08-07:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Running on WSL with PIP

**ESP (ESP32/ESP8266, Board/Sonoff):**

**ESPHome version (latest production, beta, dev branch)**

```dclobato@DESK-DANIEL:~/ESPhome/HA$ esphome version
Version: 1.14.5
dclobato@DESK-DANIEL:~/ESPhome/HA$```

**Affected component:**
<!--
Please add the link to the documentation at https://esphome.io/index.html of the component in question.
-->

**Description of problem:**
For some reason, esphome is not compiling anymore.  I updated platformio to latest version, and a lot of bugs emerged on configurations that were working fine

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: ${nomeDoNodo}
  platform: ESP8266
  board: esp8285
  build_path: binarios/${nomeDoNodo}

<<: !include common/common.yaml
<<: !include common/status_led.yaml

sensor:
  - !include sensor/wifi-signal.yaml
  - !include sensor/uptime.yaml

switch:
  - !include switch/restart.yaml

# Enable logging
logger:

binary_sensor:
  - !include binary_sensor/status.yaml
  - platform: gpio
    pin: GPIO00
    id: reset
    internal: true
    filters:
      - invert:
      - delayed_off: 10ms
    on_press:
      - light.toggle:
          id: light_1

  - platform: gpio
    name: "Botão ${dePrep} ${Nome}"
    pin: GPIO04
    id: switch_1
    on_press:
      then:
        - light.toggle:
            id: light_1
    on_release:
      then:
        - light.toggle:
            id: light_1

output:
  - platform: gpio
    pin: GPIO12
    id: relay_1
[... 226 lines cut ...]
   13 | class APIServer;
      |       ^~~~~~~~~
In file included from src/esphome/components/api/api_server.h:12,
                 from src/esphome/components/api/api_connection.h:7,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/components/api/homeassistant_service.h: In member function 'void esphome::api::HomeAssistantServiceCallAction<Ts>::play(Ts ...)':
src/esphome/components/api/homeassistant_service.h:54:18: warning: invalid use of incomplete type 'class esphome::api::APIServer'
   54 |     this->parent_->send_homeassistant_service_call(resp);
      |                  ^~
In file included from src/esphome/components/api/api_server.h:9,
                 from src/esphome/components/api/api_connection.h:7,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/components/api/util.h:13:7: note: forward declaration of 'class esphome::api::APIServer'
   13 | class APIServer;
      |       ^~~~~~~~~
*** [.pioenvs/luzcorredor/src/esphome/components/api/api_pb2.cpp.o] Error 1
*** [.pioenvs/luzcorredor/src/esphome/components/api/subscribe_state.cpp.o] Error 1
*** [.pioenvs/luzcorredor/src/esphome/components/api/list_entities.cpp.o] Error 1
*** [.pioenvs/luzcorredor/src/esphome/components/api/api_server.cpp.o] Error 1
*** [.pioenvs/luzcorredor/src/esphome/components/api/api_connection.cpp.o] Error 1
================================================ [FAILED] Took 61.04 seconds ================================================```

**Additional information and things you've tried:**

<!-- LEAVE THIS LINE AS-IS AND DON'T DELETE IT, OTHERWISE THE ISSUE WILL BE CLOSED AUTOMATICALLY. -->

dclobato, 2020-08-07:
The same happens on fresh install of esphome on Windows 10

```
C:\Windows\system32>pip3 install esphome
Collecting esphome
  Using cached esphome-1.14.5-py2.py3-none-any.whl (1.4 MB)
Collecting colorlog==4.1.0
  Using cached colorlog-4.1.0-py2.py3-none-any.whl (14 kB)
Collecting tornado==5.1.1
  Using cached tornado-5.1.1.tar.gz (516 kB)
Collecting tzlocal==2.0.0
  Using cached tzlocal-2.0.0-py2.py3-none-any.whl (15 kB)
Collecting voluptuous==0.11.7
  Using cached voluptuous-0.11.7.tar.gz (45 kB)
Collecting protobuf==3.11.3
  Using cached protobuf-3.11.3-py2.py3-none-any.whl (434 kB)
Collecting esptool==2.8
  Using cached esptool-2.8.tar.gz (84 kB)
Collecting ifaddr==0.1.6
  Using cached ifaddr-0.1.6.tar.gz (6.5 kB)
Collecting pyserial==3.4
  Using cached pyserial-3.4-py2.py3-none-any.whl (193 kB)
Collecting paho-mqtt==1.5.0
  Using cached paho-mqtt-1.5.0.tar.gz (99 kB)
Collecting pytz==2020.1
  Using cached pytz-2020.1-py2.py3-none-any.whl (510 kB)
Collecting platformio==4.3.4
  Using cached platformio-4.3.4.tar.gz (190 kB)
Collecting PyYAML==5.3.1
  Using cached PyYAML-5.3.1-cp38-cp38-win_amd64.whl (219 kB)
Collecting colorama; sys_platform == "win32"
  Using cached colorama-0.4.3-py2.py3-none-any.whl (15 kB)
Requirement already satisfied: setuptools in c:\program files\python38\lib\site-packages (from protobuf==3.11.3->esphome) (41.2.0)
Requirement already satisfied: six>=1.9 in c:\program files\python38\lib\site-packages (from protobuf==3.11.3->esphome) (1.15.0)
Collecting pyaes
  Using cached pyaes-1.6.1.tar.gz (28 kB)
Collecting ecdsa
  Using cached ecdsa-0.15-py2.py3-none-any.whl (100 kB)
Collecting bottle<0.13
  Using cached bottle-0.12.18-py3-none-any.whl (89 kB)
Collecting click!=7.1,!=7.1.1,<8,>=5
  Using cached click-7.1.2-py2.py3-none-any.whl (82 kB)
Requirement already satisfied: requests<3,>=2.4.0 in c:\program files\python38\lib\site-packages (from platformio==4.3.4->esphome) (2.24.0)
Collecting semantic_version<3,>=2.8.1
  Using cached semantic_version-2.8.5-py2.py3-none-any.whl (15 kB)
Collecting tabulate<1,>=0.8.3
  Using cached tabulate-0.8.7-py3-none-any.whl (24 kB)
Collecting pyelftools<1,>=0.25
  Using cached pyelftools-0.26-py2.py3-none-any.whl (136 kB)
Collecting marshmallow>=2
  Downloading marshmallow-3.7.1-py2.py3-none-any.whl (45 kB)
     |████████████████████████████████| 45 kB 3.4 MB/s
Requirement already satisfied: urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1 in c:\program files\python38\lib\site-packages (from requests<3,>=2.4.0->platformio==4.3.4->esphome) (1.25.10)
Requirement already satisfied: chardet<4,>=3.0.2 in c:\program files\python38\lib\site-packages (from requests<3,>=2.4.0->platformio==4.3.4->esphome) (3.0.4)
Requirement already satisfied: certifi>=2017.4.17 in c:\program files\python38\lib\site-packages (from requests<3,>=2.4.0->platformio==4.3.4->esphome) (2020.6.20)
Requirement already satisfied: idna<3,>=2.5 in c:\program files\python38\lib\site-packages (from requests<3,>=2.4.0->platformio==4.3.4->esphome) (2.10)
Using legacy setup.py install for tornado, since package 'wheel' is not installed.
Using legacy setup.py install for voluptuous, since package 'wheel' is not installed.
Using legacy setup.py install for esptool, since package 'wheel' is not installed.
Using legacy setup.py install for ifaddr, since package 'wheel' is not installed.
Using legacy setup.py install for paho-mqtt, since package 'wheel' is not installed.
Using legacy setup.py install for platformio, since package 'wheel' is not installed.
Using legacy setup.py install for pyaes, since package 'wheel' is not installed.
Installing collected packages: colorama, colorlog, tornado, pytz, tzlocal, voluptuous, protobuf, pyserial, pyaes, ecdsa, esptool, ifaddr, paho-mqtt, bottle, click, semantic-version, tabulate, pyelftools, marshmallow, platformio, PyYAML, esphome
    Running setup.py install for tornado ... done
    Running setup.py install for voluptuous ... done
  Attempting uninstall: protobuf
    Found existing installation: protobuf 3.12.2
    Uninstalling protobuf-3.12.2:
      Successfully uninstalled protobuf-3.12.2
[... 223 lines cut ...]
                 from src\esphome\components\api\api_server.cpp:1:
src\esphome\components\api\util.h:13:7: note: forward declaration of 'class esphome::api::APIServer'
   13 | class APIServer;
      |       ^~~~~~~~~
*** [.pioenvs\luzcorredor\src\esphome\components\api\proto.cpp.o] Error 1
In file included from src\esphome\components\api\api_server.h:12,
                 from src\esphome\components\api\api_connection.h:7,
                 from src\esphome\components\api\api_connection.cpp:1:
src\esphome\components\api\homeassistant_service.h: In member function 'void esphome::api::HomeAssistantServiceCallAction<Ts>::play(Ts ...)':
src\esphome\components\api\homeassistant_service.h:54:18: warning: invalid use of incomplete type 'class esphome::api::APIServer'
   54 |     this->parent_->send_homeassistant_service_call(resp);
      |                  ^~
In file included from src\esphome\components\api\api_server.h:9,
                 from src\esphome\components\api\api_connection.h:7,
                 from src\esphome\components\api\api_connection.cpp:1:
src\esphome\components\api\util.h:13:7: note: forward declaration of 'class esphome::api::APIServer'
   13 | class APIServer;
      |       ^~~~~~~~~
*** [.pioenvs\luzcorredor\src\esphome\components\api\list_entities.cpp.o] Error 1
*** [.pioenvs\luzcorredor\src\esphome\components\api\api_pb2.cpp.o] Error 1
*** [.pioenvs\luzcorredor\src\esphome\components\api\subscribe_state.cpp.o] Error 1
*** [.pioenvs\luzcorredor\src\esphome\components\api\api_server.cpp.o] Error 1
*** [.pioenvs\luzcorredor\src\esphome\components\api\api_connection.cpp.o] Error 1
=============================================================== [FAILED] Took 72.66 seconds ===============================================================
```

pascaldornfeld, 2020-08-09:
I had the same issue on my fresh linux installation. I installed vs code and platform io plugin, updated 8266 library with platform io and now compilation works. try that.

shortRound1911, 2020-08-10:
I was getting this same error on version 1.14.3. I found some recent posts on the hass forums that said basically the same thing as @pascaldornfeld mentioned. Update all the things and use VSCode. I've been using the older Atom setup personally, so I tried updating everything and moving to VSCode. My setup was still failing to compile. What I found was that even though I had a newer version of ESP8266 environment (v2.6.2) installed on my machine, esphome was always using an older one v2.2.3. Esphome even reinstalled the older version after i updated it via platformio. It seems like this older version was causing the compile errors for me. I upgraded my esphome to the `--pre` tag (1.15.0b3 as of writing) and it seemed to update the version of the compiler being used to 2.6.0. That seemed to fix my compile issues.

Command to go to beta version:
`pip install --pre -U esphome`

Info on my platformio versions:

```PS F:\backup\2020-1-25\PlatformIO\ESPHome\ESPHome\test> esphome version
Version: 1.15.0b3
PS F:\backup\2020-1-25\PlatformIO\ESPHome\ESPHome\test> platformio update
Updating contrib-piohome                 @ 3.2.3          [Up-to-date]
Updating tool-unity                      @ 1.20500.200612 [Up-to-date]
Updating tool-scons                      @ 3.30102.0      [Up-to-date]

Platform Manager
================
Platform Espressif 32
--------
Updating espressif32                     @ 1.12.4         [Up-to-date]
Updating toolchain-xtensa32              @ 2.50200.80     [Up-to-date]
Updating framework-arduinoespressif32    @ 3.10004.200129 [Up-to-date]
Updating tool-esptoolpy                  @ 1.20600.0      [Up-to-date]
Updating tool-mkspiffs                   @ 2.230.0        [Up-to-date]

Platform Espressif 8266
--------
Updating espressif8266                   @ 2.6.2          [Up-to-date]
Updating toolchain-xtensa                @ 2.40802.200502 [Up-to-date]
Updating framework-arduinoespressif8266  @ 3.20704.0      [Up-to-date]
Updating tool-esptool                    @ 1.413.0        [Up-to-date]
Updating tool-esptoolpy                  @ 1.20800.0      [Up-to-date]

Platform Espressif 8266
--------
Updating espressif8266                   @ 2.2.3          [Detached]
Updating toolchain-xtensa                @ 2.100100.200706[Up-to-date]
Updating framework-arduinoespressif8266  @ 2.20502.0      [Up-to-date]
Updating tool-esptool                    @ 1.413.0        [Up-to-date]
Updating tool-esptoolpy                  @ 1.20600.0      [Up-to-date]

Platform Espressif 8266
--------
Updating espressif8266                   @ 2.6.0          [Detached]
Updating toolchain-xtensa                @ 2.40802.200502 [Up-to-date]
Updating framework-arduinoespressif8266  @ 3.20702.0      [Up-to-date]
Updating tool-esptool                    @ 1.413.0        [Up-to-date]
Updating tool-esptoolpy                  @ 1.20800.0      [Up-to-date]```

OttoWinter, 2020-08-11:
Yeah something appears broken with PlatformIO and older toolchain versions. As @shortRound1911 said, please try the most recent pre-release.

richardweinberger, 2020-08-18:
I' facing the same problem. Isn't the release I get via `pip install esphome` supposed to work? :-(

ilkkaolavas, 2020-09-30:
I had same issue. I upgraded from 1.14.5 to 1.15.2 with command:
pip3 install esphome
esphome version
Version: 1.15.2
And now it works. Using Ubuntu and Python3.

Drizzt321, 2020-11-05:
I'm getting this via a fresh install of HomeAssistant w/ESPHome Add-On. Running latest HassOS 4.15, and latest ESPHome Add-On which is reporting v1.15.3 of ESPHome.

closed 2021-03-26 by stale[bot]