esphome/issues #1404: WLED does not work with Hyperion as a UDP device

asonjay90, 2020-08-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

NodeMCU - ESP8266

**ESPHome version (latest production, beta, dev branch)**

beta, dev

**Affected component:**

wled
https://beta.esphome.io/components/light/index.html#wled

**Description of problem:**
I am unable to get Hyperion to successfully output to a nodeMCU setup with esphome (beta) using the wled component. Its not immediately clear if this is supported, however it is explicitly mentioned in the WLED documentation: https://github.com/Aircoookie/WLED/wiki/UDP-Realtime-Control (which is also called out in the wled_light_effect.cpp#L17

Using protocol 0 and type udp (or udpraw), I am unable to get the LEDs to turn on. The following is endlessly seen in the log while the Hyperion server is enabled. Stopping the Hyperion service results in this messages no longer being written to the log, so I am confident I have this setup correctly.

`[23:41:30][D][wled_light_effect:067]: Frame: Invalid (size=894, first=0x00).`

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: livingroom_tv_leds
  platform: ESP8266
  board: nodemcuv2

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

wled:

# LED Config
light:
  - platform: neopixelbus
    name: "TV LEDs"
    id: TV_leds
    variant: WS2812X
    pin: GPIO3
    num_leds: 298
    type: GRB
    effects:
    - wled

```
The YAML is shortened to the relevant parts.

**Logs (if applicable):**

```
INFO Reading configuration /config/esphome/livingroom_tv_leds.yaml...
INFO Starting log output from livingroom_tv_leds.local using esphome API
INFO Connecting to livingroom_tv_leds.local:6053 (10.10.0.63)
INFO Successfully connected to livingroom_tv_leds.local
[23:49:52][I][app:105]: ESPHome version 1.16.0-dev compiled on Aug  7 2020, 23:16:47
[23:49:52][C][wifi:443]: WiFi:
[23:49:52][C][wifi:303]:   SSID: [redacted]
[23:49:52][C][wifi:304]:   IP Address: 10.10.0.63
[23:49:52][C][wifi:306]:   BSSID: [redacted]
[23:49:52][C][wifi:307]:   Hostname: 'livingroom_tv_leds'
[23:49:52][C][wifi:311]:   Signal strength: -61 dB ▂▄▆█
[23:49:52][C][wifi:315]:   Channel: 6
[23:49:52][C][wifi:316]:   Subnet: 255.255.255.0
[23:49:52][C][wifi:317]:   Gateway: 10.10.0.1
[23:49:52][C][wifi:318]:   DNS1: 10.10.0.2
[23:49:52][C][wifi:319]:   DNS2: (IP unset)
[23:49:52][D][wled_light_effect:067]: Frame: Invalid (size=894, first=0x00).
[23:49:52][D][wled_light_effect:067]: Frame: Invalid (size=894, first=0x00).
[23:49:52][C][logger:185]: Logger:
[23:49:52][C][logger:186]:   Level: DEBUG
[23:49:52][C][logger:187]:   Log Baud Rate: 115200
[23:49:52][C][logger:188]:   Hardware UART: UART0
[23:49:52][C][light:177]: Light 'TV LEDs'
[23:49:52][C][light:179]:   Default Transition Length: 1.0s
[23:49:52][C][light:180]:   Gamma Correct: 2.80
[23:49:52][D][wled_light_effect:067]: Frame: Invalid (size=894, first=0x00).
[23:49:52][C][light:177]: Light 'TV LED Top'
[23:49:52][C][light:179]:   Default Transition Length: 1.0s
[23:49:52][C][light:180]:   Gamma Correct: 2.80
[23:49:52][C][light:177]: Light 'TV LED Left'
[23:49:52][C][light:179]:   Default Transition Length: 1.0s
[23:49:52][C][light:180]:   Gamma Correct: 2.80
[23:49:52][D][wled_light_effect:067]: Frame: Invalid (size=894, first=0x00).
[23:49:52][C][light:177]: Light 'TV LED Bottom'
[23:49:52][C][light:179]:   Default Transition Length: 1.0s
[23:49:52][C][light:180]:   Gamma Correct: 2.80
[23:49:52][C][light:177]: Light 'TV LED Right'
[23:49:52][C][light:179]:   Default Transition Length: 1.0s
[23:49:52][C][light:180]:   Gamma Correct: 2.80
[23:49:52][D][wled_light_effect:067]: Frame: Invalid (size=894, first=0x00).
[23:49:52][D][wled_light_effect:067]: Frame: Invalid (size=894, first=0x00).
[23:49:53][D][wled_light_effect:067]: Frame: Invalid (size=894, first=0x00).
[23:49:53][C][captive_portal:169]: Captive Portal:
[23:49:53][D][wled_light_effect:067]: Frame: Invalid (size=894, first=0x00).
[23:49:53][C][ota:029]: Over-The-Air Updates:
[23:49:53][C][ota:030]:   Address: livingroom_tv_leds.local:8266
[23:49:53][C][api:095]: API Server:
[23:49:53][C][api:096]:   Address: livingroom_tv_leds.local:6053
[23:49:53][D][wled_light_effect:067]: Frame: Invalid (size=894, first=0x00).
[23:49:53][D][wled_light_effect:067]: Frame: Invalid (size=894, first=0x00).
[23:49:53][D][wled_light_effect:067]: Frame: Invalid (size=894, first=0x00).
[23:49:53][D][wled_light_effect:067]: Frame: Invalid (size=894, first=0x00).
....
```

**Additional information and things you've tried:**

This is my Hyperion device config:

```
        // DEVICE CONFIGURATION 
        "device" :
        {
                "name"       : "Livingroom-TV",
                "type"       : "udpraw",
                "output"     : "10.10.0.63:21324",
                "rate"     : 1000000,
                "maxpacket"     : 1450,
                "protocol"     : 0,
                "colorOrder" : "rgb"
        },
```
I have 298 leds setup, so I am clearly only including the relevant part here.

https://hyperion-project.org/wiki/UDP-Device
Looking at Hyperion documentation, its clear the expected information is being sent for protocol 0:

> Packet Format protocol 0:
> 3 bytes per LED as R, G, B

298 LEDs * 3 bytes per LED = 894

Additionally, i know I have this (somewhat) working correctly as when I use protocol 2 (DRGB) I am able to get the leds to light the expected color. However the refresh rate is extremely slow (1fps) and the leds flash funky colors in between transitions.

jakehigg, 2020-10-02:
I am having this issue as well.  I get frame invalid when trying to sync from wled to esphome light effects.  I have tried both wled_notify and WARLS protocol in wled 0.10.0 and 0.10.2.  Running esphome 1.15.2.  When I sync from wled devices I receive [wled_light_effect:067]: Frame: Invalid (size=29, first 0x00) and 0x01 (when using WARLS). It appears to be a bug in wled_light_effect but I cant easily troubleshoot.  Have you tried to sync from wled?

d-two, 2020-10-28:
I have make a PR

https://github.com/esphome/esphome/pull/1339

DrBlokmeister, 2021-01-23:
@d-two which version of Hyperion do you use? Do you use hyperion.ng? I cannot find the 'protocol' setting in hyperion.ng and can therefore not get this to work.

Is there something I can do in order to get this to work before the pull request comes through?

d-two, 2021-01-23:
> @d-two which version of Hyperion do you use? Do you use hyperion.ng? I cannot find the 'protocol' setting in hyperion.ng and can therefore not get this to work.
> 
> Is there something I can do in order to get this to work before the pull request comes through?

This is not working for Hyperion.NG only for the first version of Hyperion...

rradar, 2021-01-31:
> This is not working for Hyperion.NG only for the first version of Hyperion...

Too bad... I tried your PR (https://github.com/esphome/esphome/pull/1339) now the second time and wondered why I always ended up with `[wled_light_effect:067]: Frame: Invalid` errors the PR actually should fix.

Do you @d-two know what's needed to make this also work with Hyperion.NG?

marrobHD, 2021-04-24:
Still the same issue when using official wled firmware. 
```
[20:58:56][D][wled_light_effect:067]: Frame: Invalid (size=44, first=0xFF).
[20:59:25][D][wled_light_effect:067]: Frame: Invalid (size=44, first=0xFF).
```

bjornstevens, 2021-04-28:
For Hyperion you can just easily use E1.31 to output?

I do experience issues syncing my led strip trough the WLED sync..

closed 2021-05-05 by jesserockz