esphome/issues #1406: ADC value as condition under interval

timelyfiasco, 2020-08-10:
Operating environment/Installation (Hass.io/Docker/pip/etc.):

   _arch/pip_
ESP (ESP32/ESP8266, Board/Sonoff):

   _esp12_
ESPHome version (latest production, beta, dev branch)

   _v1.14.5_
Affected component:

   _ADC's state parameter_  https://esphome.io/components/sensor/adc.html

Description of problem:
   _I want to get the value of ADC which is given the unit ms and use it as timer to turn off a switch. I cant get the value of ADC using ID.state. Am I wrong to use state?_

Problem-relevant YAML-configuration entries:
```yaml
  - platform: adc 
	id: weld_duration
	name: "Duration"
	pin: A0
	unit_of_measurement: "ms"
	accuracy_decimals: 4
	update_interval: 1s

interval:
  - interval: 1s
	then:
		- if:
			condition:
			  - lambda: 'return id(weld_duration).state <= id(start_stopwatch);'
			then:
			  - switch.turn_off: relay0
			  - lambda: |- 
				  id(start_stopwatch) = 0;
			  - logger.log: 
				  format: "weld_duration %.4d"
				  args: [ 'id(weld_duration).state' ]

```

Logs (if applicable):

```
[09:53:51][D][adc:056]: 'Duration': Got voltage=0.54V
[09:53:51][D][sensor:092]: 'Duration': Sending state 0.54492 ms with 4 decimals of accuracy
[09:53:52][D][main:608]: weld_duration 0000
[09:53:53][D][main:608]: weld_duration 0000
[09:53:53][D][adc:056]: 'Duration': Got voltage=0.54V
[09:53:53][D][sensor:092]: 'Duration': Sending state 0.54297 ms with 4 decimals of accuracy
[09:53:54][D][main:608]: weld_duration 0000
[09:53:55][D][main:608]: weld_duration 0000

```

Additional information and things you've tried:

glmnet, 2020-08-10:
that should work as is, the id(weld_duration).state will give you values from 0.0 to 1.0, i.e. that is a float value. You can use filters if you need to change the value.

You see `weld_duration 0000` because of the format specifier `%.4d` is for integers, try `%.4f` instead.

timelyfiasco, 2020-08-12:
Thank you glmnet. rookie mistake.

closed 2020-08-12 by timelyfiasco