esphome/issues #1407: ESPHome 1.14.3 no updates Hass.io Addon

vvv850, 2020-08-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

NUC
Ubunt 19.10
Docker version 19.03.8
HA 0.113.3

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**ESPHome version (latest production, beta, dev branch)**

ESPHome 1.14.3

**Affected component:**

-

**Description of problem:**

When the 1.14.4 version was available, I saw for a couple of days and it disappeared. So I remained on 1.14.3 after seeing there were some issues with 1.14.4.
Now I saw that 1.14.5 is available but it does not show up in the Supervisor section. Checking the add-on store I see it's available, but not as an update but as a new install. It seems like my current install is not tied to what's available in the add-on store.

**Problem-relevant YAML-configuration entries:**
```yaml

no special config, just:
status_use_ping: true

```

**Logs (if applicable):**

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] 00-banner.sh: executing... 

-----------------------------------------------------------
 Hass.io Add-on: ESPHome
 ESPHome Hass.io add-on for intelligently managing all your ESP8266/ESP32 devices.
-----------------------------------------------------------
 Add-on version: 1.14.3
 There is an update available for this add-on!
 **Latest add-on version: null**
 Please consider upgrading as soon as possible.
 System: Ubuntu 19.10  (amd64 / intel-nuc)
![esp3](https://user-images.githubusercontent.com/50616451/89774654-f387a680-db0e-11ea-9910-ae0518457f45.png)
![esp2](https://user-images.githubusercontent.com/50616451/89774656-f4b8d380-db0e-11ea-8945-909e54407ebe.png)
![esp1](https://user-images.githubusercontent.com/50616451/89774658-f5516a00-db0e-11ea-8466-12e5b1c131b4.png)

 Home Assistant version: 0.113.3
 Supervisor version: 229
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
[cont-init.d] 00-banner.sh: exited 0.
[cont-init.d] 01-log-level.sh: executing... 
[cont-init.d] 01-log-level.sh: exited 0.
[cont-init.d] 10-requirements.sh: executing... 
[cont-init.d] 10-requirements.sh: exited 0.
[cont-init.d] 20-nginx.sh: executing... 
[cont-init.d] 20-nginx.sh: exited 0.
[cont-init.d] 30-esphome.sh: executing... 
[cont-init.d] 30-esphome.sh: exited 0.
[cont-init.d] 40-migrate.sh: executing... 
[cont-init.d] 40-migrate.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[22:15:43] INFO: Waiting for dashboard to come up...
[22:15:43] INFO: Starting ESPHome dashboard...
INFO Starting dashboard web server on unix socket /var/run/esphome.sock and configuration dir /config/esphome...
[22:15:43] INFO: Starting NGINX...

```

**Additional information and things you've tried:**

strumf666, 2020-08-30:
I have the same issue running home assistant 0.114.4 on rpi4

strumf666, 2020-11-02:
A bit of an update, I have another issue, possibly connected with this; compiling doesn't work; logs bellow.

`INFO Reading configuration /config/esphome/Esp32_dnevna.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/esp32_living_room
Processing esp32_living_room (board: lolin_d32_pro; framework: arduino; platform: espressif32@1.11.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Looking for AsyncTCP-esphome library in registry
Warning! You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchains.
Warning! You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchains.
Dependency Graph
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <Wire> 1.0.1
|-- <Update> 1.0
|-- <WiFi> 1.0
Compiling /data/esp32_living_room/.pioenvs/esp32_living_room/src/esphome/components/api/api_connection.cpp.o
Compiling /data/esp32_living_room/.pioenvs/esp32_living_room/src/esphome/components/api/api_server.cpp.o
Compiling /data/esp32_living_room/.pioenvs/esp32_living_room/src/esphome/components/api/list_entities.cpp.o
Compiling /data/esp32_living_room/.pioenvs/esp32_living_room/src/esphome/components/api/subscribe_state.cpp.o
In file included from src/esphome/components/api/api_connection.h:7:0,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/components/api/api_server.h:16:22: fatal error: AsyncTCP.h: No such file or directory

******************************************************************
* Looking for AsyncTCP.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:AsyncTCP.h"
* Web  > https://platformio.org/lib/search?query=header:AsyncTCP.h
*
******************************************************************

compilation terminated.
*** [/data/esp32_living_room/.pioenvs/esp32_living_room/src/esphome/components/api/api_connection.cpp.o] Error 1
In file included from src/esphome/components/api/api_server.cpp:1:0:
src/esphome/components/api/api_server.h:16:22: fatal error: AsyncTCP.h: No such file or directory

******************************************************************
* Looking for AsyncTCP.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:AsyncTCP.h"
* Web  > https://platformio.org/lib/search?query=header:AsyncTCP.h
*
******************************************************************

compilation terminated.
*** [/data/esp32_living_room/.pioenvs/esp32_living_room/src/esphome/components/api/api_server.cpp.o] Error 1
In file included from src/esphome/components/api/list_entities.h:52:0,
                 from src/esphome/components/api/list_entities.cpp:1:
src/esphome/components/api/api_server.h:16:22: fatal error: AsyncTCP.h: No such file or directory

******************************************************************
* Looking for AsyncTCP.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:AsyncTCP.h"
* Web  > https://platformio.org/lib/search?query=header:AsyncTCP.h
*
******************************************************************

compilation terminated.
*** [/data/esp32_living_room/.pioenvs/esp32_living_room/src/esphome/components/api/list_entities.cpp.o] Error 1
In file included from src/esphome/components/api/subscribe_state.h:47:0,
                 from src/esphome/components/api/subscribe_state.cpp:1:
src/esphome/components/api/api_server.h:16:22: fatal error: AsyncTCP.h: No such file or directory

******************************************************************
[... 36 lines cut ...]
[cont-init.d] 10-requirements.sh: exited 0.
[cont-init.d] 20-nginx.sh: executing... 
[cont-init.d] 20-nginx.sh: exited 0.
[cont-init.d] 30-esphome.sh: executing... 
[cont-init.d] 30-esphome.sh: exited 0.
[cont-init.d] 40-migrate.sh: executing... 
[cont-init.d] 40-migrate.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[01:42:36] INFO: Waiting for dashboard to come up...
[01:42:36] INFO: Starting ESPHome dashboard...
INFO Starting dashboard web server on unix socket /var/run/esphome.sock and configuration dir /config/esphome...
[01:42:38] INFO: Starting NGINX...
INFO 304 GET / (0.0.0.0) 46.29ms
INFO 101 GET /ace (0.0.0.0) 1.92ms
INFO Running command 'esphome --dashboard -q /config/esphome vscode --ace'
INFO 101 GET /upload (0.0.0.0) 1.65ms
INFO Running command 'esphome --dashboard /config/esphome/Esp32_dnevna.yaml run --upload-port /dev/ttyUSB0'
INFO Process exited with return code 1
INFO 101 GET /logs (0.0.0.0) 1.77ms
INFO Running command 'esphome --dashboard /config/esphome/esp32_kino.yaml logs --serial-port OTA'
INFO 304 GET / (0.0.0.0) 8.61ms
INFO 101 GET /ace (0.0.0.0) 2.13ms
INFO Running command 'esphome --dashboard -q /config/esphome vscode `--ace'`

khenderick, 2021-02-16:
Not sure if you ever found out the solution yourself, but this helped me:

```
esphome:
  ...
  platformio_options:
    lib_deps:
      ...
      - AsyncTCP
...
```

closed 2021-06-30 by stale[bot]