esphome/issues #1409: Dallas - 85 degree postings

trp79, 2020-08-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**ESPHome version (latest production, beta, dev branch)**

1.14.5 (Home Assistant 0.113.3)

**Affected component:**

https://esphome.io/components/sensor/dallas.html

**Description of problem:**
As shown in below screen dump I get incorrect temperature readings from time to time as it reads 85 degree Celsius. I do not see a specific pattern, it seems to occur randomly. I have the one-wire version.
I see, when googling, that I'm not the only one having this issue with the sensor, it does as well not seem to be esphome related, I do however see a solution for the problem describe [here](https://github.com/espressif/arduino-esp32/issues/1050): however not sure if esphome is already implemented like described there
"_You need to to add sensor_t1.begin() to your setup() and you have tell the sensors to do a conversion with sensor_t1.requestTemperatures(). Before you can read the temperature with sensor_t1.getTempCByIndex(0)._"

[image]

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  display_name: DIY_Solar_1
  
esphome:
  name: solar_1
  platform: ESP32
  board: esp-wrover-kit

status_led:
  pin: GPIO2
  
wifi:
  ssid: !secret wifi_iot_ssid
  password: !secret wifi_iot_pass
  fast_connect: true
  
  # Optional manual IP
  manual_ip:
    static_ip: !secret ip_static_diy_solar_1
    gateway: !secret ip_gateway
    subnet: !secret ip_subnet
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
#    ssid: "Altan Fallback Hotspot"
    ssid: ${display_name}_AP
    password: !secret esphome_fallback_hotspot

captive_portal:

# Enable logging
logger:
  #level: INFO #set to INFO to save a bit of time (hence battery)

# MQTT preferred over API when using deep sleep
mqtt:
  broker: !secret mqttip
  port: !secret mqttport
  username: !secret mqttuser
  password: !secret mqttpass
  birth_message: #Setting the MQTT Birth and Will message to blank stops the device from going Unavailable while it is asleep which messes with history graphs (lots of dots/broken lines)
  will_message:
  on_message:
    - topic: ${display_name}/ota_mode
      payload: 'ON'
      then:
        - deep_sleep.prevent: deep_sleep_1
    - topic: ${display_name}/sleep_mode
      payload: 'ON'
      then:
        - deep_sleep.enter: deep_sleep_1

# Enable Home Assistant API
#api:
#  password: !secret esphome_oat_pass

ota:
  password: !secret esphome_oat_pass

#Deep sleep
deep_sleep:
#  run_duration: 30min
  run_duration: 20s
  sleep_duration: 15min
  id: deep_sleep_1
  
# Dallas sensor/hub component (in my case for the DS18B20 Waterproof Digital Temperature Temp Sensor Probe)
dallas:
  - pin: GPIO4
    update_interval: 5s
    id: dallas_1
  
sensor:
  - platform: dallas
    address: 0x5E02131E08F5AA28
    name: ${display_name} Temperature
    id: dallas_temp_1
    
  - platform: adc
    pin: 39
    name: ${display_name} Battery Voltage
    update_interval: 5s
    attenuation: 2.5db
    accuracy_decimals: 1
    id: battery_voltage_1
    filters:
      - multiply: 4.268367

```

**Logs (if applicable):**

```
[15:29:18][I][deep_sleep:067]: Beginning Deep Sleep
[15:32:36][I][mqtt:202]: MQTT Connected!
[15:32:36][C][deep_sleep:013]: Setting up Deep Sleep...
[15:32:36][I][app:058]: setup() finished successfully!
[15:32:36][I][app:100]: ESPHome version 1.14.5 compiled on Aug 10 2020, 14:30:51
[15:32:36][C][status_led:019]: Status LED:
[15:32:36][C][status_led:020]:   Pin: GPIO2 (Mode: OUTPUT)
[15:32:36][C][wifi:415]: WiFi:
[15:32:36][C][wifi:283]:   SSID: [redacted]
[15:32:36][C][wifi:284]:   IP Address: 10.0.30.50
[15:32:36][C][wifi:286]:   BSSID: [redacted]
[15:32:36][C][wifi:287]:   Hostname: 'solar_1'
[15:32:36][C][wifi:291]:   Signal strength: -82 dB ▂▄▆█
[15:32:36][C][wifi:295]:   Channel: 1
[15:32:36][C][wifi:296]:   Subnet: 255.255.255.0
[15:32:36][C][wifi:297]:   Gateway: 10.0.30.1
[15:32:36][C][wifi:298]:   DNS1: 0.0.0.0
[15:32:36][C][wifi:299]:   DNS2: 0.0.0.0
[15:32:36][C][logger:175]: Logger:
[15:32:36][C][logger:176]:   Level: DEBUG
[15:32:36][C][logger:177]:   Log Baud Rate: 115200
[15:32:36][C][logger:178]:   Hardware UART: UART0
[15:32:36][C][adc:026]: ADC Sensor 'DIY_Solar_1 Battery Voltage'
[15:32:36][C][adc:026]:   Unit of Measurement: 'V'
[15:32:36][C][adc:026]:   Accuracy Decimals: 1
[15:32:36][C][adc:026]:   Icon: 'mdi:flash'
[15:32:36][C][adc:035]:   Pin: 39
[15:32:36][C][adc:041]:  Attenuation: 2.5db (max 1.5V)
[15:32:36][C][adc:051]:   Update Interval: 5.0s
[15:32:36][C][dallas.sensor:072]: DallasComponent:
[15:32:36][C][dallas.sensor:073]:   Pin: GPIO4 (Mode: INPUT)
[15:32:36][C][dallas.sensor:074]:   Update Interval: 5.0s
[15:32:36][D][dallas.sensor:079]:   Found sensors:
[15:32:36][D][dallas.sensor:082]:     0x5E02131E08F5AA28
[15:32:36][C][dallas.sensor:087]:   Device 'DIY_Solar_1 Temperature'
[15:32:36][C][dallas.sensor:087]:     Unit of Measurement: '°C'
[15:32:36][C][dallas.sensor:087]:     Accuracy Decimals: 1
[15:32:36][C][dallas.sensor:087]:     Icon: 'mdi:thermometer'
[15:32:36][C][dallas.sensor:095]:     Address: 0x5E02131E08F5AA28
[15:32:36][C][dallas.sensor:096]:     Resolution: 12
[15:32:36][C][captive_portal:169]: Captive Portal:
[15:32:36][C][ota:029]: Over-The-Air Updates:
[15:32:36][C][ota:030]:   Address: 10.0.30.50:3232
[15:32:36][C][ota:032]:   Using Password.
[15:32:36][C][mqtt:051]: MQTT:
[15:32:36][C][mqtt:053]:   Server Address: 10.0.30.204:1883 (10.0.30.204)
[15:32:36][C][mqtt:054]:   Username: [redacted]
[15:32:36][C][mqtt:055]:   Client ID: [redacted]
[15:32:36][C][mqtt:057]:   Discovery prefix: 'homeassistant'
[15:32:36][C][mqtt:058]:   Discovery retain: YES
[15:32:36][C][mqtt:060]:   Topic Prefix: 'solar_1'
[15:32:36][C][mqtt:062]:   Log Topic: 'solar_1/debug'
[15:32:36][C][mqtt:551]: MQTT Message Trigger:
[15:32:36][C][mqtt:552]:   Topic: 'DIY_Solar_1/ota_mode'
[15:32:36][C][mqtt:553]:   QoS: 0
[15:32:36][C][mqtt.sensor:024]: MQTT Sensor 'DIY_Solar_1 Temperature':
[15:32:36][C][mqtt.sensor:028]:   State Topic: 'solar_1/sensor/diy_solar_1_temperature/state'
[15:32:36][C][mqtt.sensor:024]: MQTT Sensor 'DIY_Solar_1 Battery Voltage':
[15:32:36][C][mqtt.sensor:028]:   State Topic: 'solar_1/sensor/diy_solar_1_battery_voltage/state'
[15:32:36][C][mqtt:551]: MQTT Message Trigger:
[15:32:36][C][mqtt:552]:   Topic: 'DIY_Solar_1/sleep_mode'
[15:32:36][C][mqtt:553]:   QoS: 0
[15:32:36][C][deep_sleep:020]: Setting up Deep Sleep...
[15:32:36][C][deep_sleep:023]:   Sleep Duration: 900000 ms
[15:32:36][C][deep_sleep:026]:   Run Duration: 20000 ms
[15:32:37][D][dallas.sensor:153]: 'DIY_Solar_1 Temperature': Got Temperature=85.0°C
[15:32:37][D][sensor:092]: 'DIY_Solar_1 Temperature': Sending state 85.00000 °C with 1 decimals of accuracy
[15:32:37][D][adc:056]: 'DIY_Solar_1 Battery Voltage': Got voltage=0.93V
[15:32:37][D][sensor:092]: 'DIY_Solar_1 Battery Voltage': Sending state 3.97130 V with 1 decimals of accuracy
[15:32:42][D][dallas.sensor:153]: 'DIY_Solar_1 Temperature': Got Temperature=85.0°C
[15:32:42][D][sensor:092]: 'DIY_Solar_1 Temperature': Sending state 85.00000 °C with 1 decimals of accuracy
[15:32:42][D][adc:056]: 'DIY_Solar_1 Battery Voltage': Got voltage=0.93V
[15:32:42][D][sensor:092]: 'DIY_Solar_1 Battery Voltage': Sending state 3.97756 V with 1 decimals of accuracy
[15:32:47][D][dallas.sensor:153]: 'DIY_Solar_1 Temperature': Got Temperature=85.0°C
[15:32:47][D][sensor:092]: 'DIY_Solar_1 Temperature': Sending state 85.00000 °C with 1 decimals of accuracy
[15:32:47][D][adc:056]: 'DIY_Solar_1 Battery Voltage': Got voltage=0.93V
[15:32:47][D][sensor:092]: 'DIY_Solar_1 Battery Voltage': Sending state 3.97756 V with 1 decimals of accuracy
[15:32:52][D][dallas.sensor:153]: 'DIY_Solar_1 Temperature': Got Temperature=85.0°C
[15:32:52][D][sensor:092]: 'DIY_Solar_1 Temperature': Sending state 85.00000 °C with 1 decimals of accuracy
[15:32:52][D][adc:056]: 'DIY_Solar_1 Battery Voltage': Got voltage=0.93V
[15:32:52][D][sensor:092]: 'DIY_Solar_1 Battery Voltage': Sending state 3.97443 V with 1 decimals of accuracy
[15:32:56][I][deep_sleep:067]: Beginning Deep Sleep

```

**Additional information and things you've tried:**

evildad, 2020-08-11:
The 1820 gives temperature 85 in case there is some sort of communication failure, so check your cables and connections.
Or just filter those 85 degree readings out using:
https://esphome.io/components/sensor/index.html#filter-out

trp79, 2020-08-11:
Thanks for the reply evildad 👍

I'm however not sure it is a pure communication error:
- I get the battery voltage without issues as you see in the log
- I have other ESPHOME devices (however not with the 1820 sensor), I do not see any communication issues with these

I will filter out 85 when I move it to my "productive" setup as a start.

Thanks and KR
Torben

> The 1820 gives temperature 85 in case there is some sort of communication failure, so check your cables and connections.
> Or just filter those 85 degree readings out using:
> https://esphome.io/components/sensor/index.html#filter-out
> 
> ` - platform: dallas`
> ` filters:`
> ` - filter_out: 85`

evildad, 2020-08-11:
Communication Error with the 1820 sensor and not with something else.
I occasionally have those 85 readings with my 1820 sensors, too and that's why i just filter them out.

Seems like a common issue for those cheap sensors... just google 1820 and 85 and we are not alone :-)

rradar, 2020-08-15:
Some more useful insights found here https://github.com/cpetrich/counterfeit_DS18B20

For the original maxim dallas sensors 85°C is actually the "power-up temperature"

CarlosGS, 2020-10-28:
Also observing some spurious values, I thought it was related to `power_save_mode: light`
[image]

CarlosGS, 2020-10-29:
Oh, were you all using ESP32?

closed 2021-03-13 by stale[bot]