esphome/issues #1411: AS3935_i2C -  Read failed!

zibous, 2020-08-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
pip, OSX High Sierra

**ESP (ESP32/ESP8266, Board/Sonoff):**
nodemcuv2 and Wemos D1 Mini

**ESPHome version (latest production, beta, dev branch)**
ESPHome v1.15.0b3

**Affected component:**
as3935

**Description of problem:**
Has anyone managed to get one of these working? I have tried and tried but not a squeak of output.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: blitzsensor
  platform: ESP8266
  board: nodemcuv2
  platformio_options:
    upload_speed: 115200
  
wifi:
  ssid:  !secret ssid_name
  password: !secret ssid_pswd

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Messstation Hotspot"
    password: "tMtkdRRAxEnO"

captive_portal:

# Enable logging
# see: https://esphome.io/components/logger.html?highlight=logging
logger:
  level: DEBUG
  
# Enable Home Assistant API
api:
  password: !secret ha_pswd

ota:
  password: !secret ota_pswd

web_server:
  port: 80

mqtt:
  broker: !secret mqtt_broker
  username: !secret mqtt_username
  password: !secret mqtt_password
  birth_message:
    topic: tele/blitzsensor/LWT
    payload: online
  will_message:
    topic: tele/blitzsensor/LWT
    payload: offline

# connection ams as3935 Franklin Lightning Sensor
i2c:
  scl: D1
  sda: D2
  scan: True
  
as3935_i2c:
  irq_pin: GPIO13
  indoor: False
  spike_rejection: 5

sensor:
  - platform: as3935
    lightning_energy:
      name: "ESP Blitz Energie"
    distance:
      name: "ESP Sturm entfernung"
  - platform: wifi_signal
    name: "WiFi Signal"
    update_interval: 60s

# see https://esphome.io/components/sensor/as3935.html?highlight=as3935
binary_sensor:
  - platform: as3935
    name: "ESP Sturmwarnung"

```

**Logs (if applicable):**

```
[14:51:15][C][ota:366]: There have been 1 suspected unsuccessful boot attempts.
[14:51:15][I][app:029]: Running through setup()...
[14:51:15][C][as3935:010]: Setting up AS3935...
[14:51:15][C][as3935:013]:   IRQ Pin: GPIO13 (Mode: INPUT)
[14:51:15][C][wifi:033]: Setting up WiFi...
[14:51:15][D][wifi:324]: Starting scan...
[14:51:15][D][binary_sensor:036]: 'ESP Sturmwarnung': Sending state OFF
....
14:52:09][D][sensor:092]: 'WiFi Signal': Sending state -76.00000 dB with 0 decimals of accuracy
[14:52:31][W][i2c:086]: Requesting 1 bytes from 0x03 failed!
[14:52:31][W][as3935_i2c:029]: Read failed!
[14:52:31][W][i2c:070]: Received NACK on transmit of address 0x03
[14:52:31][W][as3935_i2c:029]: Read failed!

```

**Additional information and things you've tried:**

Name | State | Actions
-- | -- | --
WiFi Signal | -79 dB |  
ESP Sturm entfernung |   |  
ESP Blitz Energie |   |  
ESP Sturmwarnung | OFF |  

Sensor:
[image]

I don't get any data (distance, energy) and the status is always "OFF".
What am I doing wrong or what is the solution?

glmnet, 2020-08-14:
You should see on the logs before the lines you're posting that an I2C device was found on your bus and it should tell you the address the device has. 

cc @TheEggi

zibous, 2020-08-14:
Log see    **Component was marked as failed.**

```text
Processing blitzsensor (board: nodemcuv2; framework: arduino; platform: espressif8266@2.6.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20702.0 (2.7.2) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)

Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.7
|   |-- <ESPAsyncTCP-esphome> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|-- <AsyncMqttClient-esphome> 0.8.4
|   |-- <ESPAsyncTCP-esphome> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <Wire> 1.0
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
....

[12:12:16][C][ota:366]: There have been 2 suspected unsuccessful boot attempts.
[12:12:16][I][app:029]: Running through setup()...
[12:12:16][C][as3935:010]: Setting up AS3935...
[12:12:16][C][as3935:013]:   IRQ Pin: GPIO13 (Mode: INPUT)
[12:12:16][W][i2c:070]: Received NACK on transmit of address 0x03
[12:12:16][E][component:092]: Component was marked as failed.
[12:12:16][W][as3935_i2c:013]: read_byte failed - increase log level for more details!
[12:12:16][D][text_sensor:015]: 'ESPHome Version': Sending state '1.15.0b3 Aug 14 2020, 11:30:31'
[12:12:16][C][wifi:033]: Setting up WiFi...
[12:12:16][D][wifi:324]: Starting scan...
[12:12:16][D][sensor:092]: 'Blitz Sensor Uptime': Sending state 0.14100 s with 0 decimals of accuracy
[12:12:27][I][app:059]: setup() finished successfully!
[12:12:27][D][sensor:092]: 'Blitzsensor WiFi Signal': Sending state -57.00000 dB with 0 decimals of accuracy
[12:12:27][I][app:105]: ESPHome version 1.15.0b3 compiled on Aug 14 2020, 11:30:31

[[12:12:27][C][i2c:028]: I2C Bus:
[12:12:27][C][i2c:029]:   SDA Pin: GPIO4
[12:12:27][C][i2c:030]:   SCL Pin: GPIO5
[12:12:27][C][i2c:031]:   Frequency: 50000 Hz
[12:12:27][I][i2c:033]: Scanning i2c bus for active devices...
[12:12:27][I][i2c:040]: Found i2c device at address 0x03

[12:12:27][D][sntp:059]: Synchronized time: Fri Aug 14 10:12:27 2020
[12:12:27][C][uptime.sensor:030]: Uptime Sensor 'Blitz Sensor Uptime'
[12:12:27][C][uptime.sensor:030]:   Unit of Measurement: 's'
[12:12:27][C][uptime.sensor:030]:   Accuracy Decimals: 0
[12:12:27][C][uptime.sensor:030]:   Icon: 'mdi:timer-outline'

[12:12:27][C][logger:185]: Logger:
[12:12:27][C][logger:186]:   Level: DEBUG
[12:12:27][C][logger:187]:   Log Baud Rate: 115200
[12:12:27][C][logger:188]:   Hardware UART: UART0
[12:12:27][C][logger:190]:   Level for 'as3935_i2c.component': DEBUG

[12:12:27][C][as3935:027]: AS3935:
[12:12:27][C][as3935:028]:   Interrupt Pin: GPIO13 (Mode: INPUT)
[12:12:27][C][as3935:029]:   Thunder alert 'Blitz Sensor Sturm Warnung'
[12:12:27][C][as3935:030]:   Distance 'Blitz Sensor Sturm Entfernung'
[12:12:27][C][as3935:030]:     Unit of Measurement: 'km'
[12:12:27][C][as3935:030]:     Accuracy Decimals: 1
[12:12:27][C][as3935:030]:     Icon: 'mdi:signal'
[12:12:27][C][as3935:031]:   Lightning energy 'blitzsensor Blitz Energie'
[12:12:27][C][as3935:031]:     Unit of Measurement: ''
[12:12:27][C][as3935:031]:     Accuracy Decimals: 1
[12:12:27][C][as3935:031]:     Icon: 'mdi:flash'

... Restart
12:17:47][C][ota:366]: There have been 0 suspected unsuccessful boot attempts.
[12:17:47][I][app:029]: Running through setup()...

[12:17:47][C][as3935:010]: Setting up AS3935...
[12:17:47][C][as3935:013]:   IRQ Pin: GPIO13 (Mode: INPUT)
[12:17:47][W][i2c:070]: Received NACK on transmit of address 0x03
[12:17:47][E][component:092]: Component was marked as failed.
[12:17:47][W][as3935_i2c:013]: read_byte failed - increase log level for more details!

[12:17:47][D][text_sensor:015]: 'ESPHome Version': Sending state '1.15.0b3 Aug 14 2020, 11:30:31'
[12:17:47][C][wifi:033]: Setting up WiFi...
...
[12:17:56][C][i2c:028]: I2C Bus:
[12:17:56][C][i2c:029]:   SDA Pin: GPIO4
[12:17:56][C][i2c:030]:   SCL Pin: GPIO5
[12:17:56][C][i2c:031]:   Frequency: 50000 Hz
[12:17:56][I][i2c:033]: Scanning i2c bus for active devices...
[12:17:56][I][i2c:040]: Found i2c device at address 0x03
[12:17:56][D][sntp:059]: Synchronized time: Fri Aug 14 10:17:56 2020
[12:17:56][C][uptime.sensor:030]: Uptime Sensor 'Blitz Sensor Uptime'
[12:17:56][C][uptime.sensor:030]:   Unit of Measurement: 's'
[12:17:56][C][uptime.sensor:030]:   Accuracy Decimals: 0
[12:17:56][C][uptime.sensor:030]:   Icon: 'mdi:timer-outline'

[12:17:56][C][logger:185]: Logger:
[12:17:56][C][logger:186]:   Level: DEBUG
[12:17:56][C][logger:187]:   Log Baud Rate: 115200
[12:17:56][C][logger:188]:   Hardware UART: UART0
[12:17:56][C][logger:190]:   Level for 'as3935_i2c.component': DEBUG

[12:17:56][C][as3935:027]: AS3935:
[12:17:56][C][as3935:028]:   Interrupt Pin: GPIO13 (Mode: INPUT)
[12:17:56][C][as3935:029]:   Thunder alert 'Blitz Sensor Sturm Warnung'
[12:17:56][C][as3935:030]:   Distance 'Blitz Sensor Sturm Entfernung'
[12:17:56][C][as3935:030]:     Unit of Measurement: 'km'
[12:17:56][C][as3935:030]:     Accuracy Decimals: 1
[12:17:56][C][as3935:030]:     Icon: 'mdi:signal'
[12:17:56][C][as3935:031]:   Lightning energy 'blitzsensor Blitz Energie'
[12:17:56][C][as3935:031]:     Unit of Measurement: ''
[12:17:56][C][as3935:031]:     Accuracy Decimals: 1
[12:17:56][C][as3935:031]:     Icon: 'mdi:flash'
[12:17:56][C][as3935_i2c:036]:   Address: 0x03
```

glmnet, 2020-08-14:
Did you try a full power cycle? Sometimes devices enter in sleep mode and they will only accept an specific wake up command only. (This is generally speaking, not sure about this sensor).

zibous, 2020-08-14:
Thanks for your information
> Did you try a full power cycle

@glmnet 
How can I set the 'full power cycle'?

This is my wiring of the sensor to the NodeMCU.
[image]

[image]

Name | State | Actions
-- | -- | --
Blitz Sensor Uptime | 388 s |  
Blitzsensor WiFi Signal | -75 dB |  
ADC Voltage | 2.89 V |  
Blitz Sensor Sturm Entfernung |   |  
blitzsensor Blitz Energie |   |  
Blitz Sensor Sturm Warnung | OFF |  
blitzsensor ESP Version | 1.15.0b3 Aug 14 2020, 19:29:45 |  
Blitz Sensor IP | 10.1.1.93 |  
Blitz Sensor SSID | smarthome |  
Blitz Sensor BSSID | 88:1F:A1:DD:3E:82

I am not receiving data in any mode, am I doing something wrong or have I bought the wrong sensor?

randybb, 2020-08-14:
How are you switching between I2C and SPI? Just by connecting it to a different bus?
Here is connection diagram from its datasheet:
[image]

glmnet, 2020-08-14:
I was referring to a complete power down and then power up again, when you download a new program or reset the ESP the other device does not know and might still be in an unexpected state.

I am curious why are you powering it via `VU` (tbh I didn't know this pin will give you 5V power from the USB directly), but this sensor seems to work with 3v3 so unless it is really power hungry you should connect it to 3v3 so SDA and SCL levels are 3v3 too

Did you try that already?

zibous, 2020-08-15:
@randybb

Thank you for the information. As I can see from this, the mode must be made via the SI wiring. I did this directly on the connection contacts.

[image]

> How are you switching between I2C and SPI? Just by connecting it to a different bus?

Remove the NodeMCU from the interface, re-route the cable (ic2 or SPI) then deploy the application.

@glmnet 
Yes, I tested it, but no change

Result of the settings - **Cannot get any to work**, most likely the SPI MODE variant. But I would much rather use ic2 MODE, as it does not occupy as many ports.

### SPI MODE: as3935

| PIN  | AS3935 | NODEMCU |                                                              |
| ---- | ------ | ------- | ------------------------------------------------------------ |
| 1    | VCC    | 3.3V    | Positive supply voltage                                      |
| 2    | GND    | GND     | Ground                                                       |
| 3    | SCL    | D5      | GPIO14 Digital input I2C clock bus or SPI clock bus (according to SI setting) |
| 4    | MOSI   | D7      | GPIO12 Digital I/O with pull-up / Digital input I2C data bus or SPI data input bus |
| 5    | MISO   | D6      | GPIO13 Digital output SPI data output bus                    |
| 6    | IRQ    | D1      | GPIO5  IRQ Digital output Interrupt                          |
| 7    | SI     | GND     | **Select Interface  GND → SPI**                              |
| 8    | CS     | D8      | GPIO15 Digital input, Chip select (active low)               |

## **Error**

When I click nearby with a piezo BBQ lighter, I get the message "Disturber was detected ", but no information about "lightning_energy" and "distance"

Almost works, only after a short time, it is restarted ...

```text
V][as3935:178]: Calling read_interrupt_register_
[V][as3935_spi:043]: read_register_: 4
[I][as3935:044]: Disturber was detected - try increasing the spike rejection value!
[D][binary_sensor:036]: 'as3935_spi Sturm Warnung': Sending state OFF
[I][ota:046]: Boot seems successful, resetting boot loop counter.

[E][api:084]: No client connected to API. Rebooting...

```

### ic2 MODE: as3935 (Preferred setting)

  | PIN | AS3935 | NODEMCU |   |  
-- | -- | -- | -- | -- | --
  | 1 | VCC | 3.3V | Positive supply voltage |  
  | 2 | GND | GND | Ground |  
  | 3 | SCL | D1 | Digital input I2C clock bus or SPI clock bus (according to SI setting) |  
  | 4 | SDA | D2 | Digital I/O with pull-up / Digital input I2C data bus or SPI data input bus |  
  | 5 | MISO | GND | Digital output SPI data output bus |  
  | 6 | IRQ | D7 (D6) | IRQ Digital output Interrupt |  
  | 7 | SI | 3.3V | Select Interface VDD → I2C Mode !! |  

## **Error**

When I click with a piezo BBQ lighter nearby, I get the following error messages:

```text
[W][i2c:070]: Received NACK on transmit of address 0x03
[W][as3935_i2c:029]: Read failed!
[W][i2c:070]: Received NACK on transmit of address 0x03
[W][as3935_i2c:029]: Read failed!
[W][i2c:070]: Received NACK on transmit of address 0x03
[W][as3935_i2c:029]: Read failed!
[W][i2c:070]: Received NACK on transmit of address 0x03
[W][as3935_i2c:029]: Read failed!
[W][i2c:070]: Received NACK on transmit of address 0x03
[W][as3935_i2c:029]: Read failed!
```

What else can I do. I shouldn't be the only one running this sensor with a NodeMCU. I am very grateful for further help.

randybb, 2020-08-16:
orry for that, I just wanted to make sure you connect it and test it properly. 
Well, I have a few of these sensors but I have trouble getting them working with esphome and on other projects. I can't blame esphome - I believe the issue is in wrongly tuned antenna... The best implementation I found is https://bitbucket.org/christandlg/as3935mi - it can tune the antenna during initialization and change the spike rejection ratio OR watchdog threshold based on the noise level. This works flawlessly, it is changing these values all the time, but I never got a storm approach warning => so probably these sensor boards are not well designed and I need to order a different one just to compare it.

So back to the esphome. I removed it from my stm32f411 board and connected it to my esp32 debug board with a bunch of other sensors and I am getting a lot of info messages:
```yaml
[I][as3935:044]: Disturber was detected - try increasing the spike rejection value!
```
Somebody on HA forum runned into the same problem, changing all default values without any success.

From the other projects I know that it is triggering IRQ and the interrupt vector sais that there is a lot of noise (in these cases it will not show you energy or distance). A few days ago there was a big thunderstorm with many lightning strikes and non of these (one board uses "as3935mi" and the other one "SparkFun AS3935 Lightning Detector Arduino Library")  triggered the alarm, so this confirms my hypothesis about the wrongly tuned antenna on these tiny boards. Unfortunately, I don't have a different type, only this small one.

randybb, 2020-08-17:
I was getting your errors only with a disconnected sensor, otherwise it acts normal.

We just had a big thunderstorm with more than 100 strikes in 8-17 km radius - my other esp32 board, where I am using AS3935MI, beeped like crazy. On ESPHome I got 0 strikes and I have never seen any lightning distance or energy. That means something is wrong with this implementation and this post https://community.home-assistant.io/t/esphome-with-as3935-franklin-lightning-sensor-try-increasing-the-spike-rejection-value/209621/4 just confirms it.

zibous, 2020-08-18:
@randybb 

I have now done different tests with different software (ESOHome, Tasmota, as3935mi ..) on different boards (NodeMCU, ESP32, Ardino mini, Wemos D1 Mini) and come to the following result:

The sensor (SDA, SDL) is correctly found on all boards during the I2C scan at address 0x03. The problem is probably with the IRQ. Even the attempt to work with a pulldown resistor on the IRQ pin was unsuccessful.

But I have no idea how I can test it, because I don't know whether the GY-AS3935 sensor works at all.

SknerusMcK, 2020-09-05:
For me increasing I2C frequency to 400kHz worked like a charm :D
[image]

glmnet, 2020-09-05:
Are you sure those are khz and not hz?

SknerusMcK, 2020-09-05:
> Are you sure those are khz and not hz?

No, this is ESPHome docs 
[image]

closed 2021-01-17 by stale[bot]

blind-oracle, 2021-03-30:
I've got the same problem. Bought 2 of them, tried SPI/I2C, either I'm getting these `Disturber was detected` floods or nothing.

Connected to Arduino in SPI mode with SparkFun library, it sees the AS board fine, sets/gets the settings (noise level etc) but there are *no* interrupts from it at all. I tried some man-made disturbers: piezo, drill which sparks like hell when stopped etc. Nothing.