esphome/issues #1414: TM1637 - time not working

SnowyTerror, 2020-08-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io (Home Assistant) Version: 0.114.0 on Raspberry Pi 4

**ESP (ESP32/ESP8266, Board/Sonoff):**

Wemos D1 Mini

**ESPHome version (latest production, beta, dev branch)**

Tested on ESPHome Beta Version: 1.15.0b3

**Affected component:**

Link: [https://beta.esphome.io/components/display/tm1637.html?highlight=tm1637](https://beta.esphome.io/components/display/tm1637.html?highlight=tm1637)

**Description of problem:**
The time function in the below link in relation to TM1637 doesn't work at all, once the sketch is uploaded via OTA as normal it would start counting up from when it has started by the minute.

**Problem-relevant YAML-configuration entries:**
```yaml
time:
  - platform: homeassistant
    id: homeassistant_time

display:
  platform: tm1637
  clk_pin: D6
  dio_pin: D5
  update_interval: 500ms
  lambda: |-
      static int i = 0;
      i++;
      if ((i % 2) == 0)
        it.strftime("%H.%M", id(homeassistant_time).now());
      else
        it.strftime("%H%M", id(homeassistant_time).now());

```

**Logs (if applicable):**

No Logs available

**Additional information and things you've tried:**
Tried different methods for the time to work by either GPS but do not seem to work, attempted multiple times to confirm that when sketch is uploaded it will start counting up by the minute rather than showing correct time.

rradar, 2020-08-15:
Are you sure your time is synced? Can you see it in the logs (please post them here)? It takes a bit (up to 1 minute) in my case till the time from home assistant is available.

I use the same code and it works exactly like it expected.

SnowyTerror, 2020-08-15:
Hi @rradar I wouldn't be so sure if they're successfully syncing, please see logs below from the Wemos that has the TM1637.

`INFO Reading configuration /config/esphome/pump_clock.yaml...
INFO Detected timezone 'GMT' with UTC offset 0 and daylight savings time from 28 March 01:00:00 to 31 October 02:00:00
INFO Starting log output from pump_clock.local using esphome API
INFO Connecting to pump_clock.local:6053 (192.168.0.16)
INFO Successfully connected to pump_clock.local
[19:19:19][I][app:105]: ESPHome version 1.15.0b3 compiled on Aug 15 2020, 05:21:18
[19:19:19][C][wifi:443]: WiFi:
[19:19:19][C][wifi:303]:   SSID: [redacted]
[19:19:19][C][wifi:304]:   IP Address: 192.168.0.16
[19:19:19][C][wifi:306]:   BSSID: [redacted]
[19:19:19][C][wifi:307]:   Hostname: 'pump_clock'
[19:19:19][C][wifi:311]:   Signal strength: -82 dB ▂▄▆█
[19:19:19][C][wifi:315]:   Channel: 11
[19:19:19][C][wifi:316]:   Subnet: 255.255.255.0
[19:19:19][C][wifi:317]:   Gateway: 192.168.0.1
[19:19:19][C][wifi:318]:   DNS1: 192.168.0.1
[19:19:19][C][wifi:319]:   DNS2: (IP unset)
[19:19:19][C][fastled:019]: FastLED light:
[19:19:19][C][fastled:020]:   Num LEDs: 41
[19:19:19][C][fastled:021]:   Max refresh rate: 400
[19:19:19][C][logger:185]: Logger:
[19:19:19][C][logger:186]:   Level: DEBUG
[19:19:19][C][logger:187]:   Log Baud Rate: 115200
[19:19:19][C][logger:188]:   Hardware UART: UART0
[19:19:19][C][light:177]: Light 'Pump Clock Lights'
[19:19:19][C][light:179]:   Default Transition Length: 1.0s
[19:19:19][C][light:180]:   Gamma Correct: 2.80
[19:19:19][C][homeassistant.time:010]: Home Assistant Time:
[19:19:19][C][homeassistant.time:011]:   Timezone: 'GMT0BST-1,M3.4.0/1,M10.5.0/2'
[19:19:20][C][display.tm1637:131]: TM1637:
[19:19:20][C][display.tm1637:132]:   INTENSITY: 7
[19:19:20][C][display.tm1637:133]:   CLK Pin: GPIO4 (Mode: OUTPUT)
[19:19:20][C][display.tm1637:134]:   DIO Pin: GPIO0 (Mode: OUTPUT)
[19:19:20][C][display.tm1637:135]:   Update Interval: 0.5s
[19:19:20][C][captive_portal:169]: Captive Portal:
[19:19:20][C][ota:029]: Over-The-Air Updates:
[19:19:20][C][ota:030]:   Address: pump_clock.local:8266
[19:19:20][C][api:095]: API Server:
[19:19:20][C][api:096]:   Address: pump_clock.local:6053
[19:20:33][I][ota:046]: Boot seems successful, resetting boot loop counter.`

SnowyTerror, 2020-08-15:
Hi, just an update in regards to this - I have tried the suggestion you provided in regards to waiting upto a minute with extending this by 10 minutes and it seems to not be updating.

randybb, 2020-08-16:
I am using a similar configuration for my clocks and no problem at all, so this is working on esp8266 and esp32 without any problems.

Do you have API configured and is your node added to HA?

SnowyTerror, 2020-08-16:
Yes I believe this is configured in the API and I can confirm this is indeed added into HA, it shows under integrations as an entity, additionally I can confirm it is interacting with HA as I have FastLED alongside it working on the same entity.

I have tried reflashing it multiple times but doesn't seem to make a difference, could there be something in HA I need to enable possibly?

glmnet, 2020-08-16:
What is all this:  `[19:19:19][C][homeassistant.time:011]: Timezone: 'GMT0BST-1,M3.4.0/1,M10.5.0/2' `

What's your real timezone?

SnowyTerror, 2020-08-16:
Hi, I discovered what the issue is, I could see that when uploading it would state that its time it'll go by is sometime and day in March 1971 so I did a bit of checking to see what the home assistant time requires and found that I could add an sensor into my configuration.yaml file, the sensor I added is the below.

`sensor:`
`  - platform: time_date`
`    display_options:`
`      - 'time'`
`      - 'date'`
`      - 'date_time'`
`      - 'date_time_utc'`
`      - 'date_time_iso'`
`      - 'time_date'`
`      - 'time_utc'`
`      - 'beat'`

(Sorry for no indentation seems Github with the code option doesn't like the "-")

The two stranges things I found with this is that on HA itself when using the entity it would output for example: 15:04 when the time is 16:04 which is likely caused by daylight savings time but ESPHome seems to correct this by setting it to 16:04 and is now updating accordingly to the correct time, the only slight issue is that the minute seems to be nearly 30 seconds behind, the last standalone system I used would have went out to the internet itself and got the correct time and matched it to the second where it would update the minute right on the dot with other devices, would there be a fix for this at all?

Could you guys that have got this working with no issues paste the time related code you would have in your configuration.yaml and or advise what timezone entry should be used under HA > Configuration > General > "Time Zone" as I would presume in my case it is logical to be "GMT".

SnowyTerror, 2020-08-16:
In regards to saying it seems to work now... it seems that this was intermittent as it's now outputting wrong timezone again.

> INFO Reading configuration /config/esphome/pump_clock.yaml...
> INFO Detected timezone 'PST' with UTC offset -8 and daylight savings time from 14 March 02:00:00 to 07 November 02:00:00
> INFO Starting log output from pump_clock.local using esphome API
> INFO Connecting to pump_clock.local:6053 (192.168.0.16)
> INFO Successfully connected to pump_clock.local
> [08:49:25][I][app:105]: ESPHome version 1.15.0b3 compiled on Aug 16 2020, 08:44:50
> [08:49:25][C][wifi:443]: WiFi:
> [08:49:25][C][wifi:303]:   SSID: [redacted]
> [08:49:25][C][wifi:304]:   IP Address: 192.168.0.16
> [08:49:25][C][wifi:306]:   BSSID: [redacted]
> [08:49:25][C][wifi:307]:   Hostname: 'pump_clock'
> [08:49:25][C][wifi:311]:   Signal strength: -82 dB ▂▄▆█
> [08:49:25][C][wifi:315]:   Channel: 6
> [08:49:25][C][wifi:316]:   Subnet: 255.255.255.0
> [08:49:25][C][wifi:317]:   Gateway: 192.168.0.1
> [08:49:25][C][wifi:318]:   DNS1: 192.168.0.1
> [08:49:25][C][wifi:319]:   DNS2: (IP unset)
> [08:49:25][C][fastled:019]: FastLED light:
> [08:49:25][C][fastled:020]:   Num LEDs: 41
> [08:49:25][C][fastled:021]:   Max refresh rate: 400
> [08:49:25][C][logger:185]: Logger:
> [08:49:25][C][logger:186]:   Level: DEBUG
> [08:49:25][C][logger:187]:   Log Baud Rate: 115200
> [08:49:25][C][logger:188]:   Hardware UART: UART0
> [08:49:25][C][light:177]: Light 'Pump Clock Lights'
> [08:49:25][C][light:179]:   Default Transition Length: 1.0s
> [08:49:25][C][light:180]:   Gamma Correct: 2.80
> [08:49:25][C][homeassistant.time:010]: Home Assistant Time:
> [08:49:25][C][homeassistant.time:011]:   **Timezone: 'PST8PDT7,M3.2.0/2,M11.1.0/2'**
> [08:49:25][C][display.tm1637:131]: TM1637:
> [08:49:25][C][display.tm1637:132]:   INTENSITY: 1
> [08:49:25][C][display.tm1637:133]:   CLK Pin: GPIO4 (Mode: OUTPUT)
> [08:49:25][C][display.tm1637:134]:   DIO Pin: GPIO0 (Mode: OUTPUT)
> [08:49:25][C][display.tm1637:135]:   Update Interval: 0.5s
> [08:49:25][C][captive_portal:169]: Captive Portal:
> [08:49:25][C][ota:029]: Over-The-Air Updates:
> [08:49:25][C][ota:030]:   Address: pump_clock.local:8266
> [08:49:25][C][api:095]: API Server:
> [08:49:25][C][api:096]:   Address: pump_clock.local:6053

SnowyTerror, 2020-08-16:
Just something I noticed when validating the code, it says the timezone was detected but my timezone is now set by the system and is correct using "Europe/London"

> INFO Reading configuration /config/esphome/pump_clock.yaml...
> INFO Detected timezone 'PST' with UTC offset -8 and daylight savings time from 14 March 02:00:00 to 07 November 02:00:00
> INFO Configuration is valid!

Additionally when uploading the code that was validated.

> INFO Reading configuration /config/esphome/pump_clock.yaml...
> INFO Detected timezone 'PST' with UTC offset -8 and daylight savings time from 14 March 02:00:00 to 07 November 02:00:00
> INFO Generating C++ source...
> INFO Compiling app...
> INFO Running:  platformio run -d /config/esphome/pump_clock
> Processing pump_clock (board: d1_mini; framework: arduino; platform: espressif8266@2.6.0)
> --------------------------------------------------------------------------------
> HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
> PACKAGES: 
>  - framework-arduinoespressif8266 3.20702.0 (2.7.2) 
>  - tool-esptool 1.413.0 (4.13) 
>  - tool-esptoolpy 1.20800.0 (2.8.0) 
>  - toolchain-xtensa 2.40802.200502 (4.8.2)
> Dependency Graph
> |-- <ESPAsyncTCP-esphome> 1.2.2
> |   |-- <ESP8266WiFi> 1.0
> |-- <ESP8266WiFi> 1.0
> |-- <ESP8266mDNS> 1.2
> |   |-- <ESP8266WiFi> 1.0
> |-- <ESPAsyncWebServer-esphome> 1.2.7
> |   |-- <ESPAsyncTCP-esphome> 1.2.2
> |   |   |-- <ESP8266WiFi> 1.0
> |   |-- <Hash> 1.0
> |   |-- <ESP8266WiFi> 1.0
> |-- <FastLED> 3.3.2
> |   |-- <SPI> 1.0
> |   |-- <EspSoftwareSerial> 6.8.5
> |-- <DNSServer> 1.1.1
> |   |-- <ESP8266WiFi> 1.0
> Compiling /data/pump_clock/.pioenvs/pump_clock/src/main.cpp.o
> src/main.cpp:178:3: warning: multi-line comment [-Wcomment]
>    //       lambda: !lambda "for( int Pixel = 0; Pixel <= it.size(); Pixel++) {\n  it[Pixel]\
>    ^
> src/main.cpp:197:3: warning: multi-line comment [-Wcomment]
>    //       lambda: !lambda "for( int Pixel = 0; Pixel <= it.size(); Pixel++) {\n  it[Pixel]\
>    ^
> Linking /data/pump_clock/.pioenvs/pump_clock/firmware.elf
> Retrieving maximum program size /data/pump_clock/.pioenvs/pump_clock/firmware.elf
> Checking size /data/pump_clock/.pioenvs/pump_clock/firmware.elf
> Building /data/pump_clock/.pioenvs/pump_clock/firmware.bin
> RAM:   [=====     ]  46.6% (used 38184 bytes from 81920 bytes)
> Flash: [====      ]  41.7% (used 435460 bytes from 1044464 bytes)
> Creating BIN file "/data/pump_clock/.pioenvs/pump_clock/firmware.bin" using "/root/.platformio/packages/framework-arduinoespressif8266/bootloaders/eboot/eboot.elf" and "/data/pump_clock/.pioenvs/pump_clock/firmware.elf"
> ========================= [SUCCESS] Took 18.17 seconds =========================
> INFO Successfully compiled program.
> INFO Resolving IP address of pump_clock.local
> INFO  -> 192.168.0.16
> INFO Uploading /data/pump_clock/.pioenvs/pump_clock/firmware.bin (439616 bytes)
> Uploading: [============================================================] 100% Done...
> 
> INFO Waiting for result...
> INFO OTA successful
> INFO Successfully uploaded program.
> INFO Starting log output from pump_clock.local using esphome API
> INFO Connecting to pump_clock.local:6053 (192.168.0.16)
> WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.0.16: timed out). Re-Trying in 1 seconds
> INFO Connecting to pump_clock.local:6053 (192.168.0.16)
> INFO Successfully connected to pump_clock.local
> [09:16:35][I][app:105]: ESPHome version 1.15.0b3 compiled on Aug 16 2020, 09:15:49
> [09:16:36][C][wifi:443]: WiFi:
> [09:16:36][C][wifi:303]:   SSID: [redacted]
> [09:16:36][C][wifi:304]:   IP Address: 192.168.0.16
> [09:16:36][C][wifi:306]:   BSSID: [redacted]
> [09:16:36][C][wifi:307]:   Hostname: 'pump_clock'
> [09:16:36][C][wifi:311]:   Signal strength: -81 dB ▂▄▆█
> [09:16:36][C][wifi:315]:   Channel: 6
> [09:16:36][C][wifi:316]:   Subnet: 255.255.255.0
> [09:16:36][C][wifi:317]:   Gateway: 192.168.0.1
> [09:16:36][C][wifi:318]:   DNS1: 192.168.0.1
> [09:16:36][C][wifi:319]:   DNS2: (IP unset)
> [09:16:36][C][fastled:019]: FastLED light:
> [09:16:36][C][fastled:020]:   Num LEDs: 41
> [09:16:36][C][fastled:021]:   Max refresh rate: 400
> [09:16:36][C][logger:185]: Logger:
> [09:16:36][C][logger:186]:   Level: DEBUG
> [09:16:36][C][logger:187]:   Log Baud Rate: 115200
> [09:16:36][C][logger:188]:   Hardware UART: UART0
> [09:16:36][C][light:177]: Light 'Pump Clock Lights'
> [09:16:36][C][light:179]:   Default Transition Length: 1.0s
> [09:16:36][C][light:180]:   Gamma Correct: 2.80
> [09:16:36][C][homeassistant.time:010]: Home Assistant Time:
> [09:16:36][C][homeassistant.time:011]:   Timezone: 'PST8PDT7,M3.2.0/2,M11.1.0/2'
> [09:16:36][C][display.tm1637:131]: TM1637:
> [09:16:36][C][display.tm1637:132]:   INTENSITY: 1
> [09:16:36][C][display.tm1637:133]:   CLK Pin: GPIO4 (Mode: OUTPUT)
> [09:16:36][C][display.tm1637:134]:   DIO Pin: GPIO0 (Mode: OUTPUT)
> [09:16:36][C][display.tm1637:135]:   Update Interval: 1.0s
> [09:16:36][C][captive_portal:169]: Captive Portal:
> [09:16:36][C][ota:029]: Over-The-Air Updates:
> [09:16:36][C][ota:030]:   Address: pump_clock.local:8266
> [09:16:36][C][api:095]: API Server:
> [09:16:36][C][api:096]:   Address: pump_clock.local:6053
> [09:16:43][D][api.connection:604]: Client 'Home Assistant 0.114.1 (192.168.0.55)' connected successfully!
> [09:16:43][D][time:029]: Synchronized time: Wed Dec 31 16:00:25 1969

To also demonstrate the time that HA sees here is a below picture showing the time on HA and the time on my machine.

[image]

SnowyTerror, 2020-08-20:
I'm guessing this has left people at a dead end for now.

glmnet, 2020-08-20:
Did you try manually setting timezone in esphome https://esphome.io/components/time.html

SnowyTerror, 2020-08-20:
Hi yes this was tried and hasn't worked.

I gave it another go today, switched from ESPHome beta to ESPHome dev and issue seems to now be resolved, there may be something in the beta build that wasn't working I'm guessing?... guess we'll never find out.

closed 2020-08-20 by SnowyTerror

glmnet, 2020-08-20:
Actually beta and dev are not much different right now