esphome/issues #1417: Manual IP assignment stops time component

RubyBailey, 2020-08-16:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Windows 10, VMWare Workstation 15 Player running Ubuntu 18.04.4 LTS
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 Node MCU V2
**ESPHome version (latest production, beta, dev branch)**

v1.14.5
**Affected component:**

https://esphome.io/components/time.html
https://esphome.io/components/wifi.html

**Description of problem:**
When adding manual_ip time component stops working.
Original code was written without manual ip and functioned flawlessly. Added manual ip and noted integration reset was no longer working. Tested by adding in interval component and integration reset functioned again, yet not on minute boundaries as with time component. Commented out interval and manual ip and time + integration reset worked again.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: watermeter
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "removed_ssid"
  password: "removed_password"
# remove comments markers from manual ip lines below and integration reset / time component stops working
#  manual_ip:
#    # IP of ESP
#    static_ip: 192.168.68.186
#    # IP of router
#    gateway: 192.168.68.1
#    # Subnet of network
#    subnet: 255.255.255.0  

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Watermeter Fallback Hotspot"
    password: "removed_password"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API - disabled by comment out 
# api:

# Over the air update enabled
ota:

mqtt:
  broker: 192.168.68.190
  # Last will topics and message
  birth_message:
    topic: watermeter/LWT
    payload: Online
  will_message:
    topic: watermeter/LWT
    payload: Offline

# Configure sensor
sensor:
  - platform: pulse_counter
    id: water_pulse
    pin: GPIO14
    name: "Pulse Counter"
    update_interval: 1s

  - platform: integration
    id: water_total_minute
    name: "Water Total Minute"
    sensor: water_pulse
    time_unit: min
    filters:
      - lambda: return x / 439.0;    

time:
  - platform: sntp
    # ...
    on_time:
      # Every minute
      - cron: '0 /1 * * * *'
        then:
          - sensor.integration.reset: water_total_minute    

# Remove comment markers to test that integration reset is still working when manual ip assigned
#interval:
# One  minute
#  - interval: 1min
#    then:
#      - sensor.integration.reset: water_total_minute

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

martgras, 2020-08-17:
You probably have to add a DNS server to your config otherwise the sntp component cannot find the ntp servers

RubyBailey, 2020-08-17:
You are correct. I added the DNS server in the wifi component and now the time component works again. Thanks for the quick response. 
Did I miss something in the documentation? I was not able to find that requirement. Could I suggest a note in the sntp section of the time component. .
Thanks again, much appreciated.

martgras, 2020-08-17:
I didn't see it in the docs either.
However since the default for the ntp server is 0.pool.ntp.org you need name resolution to work. The alternative is setting
The same applies to all settings where you use a server name instead of an IP address (like MQTT server)
Maybe it wouldn't hurt to add that piece of information to the wifi documentation part.
Just in case you are not really familiar with DNS. It's purpose is translating names to IP addresses.

RubyBailey, 2020-08-17:
I am familiar with DNS. I thought that name resolution happened at the router level, WAN to LAN. Similar to my computer, when it requests a site by name, the router is responsible for providing DNS services. It's in the router where, if required, the primary and secondary DNS ip addresses are entered. I have not had to set them at the device level before. 
Since DNS is listed as optional in wifi, some documentation in wifi (if this effects other components) or time (if only this one is effected) would be helpful. Thanks.

randybb, 2020-08-18:
You are not familiar with DNS :) Your device needs know IP address for every IP communication (https://en.wikipedia.org/wiki/IPv4#Packet_structure). If you have set domain name (sntp component defaults to 0.pool.ntp.org, 1.pool.ntp.org and 2.pool.ntp.org), then it needs to resolve IP address from DNS.
Normally you don't need to set DNS on esphome, but in this case it is needed.

martgras, 2020-08-18:
You probably never noticed that your PC needs a DNS Server as well  
````
C:\Users\fred>ipconfig /all
..
Ethernet adapter Ethernet:
...
   Default Gateway . . . . . . . . . : 192.168.1.1
   DHCP Server . . . . . . . . . . . : 192.168.1.1
   DNS Servers . . . . . . . . . . . : 192.168.1.1  <---
...
````

Anyways I noticed that this is not the first time this issue is reported and just submitted https://github.com/esphome/esphome-docs/pull/735

closed 2020-08-19 by glmnet

RubyBailey, 2020-08-29:
I've been away and unable to respond. Wanted to let you know that I appreciate the quick response and feedback. Thanks again everyone.