esphome/issues #1419: ERROR Error binary size: Error: ESP has been flashed with wrong flash size. Please choose the correct 'board' option

Sennevds, 2020-08-19:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io and pip(on windows 10

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP8266 board sparkfun thing

**ESPHome version (latest production, beta, dev branch)**
Latest on hass.io(1.14.4) and latest on pip (1.14.5)

**Affected component:**
OTA

**Description of problem:**
Can't upload new version over OTA to the thing board. It worked for the pip version 1.14.3 but after updating to the latest version I get following message:
`ERROR Error binary size: Error: ESP has been flashed with wrong flash size. Please choose the correct 'board' option`

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: temperature
  platform: ESP8266
  board: thing

wifi:
  ssid: "*****"
  password: "*****"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Temperature Fallback Hotspot"
    password: "*****"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "****"

ota:
  password: "****"

dallas:
  - pin: GPIO14

sensor:
  - platform: dallas    
    address: 0x7200000A987F8E28
    name: "Temperature outside"
  - platform: dallas
    address: 0x8D00000AEEF94728
    name: "Temperature pool"

```

**Logs (if applicable):**
```
INFO Reading configuration /config/esphome/temperature.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/temperature
Processing temperature (board: thing; framework: arduino; platform: espressif8266@2.2.3)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 512KB Flash
PACKAGES: 
 - framework-arduinoespressif8266 2.20502.0 (2.5.2) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - toolchain-xtensa 1.40802.2 (4.8.2)
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.6
|   |-- <ESPAsyncTCP-esphome> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
Compiling /data/temperature/.pioenvs/temperature/src/main.cpp.o
Linking /data/temperature/.pioenvs/temperature/firmware.elf
Retrieving maximum program size /data/temperature/.pioenvs/temperature/firmware.elf
Checking size /data/temperature/.pioenvs/temperature/firmware.elf
Building /data/temperature/.pioenvs/temperature/firmware.bin
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=====     ]  45.5% (used 37288 bytes from 81920 bytes)
Flash: [========  ]  81.3% (used 406244 bytes from 499696 bytes)
Creating BIN file "/data/temperature/.pioenvs/temperature/firmware.bin" using "/data/temperature/.pioenvs/temperature/firmware.elf"
========================= [SUCCESS] Took 16.43 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of temperature.local
INFO  -> 192.168.0.39
INFO Uploading /data/temperature/.pioenvs/temperature/firmware.bin (410400 bytes)
ERROR Error binary size: Error: ESP has been flashed with wrong flash size. Please choose the correct 'board' option (esp01_1m always works) and then flash over USB.
```

res1mp7q, 2020-08-25:
Getting the same with the ESP8266 Thing

YAML
`esphome:
  name: sprinkler_contr
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: 'espota'
  password: 'iOtmPi0j8TPx6rV9EBZ57qE'
  manual_ip:
    static_ip: X
    gateway: X
    subnet: 255.255.255.192

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Sprinkler Contr Fallback Hotspot"
    password: "X"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "X"

ota:
  password: "X"

time:
  - platform: sntp
    id: sntp_time
    servers: X
    `
Logs
`INFO Reading configuration /config/sprinkler_contr.yaml...
INFO Detected timezone 'PST' with UTC offset -8 and daylight savings time from 14 March 02:00:00 to 07 November 02:00:00
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/sprinkler_contr
Processing sprinkler_contr (board: esp01_1m; framework: arduino; platform: espressif8266@2.2.3)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 2.20502.0 (2.5.2) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - toolchain-xtensa 1.40802.0 (4.8.2)
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.6
|   |-- <ESPAsyncTCP-esphome> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|-- <AsyncMqttClient-esphome> 0.8.4
|   |-- <ESPAsyncTCP-esphome> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
Compiling .pioenvs/sprinkler_contr/src/main.cpp.o
Linking .pioenvs/sprinkler_contr/firmware.elf
Retrieving maximum program size .pioenvs/sprinkler_contr/firmware.elf
Checking size .pioenvs/sprinkler_contr/firmware.elf
Building .pioenvs/sprinkler_contr/firmware.bin
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=====     ]  46.0% (used 37660 bytes from 81920 bytes)
Flash: [====      ]  41.5% (used 424541 bytes from 1023984 bytes)
Creating BIN file ".pioenvs/sprinkler_contr/firmware.bin" using ".pioenvs/sprinkler_contr/firmware.elf"
========================= [SUCCESS] Took 7.94 seconds =========================
INFO Successfully compiled program.
INFO Connecting to XX.XX.XX.XX
INFO Uploading /config/sprinkler_contr/.pioenvs/sprinkler_contr/firmware.bin (428688 bytes)
ERROR Error binary size: Error: ESP has been flashed with wrong flash size. Please choose the correct 'board' option (esp01_1m always works) and then flash over USB.
`

res1mp7q, 2020-10-15:
Any updates? 1.15.2 is currently throwing the same error:

`Creating BIN file ".pioenvs/gr/firmware.bin" using "/root/.platformio/packages/framework-arduinoespressif8266/bootloaders/eboot/eboot.elf" and ".pioenvs/gr/firmware.elf"
========================= [SUCCESS] Took 92.80 seconds =========================
INFO Successfully compiled program.
INFO Connecting to X
INFO Uploading /config/gr/.pioenvs/garage_door/firmware.bin (479152 bytes)
ERROR Error binary size: Error: ESP has been flashed with wrong flash size. Please choose the correct 'board' option (esp01_1m always works) and then flash over USB.`

res1mp7q, 2020-11-08:
1.15.3 same error

ERROR Error binary size: Error: ESP has been flashed with wrong flash size. Please choose the correct 'board' option (esp01_1m always works) and then flash over USB.

ioios-io, 2020-11-23:
Same error here when trying first OTA with Shelly 1PM which I flashed with ESPHome bin via ESPTool. The device shows up and is talking with HA just fine via native API.

ioios-io, 2020-11-23:
Update! I fixed my error by adjusting the ESPTool command. I was writing 4MB and when I dropped it to 1MB I could flash it OTA. I thought that was a transfer rate rather than a board setting but I don't care, it works now!

closed 2021-04-07 by stale[bot]