esphome/issues #1420: esphome constantly resetting

richardweinberger, 2020-08-19:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

environment is pip on Linux x64_64, esphome version 1.14.4

```
Processing test1 (board: nodemcuv2; framework: arduino; platform: espressif8266@2.2.3)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 2.20502.0 (2.5.2) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.6
|   |-- <ESPAsyncTCP-esphome> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|-- <ArduinoJson-esphomelib> 5.13.3
```
**ESP (ESP32/ESP8266, Board/Sonoff):**

nodemcuv2

**ESPHome version (latest production, beta, dev branch)**

1.14.4

**Affected component:**

Core, I guess.

**Description of problem:**

esphome resets after a few minutes, always.
I've ruled out WIfi, since it happens with different APs.

After at most 1000 seconds uptime, it resets.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: test1
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "mywifi"
  password: "mypassword"

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "secret"

ota:
  password: "secret"
  
web_server:
  port: 80
  auth:
    username: "admin"
    password: "secret"

dallas:
  - pin: D0

sensor:
  - platform: dallas
    address: 0x6100000530FE3528
    name: "Test1"
  - platform: uptime
    name: "uptime"
```

**Logs (if applicable):**

```
[D][dallas.sensor:153]: 'Test1': Got Temperature=26.4�°C
[D][sensor:092]: 'Test1': Sending state 26.37500 �°C with 1 decimals of accuracy
[D][sensor:092]: 'uptime': Sending state 891.13599 s with 0 decimals of accuracy
[D][dallas.sensor:153]: 'Test1': Got Temperature=26.4�°C
[D][sensor:092]: 'Test1': Sending state 26.43750 �°C with 1 decimals of accuracy
[E][api:084]: No client connected to API. Rebooting...
[I][app:127]: Forcing a reboot...
[W][wifi_esp8266:394]: Event: Disconnected ssid='mywifi' bssid=xxx reason='Association Leave'

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)                

load 0x4010f000, len 1384, room 16
tail 8                   
chksum 0x2d                            
csum 0x2d                                            
v8b899c12
~ld       
```

How can i further debug the issue?
I've ruled out WIFI already.

glmnet, 2020-08-19:
This is not a bug it is well documented in the api component that it will reboot if home assistant is not connecting to the node. 
And that's exactly what the logs are telling you there.

closed 2020-08-19 by glmnet

richardweinberger, 2020-08-19:
> This is not a bug it is well documented in the api component that it will reboot if home assistant is not connecting to the node.
> And that's exactly what the logs are telling you there.

Oh, wow. Thanks for your quick answer!
I've completely misinterpreted "No client connected to API. Rebooting...". I though it tries to log the next messages to the API client because it is rebooting due to a fatal exception.