esphome/issues #1422: Over the Air update esp32 [Errno 111] Connection refused

ImmanuelT8, 2020-08-19:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**ESPHome version (latest production, beta, dev branch)**

**Affected component:**

 https://esphome.io/index.html

**Description of problem:**
Hello,

i had install home assistent and install here ESP Home. Then i use ESP32 an install the Over the air example code from the arduino IDE an name the port "pumpe_groom", the connection from Arduino IDE to ESP via Over the Air with "pumpe_groom" works.

Then i creat "pumpe_groom" on the ESP Home Over the air, i get this log:

======================== [SUCCESS] Took 10.74 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of pumpe_groom.local
INFO -> 192.168.8.106
ERROR Connecting to pumpe_groom.local:3232 failed: [Errno 111] Connection refused

The program is

esphome:
name: pumpe_groom
platform: ESP32
board: nodemcu-32s

wifi:
ssid: "WLAN1-XXXX
password: "XXXX!"

Enable fallback hotspot (captive portal) in case wifi connection fails
ap:
ssid: "Esp32 Dallas Temp Wand1"
password: "XXXXX"

captive_portal:

Enable logging
logger:

Enable Home Assistant API
api:

ota:
safe_mode: True
id: ESP1

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: pumpe_groom
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "XXXX"
  password: "XXXX"
  
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32 Dallas Temp Wand1"
    password: "XXXXX"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  safe_mode: True
  id: ESP1
```

**Logs (if applicable):**

```
INFO Reading configuration /config/esphome/esp32_dallas_temp_wand1.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/pumpe_groom
Processing pumpe_groom (board: nodemcu-32s; framework: arduino; platform: espressif32@1.12.1)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif32 3.10004.200129 (1.0.4) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
Dependency Graph
|-- <AsyncTCP-esphome> 1.1.1
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <FS> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.6
|   |-- <AsyncTCP-esphome> 1.1.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <Update> 1.0
|-- <WiFi> 1.0
Compiling /data/pumpe_groom/.pioenvs/pumpe_groom/src/main.cpp.o
Linking /data/pumpe_groom/.pioenvs/pumpe_groom/firmware.elf
Building /data/pumpe_groom/.pioenvs/pumpe_groom/firmware.bin
Retrieving maximum program size /data/pumpe_groom/.pioenvs/pumpe_groom/firmware.elf
Checking size /data/pumpe_groom/.pioenvs/pumpe_groom/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  12.9% (used 42320 bytes from 327680 bytes)
Flash: [=====     ]  53.1% (used 870254 bytes from 1638400 bytes)
========================= [SUCCESS] Took 10.91 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of pumpe_groom.local
INFO  -> 192.168.8.106
ERROR Connecting to pumpe_groom.local:3232 failed: [Errno 111] Connection refused
```

**Additional information and things you've tried:**

closed 2020-08-20 by ImmanuelT8

edtsoi-ultra, 2021-01-22:
i have the same issue, my board is esp8266

janekjelonek, 2021-02-05:
can you sahre with us what the solution was?

ImmanuelT8, 2021-02-05:
> can you share with us what the solution was?

Yes. With the new ESP Home Version the Sensor works.