esphome/issues #1423: esphome wizard for esp8266 (esp12e) board yields a build failure

joekhoobyar, 2020-08-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Command line, via pip3

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP (ESP8266, Board/esp12f)

**ESPHome version (latest production, beta, dev branch)**

ESPHome version 1.14.5

**Affected component:**

Getting started from the Command Line.

**Description of problem:**

ESPHome is apparently now using Arduino 2.5.2 to build for this board, but it doesn't work.
I was able to get the build to work by switching to Arduino 2.4.2.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
sphome:
  name: dummy1
  platform: ESP8266
  board: esp12e
  # UNCOMMENT this line for the fix
  # arduino_version: 2.4.2

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
$ esphome dummy.yaml run
INFO Reading configuration dummy.yaml...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
INFO Running:  platformio run -d dummy1
***********************************************************************************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for embedded development > https://platformio.org/platformio-ide
***********************************************************************************************************************************************

Processing dummy1 (board: esp12e; framework: arduino; platform: espressif8266@2.2.3)
-----------------------------------------------------------------------------------------------------------------------------------------------
PlatformManager: Installing espressif8266 @ 2.2.3
espressif8266 @ 2.2.3 has been successfully installed!
The platform 'espressif8266@2.2.3' has been successfully installed!
The rest of packages will be installed automatically depending on your build environment.
PackageManager: Installing toolchain-xtensa @ <3
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
toolchain-xtensa @ 2.100100.200706 has been successfully installed!
PackageManager: Installing framework-arduinoespressif8266 @ ~2.20502.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
framework-arduinoespressif8266 @ 2.20502.0 has been successfully installed!
PackageManager: Installing tool-esptool @ <2
tool-esptool @ 1.413.0 has been successfully installed!
PackageManager: Installing tool-esptoolpy @ ~1.20600.0
tool-esptoolpy @ 1.20600.0 has been successfully installed!
CorePackageManager: Installing tool-scons @ ~3.30102.0
Downloading  [####################################]  100%          
[... 83 lines cut ...]
src/esphome/components/api/util.h:13:7: note: forward declaration of 'class esphome::api::APIServer'
   13 | class APIServer;
      |       ^~~~~~~~~
In file included from src/esphome/components/api/api_server.h:12,
                 from src/esphome/components/api/api_connection.h:7,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/components/api/homeassistant_service.h: In member function 'void esphome::api::HomeAssistantServiceCallAction<Ts>::play(Ts ...)':
src/esphome/components/api/homeassistant_service.h:54:18: warning: invalid use of incomplete type 'class esphome::api::APIServer'
   54 |     this->parent_->send_homeassistant_service_call(resp);
      |                  ^~
In file included from src/esphome/components/api/api_server.h:9,
                 from src/esphome/components/api/api_connection.h:7,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/components/api/util.h:13:7: note: forward declaration of 'class esphome::api::APIServer'
   13 | class APIServer;
      |       ^~~~~~~~~
*** [.pioenvs/dummy1/src/esphome/components/api/api_pb2_service.cpp.o] Error 1
*** [.pioenvs/dummy1/src/esphome/components/api/api_connection.cpp.o] Error 1
*** [.pioenvs/dummy1/src/esphome/components/api/api_server.cpp.o] Error 1
*** [.pioenvs/dummy1/src/esphome/components/api/api_pb2.cpp.o] Error 1
========================================================= [FAILED] Took 36.79 seconds =========================================================

```

**Additional information and things you've tried:**

tairabiteru, 2020-08-21:
Exact same issue for me.

NathanSalapat, 2020-08-23:
So I'm not alone, same issue here. Tried uploading code over WiFi and with a USB connection, although I don't think that makes any difference.

closed 2021-01-05 by stale[bot]