esphome/issues #1425: No connection to esp home after adding dallas sensor ds18b20 #14279

ImmanuelT8, 2020-08-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 Dev Modul, dallas sensor ds18b20, ESP Home on homeassi, flashed over the air and also on USB ESP Home Flasher

**ESPHome version (latest production, beta, dev branch)**

Last version on home assi plugin

**Affected component:**

https://esphome.io/components/sensor/dallas.html

**Description of problem:**

After i add the code for dallas, first the :

dallas:
 - pin: GPIO4

to get the adresse, the esp is no more reachable.

Here is the full code from esp home ( i use to insert it via "over the air" and also via ESP Home Flasher). Maybe to less memory on esp for the dallas lib?

**Problem-relevant YAML-configuration entries:**
```
esphome:
  name: wohnzimmer
  platform: ESP32
  board: esp32dev

wifi:
  ssid: "XXX"
  password: "XXX"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Wohnzimmer Fallback Hotspot"
    password: "DpyxMAFDvHIr"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

dallas:
  - pin: GPIO4

# Individual sensors
sensor:
  - platform: dallas
    address: 0x281C5E7E121901B7
    name: "Living Room Temperature"
    

  
switch:
  - platform: gpio
    name: "relai"
    pin: 16
    restore_mode: RESTORE_DEFAULT_OFF

ota:  
```

**Logs (if applicable):**
[... 534 lines cut ...]
[10:08:07]A2      : 0x3ffbf33c  A3      : 0xb33fffff  A4      : 0x00060023  A5      : 0x003fffff  
[10:08:07]A6      : 0x00060021  A7      : 0x00000000  A8      : 0x0000abab  A9      : 0x0000cdcd  
[10:08:07]A10     : 0x0000abab  A11     : 0x00060023  A12     : 0x00060021  A13     : 0x3ffbbff0  
[10:08:07]A14     : 0x00000003  A15     : 0x00060f23  SAR     : 0x0000001f  EXCCAUSE: 0x00000006  
[10:08:07]EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
[10:08:07]
[10:08:07]Backtrace: 0x400895ee:0x3ffbbf40 0x40089d4a:0x3ffbbf70 0x4008b964:0x3ffbbf90 0x4008b91a:0x3ffbbfb0 0x4008c66f:0x00000000
[10:08:07]
[10:08:07]Rebooting...
```

**Additional information and things you've tried:**

I add the  address: 0x281C5E7E121901B7 also (see code) there are 3 sensors on it, but if i add it on this way:

```
sensor:
  - platform: dallas
    address: 0xA40000031F055028
    name: "Temperature #1"
  - platform: dallas
    address: 0xDD0000031EFB0428
    name: "Temperature #2"
```
etc i cand update?...

ImmanuelT8, 2020-09-04:
> I'm having this same issue as well with only 1 sensor. My logs are at https://pastebin.com/j5fA6r2L but they show pretty much the same thing. As soon as I disable the Dallas component the device will boot and work normally. Hoping there is a solution.
> 
> Edit: It seems to be the same issue as #1276
> [#1276 (comment)](https://github.com/esphome/issues/issues/1276#issuecomment-648538294) has the fix

Did you found the bug?

glmnet, 2020-09-16:
@ImmanuelT8  make sure you are running latest ESPHome version v1.15.1, then try again.

closed 2020-09-16 by ImmanuelT8