esphome/issues #1426: Home assistant log spam with deep sleep devices

WoLpH, 2020-08-20:
Not entirely sure if this classifies as a bug or if it should be a feature request, but here goes anyway :)

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Manual installation using `pip` in a `virtualenv` on Python 3.8 / Ubuntu 20.04

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32

**ESPHome version (latest production, beta, dev branch)**
Latest production, esphome 1.14.5

**Affected component:**

deep sleep: https://esphome.io/components/deep_sleep.html

**Description of problem:**

When deep sleep is enabled the devices are obviously not available all the time anymore. The problem is that the home assistant component doensn't seem to be aware of that and keeps logging connection errors.

And what's worse, since it doesn't look at the `run_duration` it can probably miss the update window easily as it's reconnecting after 60 seconds (which is much more than the 10 seconds run duration I'm currently using).

**Problem-relevant YAML-configuration entries:**
```yaml
deep_sleep:
  run_duration: 10s
  sleep_duration: 15min
```

**Logs (if applicable):**
```
2020-08-20 15:42:28 INFO (MainThread) [homeassistant.components.esphome] Can't connect to ESPHome API for 1.2.3.4: Error connecting to ('1.2.3.4', 6053): [Errno 113] Connect call failed ('1.2.3.4', 6053)
2020-08-20 15:42:28 INFO (MainThread) [homeassistant.components.esphome] Trying to reconnect to 1.2.3.4 in 60 seconds
2020-08-20 15:43:18 INFO (MainThread) [homeassistant.components.esphome] Can't connect to ESPHome API for 2.3.4.5: Error connecting to ('2.3.4.5', 6053): [Errno 113] Connect call failed ('2.3.4.5', 6053)
2020-08-20 15:43:18 INFO (MainThread) [homeassistant.components.esphome] Trying to reconnect to 2.3.4.5 in 60 seconds
2020-08-20 15:43:31 INFO (MainThread) [homeassistant.components.esphome] Can't connect to ESPHome API for 1.2.3.4: Error connecting to ('1.2.3.4', 6053): [Errno 113] Connect call failed ('1.2.3.4', 6053)
2020-08-20 15:43:31 INFO (MainThread) [homeassistant.components.esphome] Trying to reconnect to 1.2.3.4 in 60 seconds
```

**Additional information and things you've tried:**

glmnet, 2020-08-20:
Yeah. I suffer this too. It's just info though. I believe a log message saying it will keep trying and a message every hour of how many times it tried will be ok.

WoLpH, 2020-08-20:
I'm also wondering about the reconnect interval though. I'm wondering if it captures all data or if it needs to be lucky to sync up.

The varying intervals between updates suggest that it's often not capturing the results. In which case MQTT might be a better alternative since that's a push system. I expected esphome to use push as well but it looks like it's using a pull system
[image]

glmnet, 2020-08-20:
This thread is about log spam, for connection retry interval, I really think that won't be changed without a good reason. That is a well known issue and yes the workaround until api connection origination is reversed (node opening connection to hass) will remain this way.

WoLpH, 2020-08-20:
Both are valid issues if you ask me, and I reported them both in the initial thread.

In any case, I guess deep sleep should be combined with mqtt instead of api in that case

closed 2021-01-02 by stale[bot]