esphome/issues #1427: Wizard claiming that a board that exist does not exist.

Barsonax, 2020-08-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32
**ESPHome version (latest production, beta, dev branch)**

v1.14.5
**Affected component:**

Wizard

**Description of problem:**

The wizard is wrongly claiming that the `az-delivery-devkit-v4` board does not exist.

Links:
https://docs.platformio.org/en/latest/platforms/espressif32.html#az-delivery
https://docs.platformio.org/en/latest/boards/espressif32/az-delivery-devkit-v4.html#board-espressif32-az-delivery-devkit-v4

Console output:
```
 Next, I need to know what board you're using.
Please go to http://docs.platformio.org/en/latest/platforms/espressif32.html#boards and choose a board.
(Type esp01_1m for Sonoff devices)

For example "nodemcu-32s".
Options: alksesp32, bpi-bit, d-duino-32, esp-wrover-kit, esp32-devkitlipo, esp32-evb, esp32-gateway, esp32-poe-iso, esp32-poe, esp32-pro, esp320, esp32cam, esp32dev, esp32doit-devkit-v1, esp32thing, esp32vn-iot-uno, espea32, espectro32, espino32, featheresp32, firebeetle32, fm-devkit, frogboard, heltec_wifi_kit_32, heltec_wifi_lora_32, heltec_wifi_lora_32_V2, heltec_wireless_stick, hornbill32dev, hornbill32minima, intorobot, iotaap_magnolia, iotbusio, iotbusproteus, lolin32, lolin_d32, lolin_d32_pro, lopy, lopy4, m5stack-core-esp32, m5stack-fire, m5stack-grey, m5stick-c, magicbit, mhetesp32devkit, mhetesp32minikit, microduino-core-esp32, nano32, nina_w10, node32s, nodemcu-32s, odroid_esp32, onehorse32dev, oroca_edubot, pico32, pocket_32, pycom_gpy, quantum, sparkfun_lora_gateway_1-channel, tinypico, ttgo-lora32-v1, ttgo-t-beam, ttgo-t-watch, ttgo-t1, turta_iot_node, vintlabs-devkit-v1, wemos_d1_mini32, wemosbat, wesp32, widora-air, xinabox_cw02
(board): az-delivery-devkit-v4
Sorry, I don't think the board "{}" exists.
```

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

rradar, 2020-11-14:
Can you reproduce this behaviour with esphome 1.15.3 @Barsonax

Almighty-Alpaca, 2021-03-14:
Still relevant

Yo1L, 2021-03-23:
Same here...
I try with esphome:latest and esphome:dev docker images...

RenaudGiot, 2021-03-23:
Same issue.
Latest production version installed this morning, and wizard claiming heltec_wifi_lora_32_V2 does not exist.
This board is listed as a valid option.

eperdeme, 2021-05-11:
Same here. az-delivery-devkit-v4

ssieb, 2021-06-24:
The board names have to be added to esphome.

Zeik0s, 2022-05-30:
Same here for heltec_wifi_lora_32_V2. But it's in the code for the dev branch @ssieb 
https://github.com/esphome/esphome/blob/d2cefbf22495b3c62956a455c6bbc63daf40c96b/esphome/components/esp32/boards.py#L359

AllanJard, 2023-09-18:
Having the same issue with the _dfrobot_ ESP32 boards. What I don't understand is that they [are present in the `BOARDS` array for (`esphome.components.esp32`)](https://github.com/esphome/esphome/blob/11f6e555f9607767991a93d9c6259ef627a39bda/esphome/components/esp32/boards.py#L1264-L1271), which is wizard appears to base its list from, but it is telling me that the board doesn't exist.

randybb, 2023-09-18:
2023.8.3:
[image]
[image]

AllanJard, 2023-09-18:
Not sure what I'm doing wrong then? I'm using the CLI

```
allan@jupiter-ubuntu:~$ esphome version
Version: 2023.8.3
allan@jupiter-ubuntu:~$ esphome wizard test.yaml
Hi there!
I'm the wizard of ESPHome :)
And I'm here to help you get started with ESPHome.
In 4 steps I'm going to guide you through creating a basic configuration file for your custom ESP8266/ESP32 firmware. Yay!

============= STEP 1 =============
    _____ ____  _____  ______
   / ____/ __ \|  __ \|  ____|
  | |   | |  | | |__) | |__
  | |   | |  | |  _  /|  __|
  | |___| |__| | | \ \| |____
   \_____\____/|_|  \_\______|

===================================
First up, please choose a name for your node.
It should be a unique name that can be used to identify the device later.
For example, I like calling the node in my living room livingroom.

(name): test
Great! Your node is now called "test".

============= STEP 2 =============
      ______  _____ _____
     |  ____|/ ____|  __ \\
     | |__  | (___ | |__) |
     |  __|  \___ \|  ___/
     | |____ ____) | |
     |______|_____/|_|

===================================
Now I'd like to know what microcontroller you're using so that I can compile firmwares for it.
Are you using an ESP32 or ESP8266 platform? (Choose ESP8266 for Sonoff devices)

Please enter either ESP32 or ESP8266.
(ESP32/ESP8266): ESP32
Thanks! You've chosen ESP32 as your platform.

Next, I need to know what board you're using.
Please go to http://docs.platformio.org/en/latest/platforms/espressif32.html#boards and choose a board.
(Type esp01_1m for Sonoff devices)

For example "nodemcu-32s".
Options: adafruit_feather_esp32s2_tft, adafruit_qtpy_esp32, adafruit_qtpy_esp32c3, adafruit_qtpy_esp32s2, alksesp32, az-delivery-devkit-v4, bpi-bit, briki_abc_esp32, briki_mbc-wb_esp32, d-duino-32, esp-wrover-kit, esp32-devkitlipo, esp32-evb, esp32-gateway, esp32-poe, esp32-poe-iso, esp32-pro, esp320, esp32cam, esp32dev, esp32doit-devkit-v1, esp32doit-espduino, esp32thing, esp32thing_plus, esp32vn-iot-uno, espea32, espectro32, espino32, etboard, featheresp32, featheresp32-s2, firebeetle32, fm-devkit, frogboard, healtypi4, heltec_wifi_kit_32, heltec_wifi_kit_32_v2, heltec_wifi_lora_32, heltec_wifi_lora_32_V2, heltec_wireless_stick, heltec_wireless_stick_lite, honeylemon, hornbill32dev, hornbill32minima, imbrios-logsens-v1p1, inex_openkb, intorobot, iotaap_magnolia, iotbusio, iotbusproteus, kits-edu, labplus_mpython, lolin32, lolin32_lite, lolin_c3_mini, lolin_d32, lolin_d32_pro, lolin_s2_mini, lopy, lopy4, m5stack-atom, m5stack-core-esp32, m5stack-core2, m5stack-coreink, m5stack-fire, m5stack-grey, m5stack-timer-cam, m5stick-c, magicbit, mgbot-iotik32a, mgbot-iotik32b, mhetesp32devkit, mhetesp32minikit, microduino-core-esp32, nano32, nina_w10, node32s, nodemcu-32s, nscreen-32, odroid_esp32, onehorse32dev, oroca_edubot, pico32, piranha_esp32, pocket_32, pycom_gpy, qchip, quantum, s_odi_ultra, sensesiot_weizen, sg-o_airMon, sparkfun_lora_gateway_1-channel, tinypico, ttgo-lora32-v1, ttgo-lora32-v2, ttgo-lora32-v21, ttgo-t-beam, ttgo-t-watch, ttgo-t1, ttgo-t7-v13-mini32, ttgo-t7-v14-mini32, turta_iot_node, upesy_wroom, upesy_wrover, vintlabs-devkit-v1, wemos_d1_mini32, wemosbat, wesp32, widora-air, wifiduino32, xinabox_cw02
(board): dfrobot_firebeetle2_esp32s3
Sorry, I don't think the board "dfrobot_firebeetle2_esp32s3" exists.

(board): 
```

The list shown is incomplete compared to what is defined in the `BOARDS` object in `esphome.components.esp32` (the board is in the list in the file I've got, just not shown in that output).

randybb, 2023-09-18:
Yea, it looks like it is showing and accepting only esp32 boards (VARIANT_ESP32), strange. Just use any ESP32 board and then change "board" in the generated yaml file. But yes, this is a bug.

AllanJard, 2023-09-18:
Many thanks for looking at this! I've just checkout the latest from git and the problem doesn't exist there. It shows the full list (formatted with a board per line showing board name and description). Entering `dfrobot_firebeetle2_esp32s3` works immediately.

I'd say this issue can be closed :).

cpatulea, 2023-11-09:
```
$ esphome version
Version: 2023.10.6
$ esphome wizard heatpump.yaml
...
Next, I need to know what board you're using.
...
 - heltec_wifi_lora_32_V2 - Heltec WiFi LoRa 32 (V2)
(board): heltec_wifi_lora_32_V2
Sorry, I don't think the board "heltec_wifi_lora_32_V2" exists.
```

SpongeManiac, 2025-07-11:
Has this seriously not been fixed yet? It's been 5 years and not even an official response. Dead software.

ssieb, 2025-07-11:
What's not fixed?  But just use `esp32dev` anyway.

SpongeManiac, 2025-07-11:
Telling someone that a current feature doesn't need to be fixed because there is a half-baked workaround is just ludicrous.

Imagine if the only way to use a Bluetooth device was to pair it, un-pair, then re-pair it because there's a bug in the pairing logic.

But your github issue has been deemed irrelevant, "Whats not fixed? Just un-pair and re-pair your device."

When framed this way, it sounds a little disingenuous, right?

How about instead of closing the issue without addressing the core bug, let's leave it open and just say, "I have no clue how to fix this yet"?

closed 2025-07-11 by kbx81