esphome/issues #1434: SNTP ignores time zones

f403, 2020-08-23:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** pip

**ESP (ESP32/ESP8266, Board/Sonoff):** ESP8266

**ESPHome version (latest production, beta, dev branch)** beta, dev; latest must be affected as well

**Affected component:** sntp

**Description of problem:**

sntp ignores time zones, both `.now()` and `.utcnow()` return same values.

Time zones support seems to be disabled for ESP8266 in [sntp_component.cpp](https://github.com/esphome/esphome/blob/beta/esphome/components/sntp/sntp_component.cpp#L38). 

Deleting [line 38](https://github.com/esphome/esphome/blob/beta/esphome/components/sntp/sntp_component.cpp#L38) there solves this issue.

Does anybody now, what were the intentions there?
```c++
#ifdef ARDUINO_ARCH_ESP8266
  // let localtime/gmtime handle timezones, not sntp
  sntp_set_timezone(0);
#endif
```

**Problem-relevant YAML-configuration entries:**
```yaml
time:
  - platform: sntp
    id: sntp_time
    timezone: 'Europe/Berlin'

display:
...
    lambda: |-
      auto time = id(sntp_time).now();
      it.strftime(1, 1, id(digit_font_main), "%H:%M", time);
```

**Logs (if applicable):**

**Additional information and things you've tried:**

thedexboy, 2020-08-23:
Can confirm. Latest dev version has this issue as well.
Man, I thought I was going nuts for a second there... :)

dclobato, 2020-08-24:
Same here.  I was about to open a issue on this 👍

allan-reynolds, 2020-08-25:
Same here, I have been going nuts.  I had just done a big rehash of my config for some of my esp8266 devices and spent several days convinced I had gone back to the original config.  then realised I had not updated these devices for a while came here searching before opening an issue and found this.

I am using the 1.15.b4 via Hass.io addon.

ssieb, 2020-08-26:
Likely related to https://github.com/esphome/esphome/pull/1231

dclobato, 2020-08-26:
I have changed the time source from SNTP to Home Assistant and, now, I always get "01/01/1970 00:00:00" as time when syncing. The machine running HA is with the right time (automations are running on time)

```yaml
time:
  - platform: homeassistant
    id: local_time
```

```
[17:05:25][D][api.connection:604]: Client 'Home Assistant 0.114.1 (10.0.1.4)' connected successfully!
[17:05:25][D][time:029]: Synchronized time: Wed Dec 31 21:00:14 1969
```

I believe the problem is not on the SNTP component, but somewhere on the time component, as @ssieb pointed above.

dclobato, 2020-08-27:
Just to report that, on version 1.15.0b4, the HA sync issue is gone, but the SNTP sync issue is still there

HA:
```
[23:21:54][C][homeassistant.time:010]: Home Assistant Time:
[23:21:54][C][homeassistant.time:011]:   Timezone: 'TZ3'
[23:21:57][D][time:040]: Synchronized time: Wed Aug 26 23:21:57 2020
[23:22:07][D][sync-test:321]: local time: 23:22:06
[23:22:07][D][sync-test:322]: utc time: 02:22:06
```

SNTP:
```
[23:23:48][C][sntp:043]: SNTP Time:
[23:23:48][C][sntp:044]:   Server 1: '0.br.pool.ntp.org'
[23:23:48][C][sntp:045]:   Server 2: '1.br.pool.ntp.org'
[23:23:48][C][sntp:046]:   Server 3: '2.br.pool.ntp.org'
[23:23:48][C][sntp:047]:   Timezone: 'TZ3'
[23:24:05][D][sync-test:326]: local time: 02:24:05
[23:24:05][D][sync-test:327]: utc time: 02:24:05
```

closed 2020-09-01 by glmnet