esphome/issues #1435: Node IP renewals cause "unavailable" in HASS - Discovery stores IPs instead of host names

CarlosGS, 2020-08-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HASSOS v0.114.3 running in VirtualBox

Network Wi-Fi router: ZXHN H298A

**ESP (ESP32/ESP8266, Board/Sonoff):**

(irrelevant) ESP8266 (Sonoff, Wemos, etc)

**ESPHome version (latest production, beta, dev branch)**

(irrelevant) beta 1.15.0b4

**Affected component:**

https://github.com/home-assistant/core/blob/dev/homeassistant/components/esphome/config_flow.py

**Description of problem:**

- Upon local IP renewals (i.e. house power cut & restore) the nodes appear as "unavailable" in Home Assistant.
- This happens because when setting up integrations, the IPs are being stored instead of the node host names.
- (irrelevant) In the ESPHome plugin the devices are still visible & connected.

**Problem-relevant YAML-configuration entries:**

**Logs (if applicable):**

**Additional information and things you've tried:**
- The problem isn't so evident until you get a Wi-Fi router with lax DHCP management.
- Deleting the nodes in HASS and manually adding them, makes them available again.
- It's also possible to set up a fixed DHCP lease for each node, but it is tedious. I don't think this is the intended behavior.
- In `core.config_entries` I see that some devices are stored by IP and others by host name. Seem to be the ones that were added manually rather than using the "new devices found" prompt.
- **To solve the issue, the integration tool could ensure nodes are stored by their host name instead of their IP.**

Related threads:
https://github.com/home-assistant/core/issues/38860
https://github.com/esphome/issues/issues/176
https://www.reddit.com/r/homeassistant/comments/a3ptqf/devices_show_in_ha_as_unavailable_when_theyre/
https://community.home-assistant.io/t/solved-some-esphome-nodes-show-unavailable-in-home-assistant-but-show-online-in-esphome-gui/125840
https://community.home-assistant.io/t/configure-esphome-in-configuration-yaml/110904
https://community.home-assistant.io/t/reproducible-configuration-for-modern-setups/170309
https://community.home-assistant.io/t/esp32-connected-but-unavailable-in-frontend/183062

glmnet, 2020-08-26:
When adding manually if you use host name or ip that is what should be stored. If you add manually by ip then you are implicit accepting that your node has a fixed ip (otherwise this should be stated in docs) 

Fixed ip is your responsibility too, either by setting it on the yaml, or using static leases on your router. 

When discovered, host name should be stored as the discovery is done via mDNS which is also used to find the IP address. 

HA team has been making changes lately on their side (also on esphome integration) regarding all this stuff so probably they made a mistake. 

Personally I use dynamic ip on the nodes but my router does a good job on keeping IP address, however I've seen an issue just two days ago after renaming a node (which should be unrelated to IP address) then I had to remove and add node again. 

**If you can come up with a consistent, reproducible action list this will help figure out the issue and fix. **

CarlosGS, 2020-08-26:
Awesome! I'll try to find a reproducible sequence and write back :)

To clarify: I'm never typing IP for the node configuration, because I don't want it to be static. Always host name.

CarlosGS, 2020-08-26:
OK, a quick simple check:
- Open the text file `/config/.storage/core.config_entries`. For every ESPHome node there is a field `"host": "XXXXX"`. If you see IPs here, and not hostnames, (p.e. `"192.168.1.123"` instead of `"mynode.local"`) then your setup is vulnerable to LAN IP changes. *This is bad unless you purposely set fixed IPs.*

A reproducible test:
1. Go to "Configuration->Integrations->ESPHome" and delete one of the working ESP devices.
2. Now go to "Configuration->Server control" and restart the server.
3. You should see a "new device discovered" prompt notification, but **first check in the "/config/.storage/core.config_entries" and verify the device is not yet present** (as it has been deleted in 1.).
4. Now add the device using the auto-detection prompt. *NOTE: If you have to manually type an IP or hostname, then you are at the wrong prompt.*
5. Once the device is added, check again the `core.config_entries` file. If you see an IP in the `"host": "XXXXX"` field for this node, then your setup is vulnerable to LAN IP changes.

Could someone please give it a try, to see if it's a commonplace bug? Thanks!

CarlosGS, 2020-08-27:
I'm might be confused by the `core.config_entries` file. It seems to be auto-generated and overwritten periodically. Devices that I added with the manual + tool and correctly had domains in the file, now show an IP there.
Maybe the hass<->node association is made in the `core.device_registry` file instead. There each node has a "connection" field specifying the ESP's MAC address.
Are nodes identified by MAC then? Why is `core.config_entries` showing some with an IP and others with a domain? :thinking: 

Side note, in case it is relevant:
-
Today I turned OFF my WiFi router and many devices became unavailable again. This has happened many times in the past months, whenever losing WiFi connection even briefly.
Having IP in mind as a likely culprit, I checked the current IP for every node and they **did** match what appears in `core.config_entries`. But still they were unavailable in HASS.

To recover I always perform the following sequence. I find delaying HASS start is essential to prevent falsely "unavailable" devices.
1. Safely turn OFF the HASS VM and the host PC.
2. Turn OFF all house power (all ESPs + router), wait 10 seconds, turn ON again.
3. Wait 5 minutes to allow router and ESPs to stabilize.
4. Turn ON computer, wait for internet connection, turn ON HASS VM.

akadna, 2020-09-04:
I can confirm, that it's the device IP address that is saved in "host" field, and not the hostname, when the device is found by auto discovery.

CarlosGS, 2020-09-20:
I think I might have been mistaken by a different issue, where powering off wifi makes it necessary to restart HASS a couple times to get all ESP devices back online.

**Experienced developers please clarify:** The devices seem to be referenced/stored internally by their MAC address. But I'm not sure - does ESPHome-HASS only support devices within the same local network?
If this is the case this issue can be closed.

CarlosGS, 2020-10-28:
Can't see this happen lately, after enforcing a 5min delayed boot of HASS (to allow time for the router to provide a stable connection and all ESPhome devices are connected to the wifi AP first).

Some times I still do need to manually restart hassos so it can catch up and show all ESPHome devices correctly. No need to re-configure them though.

Cheers! :D

closed 2020-10-28 by CarlosGS

WizBangCrash, 2021-01-14:
@CarlosGS I have a very similar issue. All my ESPHome devices are added automatically and have a dns name, but their IP address is stored in the `core.config_entries` file.
I recently changed my IP address allocation by stopping the DHCP server giving out addresses above `.200`.
I have a 24 hour lease time and the following day I lost 4 ESPHome devices (all the ones that had a previous IP address above `.200`) i.e they became `unavailable` in Home Assistant.
I had to reboot Home Assistant for it to recognise the IP address changes.

Surely Home Assistant should automatically recognise the IP address change and update its `core.config_entries` when this happens?