esphome/issues #1436: MQTT - PAYLOAD IS NOT VALID JSON DATA when value NaN

zibous, 2020-08-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip, OSX High Sierra

**ESP (ESP32/ESP8266, Board/Sonoff):**

nodemcuv2 and Wemos D1 Mini

**ESPHome version (latest production, beta, dev branch)**

ESPHome v1.15.0b3
**Affected component:**

MQTT

**Description of problem:**
PAYLOAD IS NOT VALID JSON DATA

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  device_id: "wmt"
  friendly_name: "Memos 163"
  device_name: "Test Wemos D1"
  board: d1_mini
  platform: ESP8266
  update_time: 60s
  publish_interval: 30s
  device_description: Simple testcase Wemos D1
  attribution: "Powered by ESHHome Device"
  appversion: "1.0.0"
  
esphome:
  name: $device_id
  comment: $device_description
  platform: $platform
  board: $board
  esp8266_restore_from_flash: true
  platformio_options:
    upload_speed: 115200

# MQTT Client Component
# see: https://esphome.io/components/mqtt.html?highlight=mqtt
# inlude mqtt broker and wifi settings w/o api
<<: !include common/mqttonly.yaml

logger:
  level: INFO
  baud_rate: 19200

# Sun component
# see: https://esphome.io/components/sun.html
sun:
  id: sun_sun
  latitude: !secret home_latitude
  longitude: !secret home_longitude
  

sensor:
  
  # info: set internal: true to disable mqtt publish 
  - platform: wifi_signal
    name: ${device_id} WiFi Signal
    id: wifisignal
    internal: true
    update_interval: ${update_time}
    
  # used as mqtt publisher
  - platform: uptime
    id: upti
    name: "${device_id} Uptime"
    update_interval: ${publish_interval}
    on_value:
      - mqtt.publish_json:
          topic: tele/$device_id/infodata
          payload: |-
            root["wifiip"] = id(wifiip).state;
            root["wifissid"] = id(wifissid).state;
            root["wifibssid"] = id(wifibssid).state;
            root["sun_elevation"] = id(sune).state;
            root["sun_azimuth"] = id(suna).state;
            root["sun_sunrise"] = id(sunup).state;
            root["sun_sunset"] = id(sundown).state;
            root["uptime"] = id(upti).state;
            root["timestamp"] = id(systime).state;
            root["version"] = id(appver).state;
            root["appversion"] = "${appversion}";
            root["attribution"] ="${attribution}";
      - logger.log: "MQTT Data published"
        
  
  # sun component 
  # see: https://esphome.io/components/sun.html?highlight=sun
  - platform: sun
    #name: "Sonne Elevation"
    id: sune
    type: elevation
    update_interval: 30s
    internal: true
    
  - platform: sun
    # name: Sonne Azimuth
    id: suna
    type: azimuth
    update_interval: 30s
    internal: true
    
# Text Sensor Component
# see: https://esphome.io/components/text_sensor/index.html?highlight=text_sensor
text_sensor:

  # sun component 
  # see: https://esphome.io/components/sun.html?highlight=sun
  - platform: sun
    name: "Nächster Sonnenaufgang"
    id: sunup
    type: sunrise
    internal: true
    
  - platform: sun
    name: "Nächster Sonnenuntergang"
    id: sundown
    type: sunset
    internal: true
    
  # Version Text Sensor
  # see: https://esphome.io/components/text_sensor/version.html?highlight=platform%20version
  - platform: version
    name: $device_name Version
    id: appver
    internal: true
  
  # WiFi Info Text Sensor
  # see: https://esphome.io/components/text_sensor/wifi_info.html?highlight=platform%20wifi_info
  - platform: wifi_info
    ip_address:
      name: $device_name IP
      id: wifiip
      icon: mdi:ip-network
      internal: true
    ssid:
      name: $device_name SSID
      id: wifissid
      internal: true
    bssid:
      id: wifibssid
      name: $device_name BSSID 
      internal: true
      
  # Template Text Sensor Timestamp
  # see: https://esphome.io/components/text_sensor/template.html?highlight=platform%20template
  - platform: template
    name: $device_name Timestamp
    id: systime
    internal: true
    lambda:
      char str[20];
      time_t currTime = id(sntp_time).now().timestamp;
      strftime(str, sizeof(str), "%Y-%m-%d %H:%M:%S", localtime(&currTime));
      return (std::string) str;
      
 
 

```

**Logs (if applicable):**

```
*** PAYLOAD IS NOT VALID JSON DATA *** 

Non-standard token 'NaN': enable JsonParser.Feature.ALLOW_NON_NUMERIC_NUMBERS to allow
 at [Source: java.io.StringReader@92ef26a; line: 1, column: 96]

 *** PAYLOAD IS NOT VALID JSON DATA *** 
  Illegal character ((CTRL-CHAR, code 27)): only regular white space (\r, \n, \t) is allowed between tokens
   at [Source: java.io.StringReader@7abc3134; line: 1, column: 2]

```

**Additional information and things you've tried:**
How can I prevent the MQTT Paylod error when a sensor delivers the value "NaN"?
```
Non-standard token 'NaN': enable JsonParser.Feature.ALLOW_NON_NUMERIC_NUMBERS to allow
 at [Source: java.io.StringReader@92ef26a; line: 1, column: 96]
 { "wifiip":"10.1.1.12",
   "wifissid":"wlan178",
   "wifibssid":"88:1F:A1:34:3E:82",
   "sun_elevation":NaN,
   "sun_azimuth":NaN,
   "sun_sunrise":"",
   "sun_sunset":"",
   "uptime":19.791,
   "timestamp":"1970-01-01 00:00:00",
   "version":"1.15.0b3 Aug 24 2020, 15:26:22",
   "appversion":"1.0.0",
   "attribution":"Powered by ESHHome Device"
  }
  
  ... 
  {
    "wifiip" : "10.1.1.12",
    "wifissid" : "wlan178",
    "wifibssid" : "88:1F:A1:34:3E:82",
    "sun_elevation" : 44.3801,
    "sun_azimuth" : 227.6101,
    "sun_sunrise" : "04:26:50",
    "sun_sunset" : "18:15:08",
    "uptime" : 109.783,
    "timestamp" : "2020-08-24 13:31:52",
    "version" : "1.15.0b3 Aug 24 2020, 15:26:22",
    "appversion" : "1.0.0",
    "attribution" : "Powered by ESHHome Device"
  }
```

  If mqtt is used, the log message is always published.
  How can I switch this off or adapt the payload so that it is published as a valid string?
```
 *** PAYLOAD IS NOT VALID JSON DATA *** 
  Illegal character ((CTRL-CHAR, code 27)): only regular white space (\r, \n, \t) is allowed between tokens
   at [Source: java.io.StringReader@7abc3134; line: 1, column: 2]
   
  tele/wmt/info
  [0;32m[I][ota:046]: Boot seems successful, resetting boot loop counter.[0m
```

OttoWinter, 2020-09-15:
There are two ways:

 - Check for NAN in your publish lambda: 

```cpp
if (!isnan(id(suna).state)) root["sun_azimuth"] = id(suna).state;
else root["sun_azimuth"] = // some other value indicating absence
```

 - Update the receiving code to accept NAN values (see `Non-standard token 'NaN': enable JsonParser.Feature.ALLOW_NON_NUMERIC_NUMBERS to allow`)

zibous, 2020-09-30:
@OttoWinter 
Thanks

But what I miss is the MQTT setting that not individual parameters are published, - better all parameters per device. I have a lot of devices and therefore it is an advantage if the MQTT Brocker only has to process one topic per device.

Example see: OpenMQTTGateway
https://github.com/1technophile/OpenMQTTGateway/tree/development/main
```
#ifndef jsonPublishing
#  define jsonPublishing true //comment if you don't want to use Json  publishing  (one topic for all the parameters)
#endif
```

closed 2021-02-11 by stale[bot]