esphome/issues #1438: Energy sensor cse7766 doesn't reset the daily watt consumation at midnight anymore

h0jeZvgoxFepBQ2C, 2020-08-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Native Linux install via pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff POW with ESP8266

**ESPHome version (latest production, beta, dev branch)**

latest pre release 1.15 (.b4?)

**Affected component:**

cse7766 energy sensor

**Description of problem:**
Energy sensor is not resetting anymore after update (see graph).
Usually my esp node resets at 00:00:00 and counts up again, but my last update the reset doesn't work anymore as it seems.

[image]

**Problem-relevant YAML-configuration entries:**
```yaml
platform: cse7766
update_interval: 30s
current:
  name: "${description} Current"
voltage:
  name: "${description} Voltage"
power:
  id: energy_sensor
  name: "${description} Power"
  filters:
    - sliding_window_moving_average:
        window_size: 60
        send_every: 15
```

**Logs (if applicable):**

No special log output available

**Additional information and things you've tried:**

Restarting didn't change anything.

7h30n3, 2020-08-30:
I had the same issue. 

For me it got solved by uprading from 1.15.0b3 to 1.15.0b4.
I'm using a Gosund SP112 and home assistant to sync time.

h0jeZvgoxFepBQ2C, 2020-08-31:
Can confirm, it works now on 1.15.0b4! Thanks for fixing it!

closed 2020-08-31 by h0jeZvgoxFepBQ2C