esphome/issues #1444: global variable does not survive power-cyle

pille, 2020-08-30:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

integrated with Home Assistant 0.114.4

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266/d1_mini

**ESPHome version (latest production, beta, dev branch)**

Version: 1.14.5
**Affected component:**

https://esphome.io/guides/automations.html#bonus-2-global-variables

**Description of problem:**
i'm reading out a meter using a hall sensor with ```pulse_counter```.
since the meter only gives pulses, i need to add an offset to get the correct value.
i've configured this using a global variable, that should be saved across reboots.
however - the stored value only survives a reset (triggered by button). during a complete power cycle this value gets lost.

**Problem-relevant YAML-configuration entries:**
```yaml
globals:
   - id: gas_meter_offset
     type: float
     restore_value: true
     initial_value: "100.0"

sensor:
- platform: pulse_counter
  name: "gas_meter"
  id: gas_meter
  pin: D4
  icon: "mdi:fire"
  update_interval: 10s
  unit_of_measurement: "m³"
  accuracy_decimals: 3
  count_mode:
    rising_edge: DISABLE
    falling_edge: INCREMENT
  internal_filter: 100ms
  filters:
    - lambda: |-
        id(gas_meter_offset) += x / 6 * 0.01;
        return id(gas_meter_offset);
```

**Logs (if applicable):**

there's noting about restoring global variables in the logs.

**Additional information and things you've tried:**

rradar, 2020-09-08:
> However, by default this data is stored in the “RTC memory” section of the ESP8266s. **This memory is cleared when the ESP8266 is disconnected from power. So by default the state cannot be recovered after power loss.**

So looks like you didn't set [esp8266_restore_from_flash](https://esphome.io/components/esphome.html#esp8266-restore-from-flash) which is necessary if you want to save your global across power cycles

pille, 2020-09-08:
@rradar, you're right. i've clarified this in the docs.

closed 2020-09-08 by pille