esphome/issues #1445: Cover, open_duration and close_duration

osalval, 2020-08-31:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**ESPHome version (latest production, beta, dev branch)**

**Affected component:**

**Description of problem:**
Maybe it's an improvement. I want to adjust these times from Home Assistant using the entity "input_number" to be able to vary the number of seconds of these variables and I get an error because it only supports "period of time with unit"

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
``
# Basic config
substitutions:
  name: vaalos_habita
  ip_address: <--------->
  gateway: <--------->
  subnet: <--------->

globals:
  - id: up_time
    type: float
    restore_value: true

esphome:
  name: ${name}_cover
  platform: ESP8266
  board: d1_mini

# ESPHome Config
  on_boot: 
    priority: -10
    then:
      - switch.turn_off: open_cover
      - switch.turn_off: close_cover
#
# WiFi config
wifi:
  networks:
  - ssid: "<--------->"
    password: "<--------->"
  - ssid: "<--------->"
    password: "<--------->"
  #fast_connect: true
  # Optional manual IP
  manual_ip:
    static_ip: ${ip_address}
    gateway: ${gateway}
    subnet: ${subnet}

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "<--------->
    password: "<--------->"

captive_portal:

# Log de información y errores
logger:
  level: DEBUG

# API and Home Assistant
api:
  password: "<--------->"

# Actualizaciones por OTA
ota:
  password: "<--------->"

## ------------------------------ SENSORS ------------------------------
#  Sensor Analógico para el control de la Persiana por Nivel de Luz
sensor:
  - platform: adc
    pin: A0
    name: "Luz Persiana Habitación"
    icon: "mdi:white-balance-sunny"
    unit_of_measurement: "%"
    update_interval: 1s
    accuracy_decimals: 0
    filters:
    # Arduino map = (x-in_min)*(out_max-out_min)/(in_max-in_min)+out_min;
    # map = (x - 0) * (100 - 0) / (1 - 0) + 0;
    # Light level in % = 100 - (map);
      - lambda: return 100.0 - ( x * 100.0 );
  #
  - platform: wifi_signal
    name: ${name} WiFi
    update_interval: 60s
  #
  - platform: homeassistant
    name: "Up time"
    id: up_time
    entity_id: input_number.cover_habita_set_uptime

## ------------------------------ SWITCH -------------------------------
#  Relays to control the cover
switch:
  - platform: gpio
    pin:
      number: D5
      inverted: yes
    interlock: &interlock [open_cover, close_cover]
    id: open_cover
  - platform: gpio
    pin:
      number: D6
      inverted: yes
    interlock: *interlock
    id: close_cover
  
  # Button to restart
  - platform: restart
    name: ${name} Reiniciar
    
## ------------------------------- COVER -------------------------------
cover:
  - platform: time_based
    device_class: shutter
    name: "Persiana Habitación"
    id: ${name}_cover
    open_action:
      - switch.turn_on: open_cover
    open_duration: id(up_time)s
    close_action:
      - switch.turn_on: close_cover
    close_duration: 34 s
    stop_action:
      - switch.turn_off: open_cover
      - switch.turn_off: close_cover
#
## -------------------------- BINARY SENSORS --------------------------
# Buttons to open, close and stop manually the cover
binary_sensor:
  - platform: gpio
    filters:
      - delayed_on: 10ms # for debouncing push button
    pin:
      number: D3
      mode: INPUT_PULLUP
      inverted: True
    name: "Botón SUBIDA"
    id: ${name}_UP_button
    on_press:
      then:
        - cover.open: ${name}_cover
    on_release:
      then:
        - cover.stop: ${name}_cover
  #  
  - platform: gpio
    filters:
      - delayed_on: 10ms # for debouncing push button
    pin:
      number: D4
      mode: INPUT_PULLUP
      inverted: True
    name: "Botón BAJADA"
    id: ${name}_DOWN_button
    on_press:
      then:
        - cover.close: ${name}_cover
    on_release:
      then:
        - cover.stop: ${name}_cover

  # Device Status
  - platform: status
    name: ${name} Estado

#
text_sensor:
  - platform: version
    name: ${name} Versión
  - platform: wifi_info
    ip_address:
      name: ${name} IP
    ssid:
      name: ${name} SSID
    bssid:
      name: ${name} BSSID
#
status_led:
  id: ${name}_Status_LED
  pin:
    number: D4
#

**Logs (if applicable):**
<!--
Please copy the debug log here. If possible, also connect to the ESP over USB and copy those logs into the backticks.
-->
```
PASTE DEBUG LOG HERE

There are no logs because it does not accept the configuration

**Additional information and things you've tried:**

kluszczyn, 2020-09-22:
You may try this:
```
cover:
  - platform: time_based
    device_class: shutter
    name: "Persiana Habitación"
    id: ${name}_cover
    open_action:
      - switch.turn_on: open_cover
    open_duration:
       seconds: id(up_time)
```

See for details https://esphome.io/guides/configuration-types.html?highlight=time#time

osalval, 2020-09-24:
Thank you @kluszczyn 
It keeps giving an error: **Expected float**.
If you put a number it does not give an error.

kluszczyn, 2020-09-24:
@osalval you have two variables with the same name `up_time`; 1) global var and 2) sensor under home assistant platform. Please change it and see if it helps.

osalval, 2020-10-03:
Thank you @kluszczyn
I have seen it, I have also tried not to use Globals:
`
# Basic configuration
substitutions:
  name: vaalos_habita
  ip_address: ---------------
  gateway: -----------------
  subnet: --------------------
  
esphome:
  name: ${name}_cover
  platform: ESP8266
  board: d1_mini

# ESPHome Config
  on_boot:
    priority: -10
    then:
      - switch.turn_off: open_cover
      - switch.turn_off: close_cover
#
# WiFi networks
wifi:
  networks:
  - ssid: "----------------"
    password: "--------------"
  - ssid: "--------------"
    password: "--------------"
  #fast_connect: true
  # Optional manual IP
  manual_ip:
    static_ip: ${ip_address}
    gateway: ${gateway}
    subnet: ${subnet}

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "-----------------------------"
    password: "---------------------------"

captive_portal:

# Log
logger:
  level: WARN
# API Home Assistant
api:
  password: "---------------"
# Updates by OTA
ota:
  password: "--------------"

## ------------------------------ SENSORS ------------------------------
#  Analog light level sensor
sensor:
  - platform: adc
    pin: A0
    name: "Luz Persiana Habitación"
    icon: "mdi:white-balance-sunny"
    unit_of_measurement: "%"
    update_interval: 1s
    accuracy_decimals: 0
    filters:
    # Arduino map = (x-in_min)*(out_max-out_min)/(in_max-in_min)+out_min;
    # map = (x - 0) * (100 - 0) / (1 - 0) + 0;
    # Light level in % = 100 - (map);
      - lambda: return 100.0 - ( x * 100.0 );
  #
  - platform: wifi_signal
    name: ${name} WiFi
    update_interval: 60s
  #
  - platform: homeassistant
    name: "Tiempo de Subida"
    id: ${name}_open_duration
    entity_id: input_number.persiana_habita_set_uptime

## ------------------------------ SWITCH -------------------------------
#  Relays for the control of the Living Room Cover
switch:
  - platform: gpio
    pin:
      number: D5
      inverted: yes
    interlock: &interlock [open_cover, close_cover]
    id: open_cover
  - platform: gpio
    pin:
      number: D6
      inverted: yes
    interlock: *interlock
    id: close_cover
  
  # Button to restart the device
  - platform: restart
    name: ${name} Reiniciar
#    
## -------------------------- COVER ---------------------------
cover:
  - platform: time_based
    device_class: shutter
    name: "Living Room Cover"
    id: ${name}_cover

    open_action:
      - switch.turn_on: open_cover
    open_duration: 
      seconds: ${name}_open_duration

    close_action:
      - switch.turn_on: close_cover
    close_duration: 23s

    stop_action:
      - switch.turn_off: open_cover
      - switch.turn_off: close_cover
#
## -------------------------- BINARY SENSORS --------------------------
#  Buttons for MANUAL control of the Living Room Blind
binary_sensor:
  - platform: gpio
    filters:
      - delayed_on: 10ms # for debouncing push button
    pin:
      number: D3
      mode: INPUT_PULLUP
      inverted: True
    name: "Botón SUBIDA"
    id: ${name}_UP_button
    on_press:
      then:
        - cover.open: ${name}_cover
    on_release:
      then:
        - cover.stop: ${name}_cover
  #  
  - platform: gpio
    filters:
      - delayed_on: 10ms # for debouncing push button
    pin:
      number: D4
      mode: INPUT_PULLUP
      inverted: True
    name: "Botón BAJADA"
    id: ${name}_DOWN_button
    on_press:
      then:
        - cover.close: ${name}_cover
    on_release:
      then:
        - cover.stop: ${name}_cover
  # Estado del dispositivo
  - platform: status
    name: ${name} Estado
#
text_sensor:
  - platform: version
    name: ${name} Versión
  - platform: wifi_info
    ip_address:
      name: ${name} IP
    ssid:
      name: ${name} SSID
    bssid:
      name: ${name} BSSID
#
status_led:
  id: ${name}_Status_LED
  pin:
    number: D4
#
`
It keeps giving an error: Expected float.

osalval, 2020-10-03:
The part of the code that I am trying to get to pass an "Input Number" from Homeassistant to "open_duration", is:

`  - platform: homeassistant
    name: "Tiempo de Subida"
    id: ${name}_open_duration
    entity_id: input_number.persiana_habita_set_uptime
`
And this "id", I use it for "open duration"
`
    open_action:
      - switch.turn_on: open_cover
    open_duration: 
      seconds: ${name}_open_duration
`
I think this should work, but I still think he doesn't admit anything but a number

osalval, 2020-10-03:
I open a new **Issue: Feature request**

napieraj, 2020-10-03:
closing this as it is not an issue, the open and close durations are assigned during compile-time, way before there is any connection to HA, I have proposed a way to achieve the desired behavior in your feature request [#930](https://github.com/esphome/feature-requests/issues/930)

closed 2020-10-03 by napieraj