esphome/issues #1447: 1.16.0-dev hassio ESPHome addon not starting because tzlocal v. 2.0.0 not found

bertdebondt, 2020-09-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HassOS 4.12 (home assistant 0.114.4 running on Proxmox) 

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESPHOME v. 1.14.4 => Dev 1.16 failed to start

**ESPHome version (latest production, beta, dev branch)**

1.16-dev (tried to upgrade from 1.14 to 1.16-dev

**Affected component:**
HA ESPHome addon

**Description of problem:**
Was using dev (maybe previous dev) for a couple of day without any problem, on reboot ESPHOME Hassio addon did not start anymore. See log below telling requirement tzlocal=2.0.0 was not met. (looks like only tzlocal=2.0.1 is installed.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
[services.d] starting services
[services.d] done.
[08:45:38] INFO: Waiting for dashboard to come up...
[08:45:38] INFO: Starting ESPHome dashboard...
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 567, in _build_master
    ws.require(__requires__)
  File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 884, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 775, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.ContextualVersionConflict: (tzlocal 2.1 (/usr/local/lib/python3.6/dist-packages), Requirement.parse('tzlocal==2.0.0'), {'esphome'})
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 3238, in <module>
    @_call_aside
  File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 3222, in _call_aside
    f(*args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 3251, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 569, in _build_master
    return cls._build_from_requirements(__requires__)
  File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 582, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 770, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'tzlocal==2.0.0' distribution was not found and is required by esphome
```

**Additional information and things you've tried:**
Removed ESPHOME addon completely and reinstalled the actual production build 1.14.4 which runs fine without problems.
Then tried again to go to ESPHome dev version 1.16.0-dev by specifiying dev as version in the config addon.
Installation of dev 1.16.0 is succesful but when the addon tries to start up it tells tzlocal version 2.0.0 was not found and is required by esphome. This message is repeated constantly in log.

holdestmade, 2020-09-01:
I came here to raise this issue, have the same problem exactly

Koumo42, 2020-09-02:
Also having this issue updating to 1.16.0-dev. Additionally I uninstalled ESPHome and reinstalled to version 1.14.4 (which worked) then switched versions to the pre-release builds v1.15.0b1-4 and all had the same issue. Also running ESPHome on HassOS 4.12, HA 0.114.4 on Proxmox.

hydrocontica, 2020-09-02:
I had exactly the same issue - running 1.16.0-dev on hassio, which arose after restarting hassio, leaving ESPHome in a looping state according to the logs, with tzlocal being the culprit. However, rebuilding ESPHome did resolve the issue.

bertdebondt, 2020-09-02:
> 
> 
> I had exactly the same issue - running 1.16.0-dev on hassio, which arose after restarting hassio, leaving ESPHome in a looping state according to the logs, with tzlocal being the culprit. However, rebuilding ESPHome did resolve the issue.

How do you rebuild ESPHome dev 1.16.0 in Hassio ? The ESPHome addon does not start up.

hydrocontica, 2020-09-02:
> How do you rebuild ESPHome dev 1.16.0 in Hassio ? The ESPHome addon does not start up.

Go to the Supervisor page, clicked the ESPHome (dev) card, and Rebuild should be one of the options near the top.

bertdebondt, 2020-09-02:
I only have ESPHome card, not ESPHome DEV card. I upgraded 1.14.4 to dev by specifying : 
esphome_version: dev  into configuration of ESPHome.
When starting the Addon you see upgrade to dev in the log file, but when after update starting it stops on the missing tzlocal version.
I don't see any rebuild option/button on the ESPHome integration card.

holdestmade, 2020-09-03:
> edit 1: think I found the dev card :
> dev card has to be installed differently:
> To install the dev version of ESPHome:
> In Hass.io: Add the ESPHome repository https://github.com/esphome/hassio https://github.com/esphome/hassio in Add-on store -> Repositories. Then install the add-on ESPHome Dev

This is the way it was done before the ESPHome addon was added to the Home Assistant Community Add-ons store.
It also works for me so have gone back to the old way for now

glmnet, 2020-09-04:
Only production and beta were promoted to the community addon store. Dev has not changed. 

Beta is having all features of dev but prod version is quite old now. 

I don't understand if there is still an issue with dev now. Rebuilding dev to check now.

holdestmade, 2020-09-04:
In the new Add-on, the docs state that dev version is available from it:

```
Option: esphome_version
Manually override which ESPHome version to use in the addon. For example to install the latest development version, use "esphome_version": "dev", or for version 1.10.0: "esphome_version": "v1.10.0"".
```

closed 2021-01-27 by stale[bot]