esphome/issues #1448: Tuya Dimmer brightness spontaneously changes from 100% to 0%.

kaners80, 2020-09-02:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Operating environment Docker
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP TYWE3S
**ESPHome version (latest production, beta, dev branch)**

ESPHome version 1.14.5

**Affected component:**

https://esphome.io/components/light/tuya.html

**Description of problem:**
Dimmer is used https://aliexpress.ru/item/33015809441.html
Home Assistant controls the brightness well, turning off/on the lights works.

If you press the light on button on the dimmer, the lamp lights up and the brightness spontaneously changes from 100% to 0%.

If the light is turned off and double-click on the light on button on the dimmer, then the light does not change brightness, but the status in the Home Assistant is indicated OFF.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: koridor_dimmer
  platform: ESP8266
  board: esp_wroom_02

wifi:
  ssid: "ssid_wifi"
  password: "ssid_wifi_pass"
  #use_address: koridor_dimmer.lan
  domain: .lan

# Enable logging
logger:
  #level: VERBOSE
  baud_rate: 0

# Enable Home Assistant API
api:
  password: "koridor_dimmer"

ota:
  password: "koridor_dimmer"

uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600

tuya:

light:
  - platform: "tuya"
    id: koridor_dimmer_light
    name: "koridor_dimmer_light"
    dimmer_datapoint: 2
    switch_datapoint: 1
    min_value: 20
    max_value: 240
    gamma_correct: 1.0
    default_transition_length: 0s
```

**Logs (if applicable):**

```
If you press the light on button on the dimmer, the lamp lights up and the brightness spontaneously changes from 100% to 0%.
[... 1145 lines cut ...]
[03:39:41][V][tuya:279]: Not sending unchanged value
[03:39:41][V][tuya:275]: Datapoint 1 set to 1
[03:39:41][V][tuya:279]: Not sending unchanged value
[03:39:41][V][tuya:098]: Received Tuya: CMD=0x07 VERSION=0 DATA=[02.02.00.04.00.00.00.F0 (8)] INIT_STATE=5
[03:39:41][V][tuya:235]: Datapoint 2 update to 240
[03:39:41][D][light:264]: 'koridor_dimmer_light' Setting:
[03:39:41][D][light:277]:   Brightness: 100%
[03:39:41][V][tuya:275]: Datapoint 2 set to 240
[03:39:41][V][tuya:279]: Not sending unchanged value
[03:39:41][V][tuya:275]: Datapoint 1 set to 1
[03:39:41][V][tuya:279]: Not sending unchanged value
[03:39:41][V][tuya:261]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=5
[03:39:41][V][tuya:098]: Received Tuya: CMD=0x00 VERSION=0 DATA=[01 (1)] INIT_STATE=5
[03:39:41][V][tuya:115]: MCU Heartbeat (0x01)
[03:39:42][V][tuya:261]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=5
[03:39:42][V][tuya:098]: Received Tuya: CMD=0x00 VERSION=0 DATA=[01 (1)] INIT_STATE=5
[03:39:42][V][tuya:115]: MCU Heartbeat (0x01)

```

**Additional information and things you've tried:**

If you remove dimmer_datapoin: 2, then the light button works correctly, but, of course, the brightness control in Home Assistant does not work.
[image]
[image]

aaliddell, 2021-01-06:
Not stale, still an issue

inspoadam, 2023-10-08:
I am having this exact issue issue right now. I have tried the fix proposed by @aaliddell, I I cannot seem to get the branch to start the dashboard (definitely user error). As a work around, I have tried hacking the fix into the official release (currently 2023.9.3) without success.

Adam, if you have time, I would appreaciate more verbose instruction after I have cloned your fork and set the current branch. Thx

aaliddell, 2023-10-26:
I just gave up and vendored the entire Tuya component folder using this: https://esphome.io/components/external_components.html

inspoadam, 2023-10-27:
Thanks @aaliddell. That helps me because I was thinking of doing the same.

inspoadam, 2023-11-04:
So, if you arrived here and are still struggling, here are a few things that may help you out. Firstly, I am not taking any credit for the updated code and any updates I have made. All I have done is taken great work by @aaliddell, which actually fixes the problem and refactored the code to work with the latest version of ESPHome (currently 2023.10.6).

That said, I started out by trying to install and use the repo locally and ended up getting nowhere fast because I was running another version of ESPHome and it kept defaulting to the official release code and ignoring the fix.

The next thing I did was install the repo in Docker and it worked perfectly. However, I very quickly realised that the version the repo was based on was too old and the firmware I created would not upload to the device.

Challenge accepted! The next thing I did was remove the "fix" repo from docker and replace it with the latest version (2023.10.6). I then copied the two files fo the tuya light (location: `esphome/esphome/components/tuya/light`) into the corresponding location in the "official" repo. I then tried to compile my firmware and quickly realised that there were a number of breaking changes between version. I worked through the problems one by one to create new `tuya_light.cpp` and `tuya_light.h` files. These files are as follows:

tuya_light.cpp
`#include "esphome/core/log.h"
#include "tuya_light.h"

namespace esphome {
namespace tuya {

static const char *TAG = "tuya.light";

void TuyaLight::setup() {
  if (this->dimmer_id_.has_value()) {
    this->parent_->register_listener(*this->dimmer_id_, [this](TuyaDatapoint datapoint) {
      // Ignore dimmer values received once switch is off, such as during switch-off
      // fade out. This allows restoring the present brightness on next switch on
      if (!this->state_->current_values.is_on()) {
        return;
      }

      this->inhibit_next_send_ = true;

      // Clip value to expected range, allowing for inverted range
      auto lower = std::min(this->min_value_, this->max_value_);
      auto upper = std::max(this->min_value_, this->max_value_);
      auto value = std::min(upper, std::max(lower, static_cast<int32_t>(datapoint.value_uint)));
      float brightness = float(value - this->min_value_) /
                         (this->max_value_ - this->min_value_);  // Don't use lower/upper here to allow inversion
      brightness = powf(brightness, 1.0 / this->state_->get_gamma_correct());  // Apply inverse gamma correction

      // Handle case where reported value is <= lower bound but not
      // zero, but we don't want light to appear off by setting
      // brightness = 0.0.
      // This can occur when we sent a value near the lower bound
      // and the returned value is not exactly what we set.
      if (lower > 0 && brightness == 0.0f) {
        brightness = 1.0 / (upper - lower);
      }

      ESP_LOGV(TAG, "Received brightness: %f %d", brightness, value);
      auto call = this->state_->make_call();
      call.set_brightness(brightness);
      call.perform();
    });
  }

  if (switch_id_.has_value()) {
    this->parent_->register_listener(*this->switch_id_, [this](TuyaDatapoint datapoint) {
      this->inhibit_next_send_ = true;
      ESP_LOGV(TAG, "Received switch: %d", datapoint.value_bool);
      auto call = this->state_->make_call();
      call.set_state(datapoint.value_bool);
      call.perform();
    });
  }
}

void TuyaLight::dump_config() {
  ESP_LOGCONFIG(TAG, "Tuya Dimmer:");
  if (this->dimmer_id_.has_value())
    ESP_LOGCONFIG(TAG, "   Dimmer has datapoint ID %u", *this->dimmer_id_);
  if (this->switch_id_.has_value())
    ESP_LOGCONFIG(TAG, "   Switch has datapoint ID %u", *this->switch_id_);
}
[... 113 lines cut ...]

`external_components:
  - source:
      type: local
      path: components
    components: [ tuya-local ]`

(don't forget to use your own naming if different).

Under the light configuration, change `platform: "tuya"` to the component defined above, which is tuya-local in my case (e.g. `platform: "tuya-local").

Here is a code snippet:

`light:
  - platform: "tuya-local"
    name: State
    id: ${device_name}
    dimmer_datapoint: 2
    switch_datapoint: 1
    min_value: 10
    max_value: 255
    restore_mode: RESTORE_DEFAULT_OFF
    icon: "mdi:lightbulb"`

Finally, save the configure, check with validation and then install. If you have previously built and installed a configuration, it's really important to delete the build files (menu option `Clean Build Files`).