esphome/issues #1449: BME280 sensor show values near two grades higher than other temperature sensors

damocles-git, 2020-09-02:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

esp32cam

**ESPHome version (latest production, beta, dev branch)**

Version: 1.14.5

**Affected component:**

BME280

**Description of problem:**

I have several temperature sensors across the floor but the BME280 managed by esphome is near two grades celsious higher than the rest.

**Problem-relevant YAML-configuration entries:**
```yaml
  - platform: xiaomi_lywsdcgq
    mac_address: 58:2D:34:38:38:76
    temperature:
      name: "Xiaomi Temperature"
      accuracy_decimals: 2
    humidity:
      name: "Xiaomi Humidity"
      accuracy_decimals: 2
    battery_level:
      name: "Xiaomi Battery Level"
      accuracy_decimals: 2
  - platform: bme280
    temperature:
      name: "esp32cam_02_Temperature"
      oversampling: '1x'
      accuracy_decimals: 2
    pressure:
      name: "esp32cam_02_Presure"
      oversampling: '1x'
      accuracy_decimals: 2
    humidity:
      name: "esp32cam_02_Humidity"
      oversampling: '1x'
      accuracy_decimals: 2
    address: 0x76
    update_interval: 60s

```

**Logs (if applicable):**

The next four logs are of sensors in the same room (hall) with all doors closed for better temperature stability.

esp32cam/esphome + BME280: **25.0°C**
```
[14:47:46][D][bme280.sensor:197]: Got temperature=25.0°C pressure=1015.8hPa humidity=52.5%
[14:47:46][D][sensor:092]: 'esp32cam_02_Temperature': Sending state 25.03000 °C with 2 decimals of accuracy
[14:47:46][D][sensor:092]: 'esp32cam_02_Presure': Sending state 1015.82623 hPa with 2 decimals of accuracy
[14:47:46][D][sensor:092]: 'esp32cam_02_Humidity': Sending state 52.52734 % with 2 decimals of accuracy
```
esp32cam/esphome + Xiaomi LYWSDCGQ: **23.0°C**
```
[14:47:59][D][xiaomi_ble:161]: Got Xiaomi LYWSDCGQ (58:2D:34:38:38:76):
[14:47:59][D][xiaomi_ble:164]:   Temperature: 23.0°C
[14:47:59][D][xiaomi_ble:167]:   Humidity: 61.8%
[14:47:59][D][sensor:092]: 'Xiaomi Temperature': Sending state 23.00000 °C with 2 decimals of accuracy
[14:47:59][D][sensor:092]: 'Xiaomi Humidity': Sending state 61.80000 % with 2 decimals of accuracy
```
wemos_mini_d1/tasmota + BME280: **22.7°C**
```
13:16:04 RSL: tele/tasmota_A7CF8A/SENSOR = {"Time":"2020-09-02T13:16:04","BME280":{"Temperature":22.7,"Humidity":54.1,"DewPoint":13.0,"Pressure":1015.8},"PressureUnit":"hPa","TempUnit":"C"}
```

zigbee2mqtt + Xiaomi Aqara WSDCGQ11LM: **22.87°C**
```
2020-09-02T14:47:49+02:00 rpi0-01 npm[8671]: #033[32mzigbee2mqtt:info #033[39m 2020-09-02 14:47:49: MQTT publish: topic 'casa/zigbee2mqtt/xiaomi-weather-01', payload '{"temperature":22.87,"linkquality":0,"humidity":63.85,"pressure":1013.5,"battery":97,"voltage":2995,"device":{"friendlyName":"xiaomi-weather-01","model":"WSDCGQ11LM","ieeeAddr":"0x00158d00034d1c9b","networkAddress":63325,"type":"EndDevice","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","applicationVersion":3,"stackVersion":2,"zclVersion":1,"hardwareVersion":30,"dateCode":"20161129","softwareBuildID":"3000-0001"}}'
```
**Additional information and things you've tried:**

I have tryed to change oversampling values from 1x to 16x and does not change the result of the readings, esphome+BME280 and is **consistently** almost two degrees higher than all other sensors.

ipetrovits, 2020-09-02:
Hi, I ran into the same issue. Have several BME280s working together, and I had to calibrate them in the first place with some offset.
Then I needed to do something about the heat dissipation, because those also produce a lot of heat (i.e. compared DHT) - and if the room is not ventilated constantly, they heat up - so I also put a small heatsink on all of them. Since then I had no issues :)
By the way this is not an ESPHome issue at all.

damocles-git, 2020-09-02:
Hi, 
The tests I ran were using identical BME280 sensors, so if it was related to heat dissipation it would also happen with wemos_mini_d1/tasmota +BME280 and it doesn't happens, also the values from tasmota are almost identical to both xiaomi sensors. So I thinks it is something related to esphome.

fkuersch, 2020-09-04:
Hi, same issue here. I found these related issues:

Duplicate of #402
Related: https://github.com/arendst/Tasmota/issues/1051

I can confirm that changing the oversampling rate from 16x (default) to 1x didn't have any effect on the temperature readings in my case either.

damocles-git, 2020-09-04:
yes, I read #402 before posting but

closed 2020-09-04 by damocles-git