esphome/issues #1452: documentation example of measuring VCC voltage does not provide sensible data

pille, 2020-09-03:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

integrated with hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266/d1_mini & nodemcuv2

**ESPHome version (latest production, beta, dev branch)**

ESPHome v1.14.5

**Affected component:**

https://esphome.io/components/sensor/adc.html#adc-esp8266-vcc
https://github.com/esphome/esphome-docs/blob/current/components/sensor/adc.rst

**Description of problem:**
when configuring the sensor according to documentation, "meassures" 2.91V.
this value does not change over time and is even the same on different devices (including different power supplies like 6V or 5V via USB).

**Problem-relevant YAML-configuration entries:**
```yaml
sensor:
  - platform: adc
    pin: VCC
    name: "VCC Voltage"
```

**Logs (if applicable):**

```
[D][adc:056]: 'VCC Voltage': Got voltage=2.91V
```

**Additional information and things you've tried:**

rradar, 2020-09-06:
@pile
> when configuring the sensor according to documentation, "meassures" 2.91V.

Sounds correct to me, as the docs state "_the voltage **the chip is getting**_"

> this value does not change over time and is even the same on different devices (including different power supplies like 6V or 5V via USB). 

This is correct because **the chip** is still getting it's ~3V despite your higher input voltage.

You could try to feed the esp with less than your measured 2.91V and observe the adv voltage again. The esp8266's should work down to a voltage of around 2.5V.

The use case for this is also mentioned in the docs btw:

> This can be useful in situations where you want to shut down the chip if the **voltage is low** when using a battery.

pille, 2020-09-06:
@rradar, you're right. i misinterpreted the docs and verified your claim by measuring the ESP8266 on a battery. it became flaky below 2.5V and died around 2.0V.

closed 2020-09-08 by pille