esphome/issues #1453: Can't compile SN74HC595 without optional oe pin

rradar, 2020-09-05:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hassio

**ESP (ESP32/ESP8266, Board/Sonoff):**
esp8266, d1 mini

**ESPHome version (latest production, beta, dev branch)**
1.15b4

**Affected component:**
SN74HC595 I/O Expander (shift register)
https://next.esphome.io/components/sn74hc595.html

**Description of problem:**
Can't compile a yaml with the SN74HC595 component without declaring the optional oe pin

**Problem-relevant YAML-configuration entries:**
```yaml
sn74hc595:
  - id: 'sn74hc595_hub'
    data_pin: D5
    clock_pin: D8
    latch_pin: D7
#    oe_pin: D6
    sr_count: 1

```

**Logs (if applicable):**

```
INFO Reading configuration /config/esphome/mysupershifter.yaml...
INFO Generating C++ source...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 11, in <module>
    load_entry_point('esphome', 'console_scripts', 'esphome')()
  File "/opt/esphome/esphome/__main__.py", line 554, in main
    return run_esphome(sys.argv)
  File "/opt/esphome/esphome/__main__.py", line 541, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "/opt/esphome/esphome/__main__.py", line 307, in command_run
    exit_code = write_cpp(config)
  File "/opt/esphome/esphome/__main__.py", line 122, in write_cpp
    generate_cpp_contents(config)
  File "/opt/esphome/esphome/__main__.py", line 134, in generate_cpp_contents
    CORE.flush_tasks()
  File "/opt/esphome/esphome/core.py", line 653, in flush_tasks
    next(task)
  File "/opt/esphome/esphome/core.py", line 438, in _wrapper_generator
    for x in var:
  File "/opt/esphome/esphome/core.py", line 434, in _wrapper_generator
    var = gen.send(var)
  File "/opt/esphome/esphome/components/sn74hc595/__init__.py", line 37, in to_code
    oe_pin = yield cg.gpio_pin_expression(config[CONF_OE_PIN])
KeyError: 'oe_pin'

```

**Additional information and things you've tried:**

Declaring the (optional) `oe_pin` makes the yaml compile successfully

lazlyhu, 2020-09-19:
I confirm this issue as wrote above.

rradar, 2020-09-19:
Actually I'm also not able to get the component working with `oe_pin` and external pull up (like mentioned in the [docs](https://esphome.io/components/sn74hc595.html)). Did you succeed using the SN74HC595 @lazlyhu ?

lazlyhu, 2020-09-19:
@rradar I added a floating pin as `oe_pin` value, without physically connect that to the IC. Looks it's working without issues.

jesserockz, 2020-09-19:
https://github.com/esphome/esphome/commit/32a46807e74d9d25e5c8860455631ff725ea97d3

closed 2021-01-08 by glmnet