esphome/issues #1454: How to creat a variable ?

OlivFabre, 2020-09-06:
Hello,
I didn't find out how to creat a variable to stock a temperature.
I want to check and stock every day at 5PM the temperature and use this value for 24H, untile replace it by the new value.
What kind of component to use ?
Thank you

ESP8266
Home Assistant 0.114.4

arch | armv7l
-- | --
chassis | embedded
dev | false
docker | true
docker_version | 19.03.11
hassio | true
host_os | HassOS 4.12
installation_type | Home Assistant OS
os_name | Linux
os_version | 4.19.127-v7l
python_version | 3.8.3
supervisor | 240
timezone | Europe/Paris
version | 0.114.4
virtualenv | false

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**ESPHome version (latest production, beta, dev branch)**

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

rradar, 2020-09-06:
@OlivFabre please [read the docs ](https://esphome.io/guides/automations.html#bonus-2-global-variables)(it even includes a search function)...

If you need help with your individual project check out the [discord](https://discord.gg/KhAMKrd) or the [forum](https://community.home-assistant.io/c/third-party/esphome)

closed 2020-09-06 by rradar

OlivFabre, 2020-09-06:
@rradar 
Thank you et have a good day