esphome/issues #1455: Custom Component; OpenTherm library won't compile

Bart-1992, 2020-09-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io
**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP8266

**ESPHome version (latest production, beta, dev branch)**
V1.14

**Affected component:**
Custom Component

**Description of problem:**
I'm trying to intergrate the following library (Opentherm library --> To control boiler using 'standardized opentherm = modulating protocol'' as a custom component into ESPhome: https://platformio.org/lib/show/5483/OpenTherm%20Library

I've followed the guide from the ESPhome docs, but the compiler keeps throwing errors. 

The custom file looks like this :

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: opentherm_controller
  platform: ESP8266
  board: nodemcuv2
  includes: 
  - custom_opentherm_controller.h
  libraries:
  - "OpenTherm Library"
  
wifi:
  ssid: ",......"
  password: "........"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "opentherm_controller"
    password: "ue95AYc0VhYC"

captive_portal:

# Enable logging
logger:
  level: VERBOSE 

# Enable Home Assistant API
api:

ota:

custom_component:
- lambda: |-
    auto my_custom = new MyCustomComponent();
    return {my_custom};

```
And the custom created .h file
```custom component.h
#include "esphome.h"
#include <Arduino.h>
#include <OpenTherm.h>

const int mInPin = 4; //for Arduino, 4 for ESP8266
const int mOutPin = 5; //for Arduino, 5 for ESP8266

const int sInPin = 12; //for Arduino, 12 for ESP8266
const int sOutPin = 13; //for Arduino, 13 for ESP8266

OpenTherm mOT(mInPin, mOutPin);
OpenTherm sOT(sInPin, sOutPin, true);

class MyCustomComponent : public Component{
 public:
 
void setup() override 
{
    Serial.begin(9600);	//9600 supported by OpenTherm Monitor App
    mOT.begin (mHandleInterrupt);
    sOT.begin(sHandleInterrupt, processRequest);
   // Serial.print("Starting the system");
}

 
void ICACHE_RAM_ATTR mHandleInterrupt() {
    mOT.handleInterrupt();
}

void ICACHE_RAM_ATTR sHandleInterrupt() {
    sOT.handleInterrupt();
}

void processRequest(unsigned long request, OpenThermResponseStatus status) {
   // Serial.println("T" + String(request, HEX));  //master/thermostat request
    unsigned long response = mOT.sendRequest(request);
    if (response) {
       // Serial.println("B" + String(response, HEX)); //slave/boiler response
        sOT.sendResponse(response);
    }
}

void loop() override
{
  //Get Boiler Temperature
    float temperature = mOT.getBoilerTemperature();
//    Serial.println("Ketel temperatuur is:");
//    Serial.println(temperature);
    temperature = sOT.getBoilerTemperature();
//    Serial.println("thermostaat temperatuur is:");
//    Serial.println(temperature);
//    sOT.process();

}
};

```
**Logs (if applicable):**

```
INFO Reading configuration /config/esphome/opentherm_controller.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/opentherm_controller
Processing opentherm_controller (board: nodemcuv2; framework: arduino; platform: espressif8266@2.2.3)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 2.20502.0 (2.5.2) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - toolchain-xtensa 1.40802.2 (4.8.2)
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <OpenTherm Library> 1.1.0
|-- <ESPAsyncWebServer-esphome> 1.2.6
|   |-- <ESPAsyncTCP-esphome> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
Compiling /data/opentherm_controller/.pioenvs/opentherm_controller/src/main.cpp.o
In file included from src/main.cpp:12:0:
src/custom_opentherm_controller.h: In member function 'virtual void MyCustomComponent::setup()':
src/custom_opentherm_controller.h:21:32: error: no matching function for call to 'OpenTherm::begin(<unresolved overloaded function type>)'
     mOT.begin (mHandleInterrupt);
                                ^
src/custom_opentherm_controller.h:21:32: note: candidates are:
In file included from src/custom_opentherm_controller.h:3:0,
                 from src/main.cpp:12:
/data/opentherm_controller/.piolibdeps/opentherm_controller/OpenTherm Library/src/OpenTherm.h:136:7: note: void OpenTherm::begin(void (*)())
  void begin(void(*handleInterruptCallback)(void));
       ^
/data/opentherm_controller/.piolibdeps/opentherm_controller/OpenTherm Library/src/OpenTherm.h:136:7: note:   no known conversion for argument 1 from '<unresolved overloaded function type>' to 'void (*)()'
/data/opentherm_controller/.piolibdeps/opentherm_controller/OpenTherm Library/src/OpenTherm.h:137:7: note: void OpenTherm::begin(void (*)(), void (*)(long unsigned int, OpenThermResponseStatus))
  void begin(void(*handleInterruptCallback)(void), void(*processResponseCallback)(unsigned long, OpenThermResponseStatus));
       ^
/data/opentherm_controller/.piolibdeps/opentherm_controller/OpenTherm Library/src/OpenTherm.h:137:7: note:   candidate expects 2 arguments, 1 provided
In file included from src/main.cpp:12:0:
src/custom_opentherm_controller.h:22:47: error: no matching function for call to 'OpenTherm::begin(<unresolved overloaded function type>, <unresolved overloaded function type>)'
     sOT.begin(sHandleInterrupt, processRequest);
                                               ^
src/custom_opentherm_controller.h:22:47: note: candidates are:
In file included from src/custom_opentherm_controller.h:3:0,
                 from src/main.cpp:12:
/data/opentherm_controller/.piolibdeps/opentherm_controller/OpenTherm Library/src/OpenTherm.h:136:7: note: void OpenTherm::begin(void (*)())
  void begin(void(*handleInterruptCallback)(void));
       ^
/data/opentherm_controller/.piolibdeps/opentherm_controller/OpenTherm Library/src/OpenTherm.h:136:7: note:   candidate expects 1 argument, 2 provided
/data/opentherm_controller/.piolibdeps/opentherm_controller/OpenTherm Library/src/OpenTherm.h:137:7: note: void OpenTherm::begin(void (*)(), void (*)(long unsigned int, OpenThermResponseStatus))
  void begin(void(*handleInterruptCallback)(void), void(*processResponseCallback)(unsigned long, OpenThermResponseStatus));
       ^
/data/opentherm_controller/.piolibdeps/opentherm_controller/OpenTherm Library/src/OpenTherm.h:137:7: note:   no known conversion for argument 1 from '<unresolved overloaded function type>' to 'void (*)()'
*** [/data/opentherm_controller/.pioenvs/opentherm_controller/src/main.cpp.o] Error 1
========================== [FAILED] Took 5.93 seconds ==========================

```

**Additional information and things you've tried:**

The strange thing is it compiles perfectly in the arduino IDE, and got it working. So what am i doing wrong? 

It looks like the compiler has some issues with the function types, i've allready tried changing them into : 

for example: 
- mOT.begin (void (*mHandleInterrupt) (void));
- mOT.begin ((void *) mHandleInterrupt);

but without succes, compiler then gives following errors :
```
In file included from src/main.cpp:12:0:
src/custom_opentherm_controller.h: In member function 'virtual void MyCustomComponent::setup()':
src/custom_opentherm_controller.h:21:25: error: invalid use of member function (did you forget the '()' ?)
     mOT.begin ((void *) mHandleInterrupt);
                         ^
src/custom_opentherm_controller.h:22:24: error: 'HandleInterrupt' was not declared in this scope
     sOT.begin((void *) HandleInterrupt, processRequest);
                        ^
*** [/data/opentherm_controller/.pioenvs/opentherm_controller/src/main.cpp.o] Error 1
```

OttoWinter, 2020-09-15:
Hi, the issue is that the library expects a function pointer (to a pure C function), but `mHandleInterrupt` is a *member function pointer* (slightly different, because it additionally needs a `this` pointer for the class). The way to work around this would be to declare a `handle_interrupt` function globally (outside the sensor class).

If you need to access the `MyCustomComponent` class methods, you can create a global `MyCustomComponent *my_global_var` object and call `my_global_var = this` in `setup()`

Bart-1992, 2020-09-26:
Hi @OttoWinter ,

Thanks alot for your responce! 

At the moment I got it partially working ( the OpenTherm handling is functioning properly now ).
However the ESPHOME part is not functioning properly anymore. 

I've used a pollingcomponent with an updateinterval of 1000 ms ( which is required by the OpenTherm protocol ). However at 1000ms polling update interval it continuously disconnects/connects from HomeAssistant., Because of this the Homeassistant sensor, and the switch component ( created in the .yaml config ) are not updated from HomeAssistant. ( please check the logs at the end of this post ) 

Currently i've used the following custom opentherm.h code :  

````
#include "esphome.h"
#include <Arduino.h>
#include <OpenTherm.h>

const int mInPin = 4; //for Arduino, 4 for ESP8266
const int mOutPin = 5; //for Arduino, 5 for ESP8266

const int sInPin = 12; //for Arduino, 12 for ESP8266
const int sOutPin = 13; //for Arduino, 13 for ESP8266

OpenTherm mOT(mInPin, mOutPin);
OpenTherm sOT(sInPin, sOutPin, true);

float boiler_out_temp = 60.0;
bool boiler_switch = false;

class MyCustomSensor : public PollingComponent, public Sensor{
 public:
 
MyCustomSensor() : PollingComponent(1000) { }

   static void ICACHE_RAM_ATTR mHandleInterrupt() {
        mOT.handleInterrupt();
    }
    
    static void ICACHE_RAM_ATTR sHandleInterrupt() {
        sOT.handleInterrupt();
    }
    static void processRequest(unsigned long request, OpenThermResponseStatus status) {
      Serial.println("T" + String(request, HEX));  //master/thermostat request
      ESP_LOGD("custom", "Thermostat request %x\n", request);
      unsigned long response = mOT.sendRequest(request);
      
      if (response) {
        Serial.println("B" + String(response, HEX)); //slave/boiler response
        ESP_LOGD("custom", "Boiler response %x\n", request);
        sOT.sendResponse(response);
      }
    }
    
   static void manualBoiler()
    {
        //Set/Get Boiler Status
        bool enableCentralHeating = true;
        bool enableHotWater = true;
        bool enableCooling = false;
        
        unsigned long response = sOT.setBoilerStatus(enableCentralHeating, enableHotWater, enableCooling);
        OpenThermResponseStatus responseStatus = sOT.getLastResponseStatus();
        if (responseStatus == OpenThermResponseStatus::SUCCESS) {
           // ESP_LOGD("custom", "Central Heating: %s", ((sOT.isCentralHeatingActive(response).c_str()) ? "on" : "off"));
            Serial.println("Central Heating: " + String(sOT.isCentralHeatingActive(response) ? "on" : "off"));
            Serial.println("Hot Water: " + String(sOT.isHotWaterActive(response) ? "on" : "off"));
            Serial.println("Flame: " + String(sOT.isFlameOn(response) ? "on" : "off"));
        }
        if (responseStatus == OpenThermResponseStatus::NONE) {
            ESP_LOGD("custom", "Error: OpenTherm is not initialized");
            Serial.println("Error: OpenTherm is not initialized");
[... 185 lines cut ...]
[22:53:19]T80190000
[22:53:19][D][custom:032]: Thermostat request 80190000
[22:53:19]
[22:53:19]Bc019356b
[22:53:19][D][custom:037]: Boiler response 80190000
[22:53:19]
[22:53:19][V][app:081]: A component took a long time in a loop() cycle (1.00 s).
[22:53:19][V][app:082]: Components should block for at most 20-30ms in loop().
[22:53:19][D][custom:101]: Boiler temp setted by user nan
[22:53:19][D][custom:102]: Boiler switch setted to false
[22:53:20]T7b0000
[22:53:20][D][custom:032]: Thermostat request 7b0000
[22:53:20]
[22:53:20]B407b006e
[22:53:20][D][custom:037]: Boiler response 7b0000
[22:53:20]
[22:53:20][V][app:081]: A component took a long time in a loop() cycle (1.07 s).
[22:53:20][V][app:082]: Components should block for at most 20-30ms in loop().
[22:53:20][D][custom:101]: Boiler temp setted by user nan
[22:53:20][D][custom:102]: Boiler switch setted to false
[22:53:21]T80000200
[22:53:21][D][custom:032]: Thermostat request 80000200
[22:53:21]
[22:53:21]B40000200
[22:53:21][D][custom:037]: Boiler response 80000200

closed 2021-02-07 by stale[bot]