esphome/issues #1457: 502 Bad gateway

Sailorsbe, 2020-09-09:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**ESPHome version (latest production, beta, dev branch)**

**Affected component:**

**Description of problem:**
Hello
For several days now I'm trying to configure a new component via de esphome addon in home assistant and get always the 502 error to see. What is happening on the ESPHOME server?

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

OttoWinter, 2020-09-15:
Hi! Could you take a look at the logs in the logs tab of the add-on in the HA dashboard? Thanks.

Also: sometimes it takes a bit for the addon to boot up (only 10 seconds max), during that time you'll see an 502 bad gateway message.

deus-ex-silicium, 2020-11-27:
I had the same problem on a fresh install of HassOS 4.17, the only things I did to it was install the official terminal add-on and the esphome one. I was getting 502's and the add-on did not want to bootup no matter how long I waited, I reinstalled it and that seems to have fixed to problem. Sorry that I forgot to grab the log while the problem was still occurring.

closed 2021-04-11 by stale[bot]