esphome/issues #1458: TTGO T5 V2.3 ePaper 2.13 inch shows artifacts in sleep mode (display content not held while off)

chris-nite, 2020-09-09:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

TTGO T5 V2.3 2.13 inch ePaper-ESP32-Module (powered by Lithium Battery or alternatively through usb)
**ESPHome version (latest production, beta, dev branch)**

ESPHome 0.1.16-dev branch
**Affected component:**

**Description of problem:**
The ePaper-module is working as expected since the 0.1.16-dev branch and displays all the content i threw it at, but every time i power off the module (through deep_sleep-function, or manually via the power-switch), the shown content on the ePaper fades into nirvana. Unfortunately because of this, the deep_sleep-function atm is of no use to me, as the temperatures i currently show are not visible anymore, everytime the esp32 goes to sleep.
I'm really not sure if this is a problem with the esphome-configuration, or the hardware itself. I got two of those TTGO T5-modules and both of them inhibit the same problems.
I am very grateful about any ideas.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: epaper_display1
  platform: ESP32
  board: esp32dev

...

spi:
  clk_pin: 18
  mosi_pin: 23

display:
  - platform: waveshare_epaper
    cs_pin: 5
    dc_pin: 17
    busy_pin: 4
    reset_pin: 16
    model: 2.13in-ttgo-b73
    rotation: 90°
    full_update_every: 30
    update_interval: 120s

...
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

chris-nite, 2020-09-14:
Okay, now it gets interesting. I contacted the hardware-vendor of the TTGO T5-modules. They were pretty curious about it and sent me test-binaries together with the esp32-flashing tool. Interestingly the problem of the screens fading out in deep sleep-mode won't happen with the stock images. Both displays now run on those images and retain the Lilygo-Image while off.
For me it seems, that there seems to be something "wrong"/different with the esphome-model-configuration "2.13in-ttgo-b73" in the dev-branch (or in the pin-settings maybe)?
I'm open for any testing.
Thanks in advance and best wishes.

chris-nite, 2020-09-23:
Any ideas? :( The technical support told me, that i should try to go the arduino ide-route with mqtt to get where i want to be, especially with functional deep sleep without fading out screens, but i really would prefer it a lot, to get this with esphome working. The question is, if other epaper-users (based on ttgo-models or waveshare) inhibit the same behavior when toggled off or going into sleep mode?

matikij, 2020-10-10:
@chris-nite I have a slightly different version of the hardware( that I added support for in https://github.com/esphome/esphome/pull/1326), but I am using deep sleep and I see no artifacts. My config is a bit different than your though:
```
esphome:
  name: $devicename
  platform: ESP32
  board: lolin32
[...]
spi:
  clk_pin:  18
  mosi_pin: 23
[...]
display:
  - platform: waveshare_epaper
    cs_pin: 5
    dc_pin: 17
    reset_pin: 16
    busy_pin: 4
    model: 2.13in-ttgo-b1
    full_update_every: 10
    update_interval: 2s
    rotation: 0
    lambda: |-
      [...]
deep_sleep:
  run_duration: 3s
  sleep_duration: 10min
  wakeup_pin:
    number: 39
    inverted: True
  wakeup_pin_mode: IGNORE
  id: deep_sleep_1
```
Notably I have `lolin32` board set. I tried with `esp32dev` set and I didn't get the behavior you are describing.
Maybe it has something to do with wakeup ?

chris-nite, 2020-10-10:
Thank you very much matikij for your reply.
I've just tried the lolin32 board in my esphome-config, but unfortunately it didn't change anything for the better, as the display still inhibits these problems.
It seems the display artifacts are quite a lot stronger in deep sleep, than when i toggle the T5 completely off. When going to deep sleep mode, the display almost instantly shows strong artifacts and shadows of previously shown contents. When it comes back it instantly refreshes the picture with great clarity and contrast.
When i turn the T5 off, it takes about 30sec or longer for the display contrast to fade out a bit. Interestingly there are almost no shadows on the display.
This behaviour is really driving me nuts, as the problems are not there at all with the "base" firmware-image from the manufacturer and i think it might be something wrong with the "2.13in-ttgo-b73" model-configuration in esphome. Maybe there are some mixed pin-settings or wrong information set up here in the code (like the pixel-dimensions as mentioned above). 

Btw. matikij, what pixel-dimensions did you set up in your code-commits for your T5-model?

matikij, 2020-10-12:
Hi, the panel itself is 122x250, but internally it uses 128x250, just like
the other ttgo-t5 panels. When changed to something different the display
is garbled.

Did you try with "2.13in-ttgo" model? Can you post the code (and links to
libraries) that your display is working with? Perhaps there's yet another
model of ttgo-t5?

chris-nite, 2020-10-23:
Thank you for your response @matikij .
When compiling with the general "2.13in-ttgo"-model unfortunately the esp won't bring anything up on the display.
While the display is actively powered and the esp32 is on everything works properly and without any issues. The problems only arise, when the esp32 goes into deep-sleep (stronger artifacts) or is powered off (fewer but still existant artifacts).

this is the code i'm currently using:
```
esphome:
  name: epaper_display1
  platform: ESP32
  board: esp32dev
#  board: lolin32

substitutions:
  devicename: epaper_display1

wifi:
  ssid: "<<removed>>"
  password: "<<removed>>"
  fast_connect: true

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Epaper Display1 Fallback Hotspot"
    password: "<<removed>>"

spi:
  clk_pin: 18
  mosi_pin: 23

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "<<removed>>"

ota:
  password: "<<removed>>"

time:
  - platform: homeassistant
    id: esptime

font:
  - file: 'fonts/materialdesignicons.ttf'
    id: icon_font_25
    size: 25
    glyphs: [
      "󰔏", # mdi-thermometer
      "󰖎", # mdi-water-percent
      # Alert
      "", # mdi-alert-outline
      # Weather
      "", # mdi-weather-sunny
      "", # mdi-weather-night
      "", # mdi-weather-cloudy
      "", # mdi-weather-pouring
      "", # mdi-weather-snowy-rainy
      "ｓ", # mdi-weather-snowy-heavy
      "", # mdi-weather-windy-variant
      "", # mdi-weather-fog
      "", # mdi-weather-partly-cloudy
      "ｎ" # mdi-weather-night-partly-cloudy
      ]
  - file: "fonts/materialdesignicons.ttf"
    id: icon_font_35
    size: 35
    glyphs: [
      "󰔏", # mdi-thermometer
      "󰖎"  # mdi-water-percent
      ]
  - file: "fonts/Roboto-Regular.ttf"
    id: bigfont
    size: 48
  - file: "fonts/Kanit-Medium.ttf"
    id: medfont
    size: 24
  - file: "fonts/Kanit-Regular.ttf"
    id: medfontreg
    size: 32
  - file: "fonts/Kanit-Regular.ttf"
    id: medfontregsmall
    size: 24

sensor:
  - platform: uptime
    name: "$devicename Uptime Sec"
    id: uptime_sec
    internal: true
  - platform: wifi_signal
    name: "$devicename WiFi Signal"
    update_interval: 360s
    
  - platform: homeassistant
    name: "Wetter Berlin - Temperatur"
    id: berlin_temperature
    internal: true
    entity_id: sensor.weather_temperature
  - platform: homeassistant
    name: "Wetter Berlin - Feuchtigkeit"
    id: berlin_humidity
    internal: true
    entity_id: sensor.weather_humidity
  - platform: homeassistant
    name: "Nikolas Zimmertemperatur"
    id: nikolas_temperature
    internal: true
    entity_id: sensor.temperature_20
  - platform: homeassistant
    name: "Elenas Zimmertemperatur"
    id: elena_temperature
    entity_id: sensor.temperature_4
    internal: true
  - platform: homeassistant
    name: "Wohnzimmer - Temperatur"
    id: wozi_temperature
    entity_id: sensor.temperature_11
    internal: true
  - platform: homeassistant
    name: "Wohnzimmer - Feuchtigkeit"
    id: wozi_humidity
    entity_id: sensor.humidity_12
    internal: true   
  - platform: homeassistant
    name: "Kühlschranktemperatur"
    id: kuehlschrank_temperature
    entity_id: sensor.temperature_14
    internal: true

text_sensor:
  - platform: version
    name: "$devicename Version"

  - platform: wifi_info
    ip_address:
      name: "$devicename IPv4"
      icon: "mdi:server-network"
    ssid:
      name: "$devicename Connected SSID"
      icon: "mdi:wifi"
  - platform: template
    name: "$devicename Uptime"
    icon: "mdi:timer"
    update_interval: 360s
    lambda: |-
      int seconds = id(uptime_sec).state;
      int days = seconds / (24 * 3600);
      seconds = seconds % (24 * 3600); 
      int hours = seconds / 3600;
      seconds = seconds % 3600;
      int minutes = seconds /  60;
      return { (String(days) +"d " + String(hours) +"h " + String(minutes) +"m").c_str() };

  - platform: homeassistant
    name: "Weather Forecast"
    entity_id: weather.home
    id: forecast
    internal: true

display:
  - platform: waveshare_epaper
    cs_pin: 5
    dc_pin: 17
    busy_pin: 4
    reset_pin: 16
    model: 2.13in-ttgo-b73
    rotation: 90°
    full_update_every: 10
    update_interval: 60s
    #info pixel dimensions: 250x122px
    lambda: |-
         it.print(125, 16, id(medfontregsmall), TextAlign::CENTER, "Berlin");
         it.printf(125, 42, id(medfontreg), TextAlign::CENTER, "%.1f°C      %.0f%%", id(berlin_temperature).state, id(berlin_humidity).state);
         it.printf(12, 44, id(icon_font_35), TextAlign::CENTER, "󰔏");
         it.printf(140, 44, id(icon_font_35), TextAlign::CENTER, "󰖎"); 
         it.line(0, 66, 250, 66);
         it.print(125, 76, id(medfontregsmall), TextAlign::CENTER, "Innen");
         it.printf(125, 102, id(medfontreg), TextAlign::CENTER, "%.1f°C      %.0f%%", id(wozi_temperature).state, id(wozi_humidity).state);
         it.printf(12, 104, id(icon_font_35), TextAlign::CENTER, "󰔏");
         it.printf(140, 104, id(icon_font_35), TextAlign::CENTER, "󰖎"); 

#deep_sleep:
#  run_duration: 8s
#  sleep_duration: 10min
#  wakeup_pin:
#    number: 39
#    inverted: True
#  wakeup_pin_mode: IGNORE
#  id: deep_sleep_1

```

matikij, 2020-10-23:
How about posting the code (and links to libraries) that work properly? Thing is that there is something with deep sleep here https://github.com/esphome/esphome/blob/dev/esphome/components/waveshare_epaper/waveshare_epaper.h#L84 but cannot know what to compare this with

chris-nite, 2020-10-24:
Actually i'm not that confident with arduino-coding, but the base-image (which is currently on my second epaper-unit) is based on this ide-code (from the manufacturer of the lilygo t5-devices):

[github-repo](https://github.com/Xinyuan-LilyGO/LilyGo-T5-ink-series/blob/master/LilyGo_T5_V2.3/LilyGo_T5_V2.3.ino)
```
// include library, include base class, make path known
#include <GxEPD.h>
#include "SD.h"
#include "SPI.h"

//! There are three versions of the 2.13 screen,
//  if you are not sure which version, please test each one,
//  if it is successful then it belongs to the model of the file name
//  关于v2.3版本的显示屏版本,如果不确定购买的显示屏型号,请每个头文件都测试一遍.

//#include <GxGDE0213B1/GxGDE0213B1.h>      // 2.13" b/w
//#include <GxGDEH0213B72/GxGDEH0213B72.h>  // 2.13" b/w new panel
#include <GxGDEH0213B73/GxGDEH0213B73.h>  // 2.13" b/w newer panel

int bmpWidth = 150, bmpHeight = 39;
//width:150,height:39
const unsigned char lilygo[] = {0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0xf7, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x07, 0x31, 0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x0f, 0xfc, 0xc0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x0d, 0xfe, 0x60, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x19, 0xff, 0x20, 0x7f, 0xc0, 0x03, 0xfc, 0x7f, 0x80, 0x07, 0xf8, 0x0f, 0xf0, 0x00, 0xfe, 0x00, 0x03, 0xff, 0x80, 0x19, 0xe7, 0x30, 0x7f, 0xc0, 0x03, 0xfc, 0x7f, 0x80, 0x07, 0xfc, 0x0f, 0xf0, 0x07, 0xff, 0xc0, 0x0f, 0xff, 0xe0, 0x19, 0xe7, 0xb0, 0x7f, 0xc0, 0x03, 0xfc, 0x7f, 0x80, 0x03, 0xfc, 0x1f, 0xe0, 0x0f, 0xff, 0xe0, 0x1f, 0xff, 0xf8, 0x19, 0xff, 0x10, 0x7f, 0xc0, 0x03, 0xfc, 0x7f, 0x80, 0x03, 0xfe, 0x1f, 0xe0, 0x1f, 0xff, 0xf0, 0x3f, 0xff, 0xfc, 0x19, 0xff, 0x10, 0x3f, 0xc0, 0x03, 0xfc, 0x7f, 0x80, 0x03, 0xfe, 0x1f, 0xc0, 0x3f, 0xff, 0xf0, 0x7f, 0xff, 0xfe, 0x19, 0xfe, 0x10, 0x3f, 0xc0, 0x03, 0xfc, 0x7f, 0x80, 0x01, 0xfe, 0x3f, 0xc0, 0x7f, 0xff, 0xe0, 0x7f, 0xff, 0xfe, 0x19, 0xfe, 0x10, 0x3f, 0xc0, 0x03, 0xfc, 0x7f, 0x80, 0x01, 0xff, 0x3f, 0x80, 0xff, 0xc7, 0xc0, 0xff, 0xff, 0xff, 0x1d, 0xfe, 0x10, 0x3f, 0xc0, 0x03, 0xfc, 0x7f, 0x80, 0x00, 0xff, 0x7f, 0x80, 0xff, 0x81, 0x80, 0xff, 0xef, 0xff, 0x1d, 0xef, 0x00, 0x3f, 0xc0, 0x03, 0xfc, 0x7f, 0x80, 0x00, 0xff, 0xff, 0x00, 0xff, 0x00, 0x00, 0xff, 0xc3, 0xff, 0x8f, 0xef, 0x00, 0x3f, 0xc0, 0x03, 0xfc, 0x7f, 0x80, 0x00, 0x7f, 0xff, 0x01, 0xff, 0x00, 0x01, 0xff, 0xc3, 0xff, 0x8f, 0x87, 0x80, 0x3f, 0xc0, 0x03, 0xfc, 0x7f, 0x80, 0x00, 0x7f, 0xfe, 0x01, 0xfe, 0x00, 0x01, 0xff, 0xc1, 0xff, 0x87, 0x81, 0xc0, 0x3f, 0xc0, 0x03, 0xfc, 0x7f, 0x80, 0x00, 0x7f, 0xfe, 0x01, 0xfe, 0x1f, 0x81, 0xff, 0x81, 0xff, 0x83, 0xff, 0x80, 0x3f, 0xc0, 0x03, 0xfc, 0x7f, 0x80, 0x00, 0x3f, 0xfc, 0x01, 0xfe, 0x3f, 0xf9, 0xff, 0x81, 0xff, 0x80, 0xfe, 0x00, 0x3f, 0xc0, 0x03, 0xfc, 0x7f, 0x80, 0x00, 0x3f, 0xfc, 0x01, 0xfe, 0x3f, 0xf9, 0xff, 0x81, 0xff, 0x80, 0x00, 0x00, 0x3f, 0xc0, 0x03, 0xfc, 0x7f, 0x80, 0x00, 0x1f, 0xf8, 0x01, 0xfe, 0x3f, 0xf9, 0xff, 0x81, 0xff, 0x80, 0x00, 0x00, 0x3f, 0xc0, 0x03, 0xfc, 0x7f, 0x80, 0x00, 0x1f, 0xf0, 0x01, 0xff, 0x3f, 0xf9, 0xff, 0xc1, 0xff, 0x80, 0x00, 0x00, 0x3f, 0xc0, 0x03, 0xfc, 0x7f, 0x80, 0x00, 0x0f, 0xf0, 0x01, 0xff, 0x3f, 0xf8, 0xff, 0xc1, 0xff, 0x80, 0x00, 0x00, 0x3f, 0xc0, 0x03, 0xfc, 0x7f, 0x80, 0x00, 0x0f, 0xf0, 0x00, 0xff, 0x9f, 0xf8, 0xff, 0xc1, 0xff, 0x80, 0x00, 0x00, 0x3f, 0xc0, 0x03, 0xfc, 0x7f, 0x80, 0x00, 0x0f, 0xf0, 0x00, 0xff, 0x83, 0xf0, 0xff, 0xe1, 0xff, 0x00, 0x00, 0x00, 0x3f, 0xfc, 0x03, 0xfc, 0x7f, 0xf8, 0x00, 0x0f, 0xf0, 0x00, 0xff, 0xe3, 0xf0, 0x7f, 0xff, 0xff, 0x00, 0x00, 0x00, 0x3f, 0xff, 0xe3, 0xfc, 0x7f, 0xff, 0xc0, 0x0f, 0xf0, 0x00, 0x7f, 0xff, 0xf0, 0x7f, 0xff, 0xfe, 0x00, 0x00, 0x00, 0x3f, 0xff, 0xe3, 0xfc, 0x7f, 0xff, 0xc0, 0x0f, 0xf0, 0x00, 0x7f, 0xff, 0xf0, 0x3f, 0xff, 0xfe, 0x00, 0x00, 0x00, 0x3f, 0xff, 0xe3, 0xfc, 0x7f, 0xff, 0xc0, 0x0f, 0xf0, 0x00, 0x3f, 0xff, 0xf0, 0x3f, 0xff, 0xfc, 0x00, 0x00, 0x00, 0x3f, 0xff, 0xe3, 0xfc, 0x7f, 0xff, 0xc0, 0x0f, 0xf0, 0x00, 0x1f, 0xff, 0xf0, 0x1f, 0xff, 0xfc, 0x00, 0x00, 0x00, 0x3f, 0xff, 0xe3, 0xf8, 0x7f, 0xff, 0xc0, 0x0f, 0xf0, 0x00, 0x0f, 0xff, 0xf0, 0x0f, 0xff, 0xf8, 0x00, 0x00, 0x00, 0x1f, 0xff, 0xc3, 0xf8, 0x1f, 0xff, 0xc0, 0x0f, 0xe0, 0x00, 0x03, 0xff, 0xe0, 0x03, 0xff, 0xe0, 0x00, 0x00, 0x00, 0x00, 0x3f, 0xc0, 0xf0, 0x00, 0x3f, 0x80, 0x07, 0xe0, 0x00, 0x00, 0xff, 0x80, 0x01, 0xff, 0xc0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x3e, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00};

// FreeFonts from Adafruit_GFX
#include <Fonts/FreeMonoBold9pt7b.h>
#include <Fonts/FreeMonoBold12pt7b.h>
[... cut ...]
        display.println("SDCard  None");
    }
    display.update();

    // goto sleep
    esp_sleep_enable_ext0_wakeup((gpio_num_t)BUTTON_PIN, LOW);

    esp_deep_sleep_start();
}

void loop()
{
}
```

As far as i interprete the code the device boots up, presents static information on the screen and then goes into deep sleep for as long as the one of the buttons is not pressed. Interestingly with that image my epaper-unit presents a clear artifact-free image in deep sleep as well as when it is turned off.

Jerzeek, 2020-11-04:
I have the same problem, "stock firmware" from TTGO works fine, esphome code causes problems

chris-nite, 2020-11-05:
Wow, thank you for joining in on us @Jerzeek. I really thought, that i would be the only one with this problem.
@matikij Whatever I can do, i'm open to any sort of testing :)

Jerzeek, 2020-11-05:
yeah it has to do with full/partial updating of the screen
If I turn it off just after the page has refreshed it stays crisp, if I wait a sec and then turn it off then it fades to gray with old artifacts

Jerzeek, 2020-11-05:
[this video](https://www.youtube.com/watch?v=UsU7p2SeZJs ) shows a guy that also does a lot of partial updates and then has gray when he turns off. I also tried full update every 1 and 2 but that did not work for me

chris-nite, 2020-11-17:
Hello @matikij , were you able to reproduce these problems, or do you need further assistance? I would love to get this to work, so both displays were not bought completely in vain. I'll help with anything i can, so just let me know. Thanks in advance :-)

alexiri, 2020-11-21:
I'm having the same issue, FWIW.

matikij, 2020-11-30:
So the waveshare_epaper component is a bit messy and I tried another approach, reusing the code from https://github.com/ZinggJM/GxEPD2 which supposedly is working for more people. In the meantime I got another module which seemed to have the same issue, but less pronounced.
Anyway, I have created a branch with a new `gxepd2_epaper` component - https://github.com/matikij/esphome/tree/gxepd2_epaper . The configuration should be the same, except for the model. Valid models are corresponding to the `GxEPD2_*` files in https://github.com/matikij/esphome/tree/gxepd2_epaper/esphome/components/gxepd2_epaper directory (to get a model you strip the extension and `GxEPD2_` prefix from the filename, so `GxEPD2_290_T5.h` contains model `290_T5`). You guys probably will want to check `213_B73` and/or other models starting with `213`.

For me the new branch solves all issues I have had up to this point.

If your esphome install is in a virtual environment you can test my branch by by running:
```
pip3 install -e git+https://github.com/matikij/esphome.git@gxepd2_epaper#egg=esphome
```

For reference my simple test config:
```
---
substitutions:
  devicename: epaper_test
  upper_devicename: Epaper Display

esphome:
  name: $devicename
  platform: ESP32
  board: lolin32

logger:

spi:
  clk_pin:  18
  mosi_pin: 23

display:
  - platform: gxepd2_epaper
    cs_pin: 5
    dc_pin: 17
    reset_pin: 16
    busy_pin: 4
    model: '213_b73'
    full_update_every: 10
    update_interval: 5s
    rotation: 0 
    lambda: |-
      it.line(0, 0, 40, 40);
```
I might still make some changes to the branch (maybe I will make the models compatible with `waveshare_epaper` component for example). If changes break the example config above I will update it.

dgsardina, 2020-12-05:
Hello @matikij . I own the TTGO T2 V2.3 with (b73 too) and I have exactly the same problems you describe. I cannot use the deep sleep function without the screen decading as you show in your pictures. Similarly, if I switch off the device, the screen decays the same way. I tried using your branch as you describe (creating the venv, then pip3 install, then setting your platform gxep2_epaper and it works exactly the same way as the waveshare_epaper, so I still have the same problems :+1: 

Were you able to overcome the issues with your branch and use deep sleep? If so, could you share more info how? I truly love this device and esphome, cannot wait to make it work with the battery for more than 8h. 

Note that I tried doing full_update_every: 1 and this has the same problem, but when the device is off with the screen decayed, it is unable to "correct" it. 

In case it helps, my current config:

```
substitutions:
  device_name: "epaper"

esphome:
  name: "${device_name}"
  platform: ESP32
  board: esp32dev
  #board: lolin32
  includes:
    - icon_map.h

wifi:
  ssid: "skynet"
  password: "434343"
  
  # Only use the following if only one AP with this SSID
  # Or when connecting to hidden ssid
  fast_connect: true

logger:

api:

ota:
time:
  - platform: homeassistant
    id: esptime
    timezone: Europe/Athens

binary_sensor:
  - platform: gpio
    name: "${device_name}_sw1"
    id: epaper_sw1 
    pin: 
      number: GPIO39
      mode: INPUT_PULLUP
      inverted: True
    filters:
        - delayed_on: 5ms
    on_click:
      min_length: 50ms
      max_length: 350ms
      then:
        - if:
            condition:
              - light.is_off:
                  id: epaper_led
            then:
              - light.turn_on:
                  id: epaper_led
                  brightness: 80%
            else:
              - light.turn_off: epaper_led
              
sensor:
  - platform: uptime
    name: "${device_name} Uptime Sec"
    id: uptime_sec
    internal: true

  - platform: wifi_signal
[... 198 lines cut ...]
       // Battery %
       float battery_perc = 100.00 * (id(VCC).state - 3.5) / (4.20-3.5);
       int battery_range = battery_perc / 10 ;
       battery_range = (battery_range > 10) ? 10 : battery_range;
       battery_range = (battery_range < 0)  ?  0 : battery_range;

       it.printf( H_LEFT_MARGIN, V_BATTERY, id(font_icon_battery), TextAlign::TOP_LEFT, 
          battery_icon_map[battery_range].c_str()
          );
       it.printf(H_RIGHT_MARGIN, V_BATTERY, id(font_big), TextAlign::TOP_RIGHT, 
          "%3.0f%%", battery_perc);
       //it.line(  0,    0,   0, 249);
       //it.line( 121,   0, 121, 249);
       //it.line(   0,   0, 121,   0);
       //it.line(   0, 249, 121, 249);    

#deep_sleep:
#  run_duration: 8s
#  sleep_duration: 60s
#  wakeup_pin:
#    number: GPIO39
#    inverted: True
#  wakeup_pin_mode: IGNORE
#  id: deep_sleep_1
```

martin3000, 2020-12-12:
I have the same TTGO T5 board with 2.13in-ttgo-b73 display and my display becomes more and more dark and has artifacts. I do not use deep sleep. At https://www.waveshare.com/wiki/E-Paper_Shield they write: "Please set the e-Paper to sleep mode or power off it directly, otherwise, the e-Paper is damaged because of working in high voltage for long time."
So there seems to be a command to let the epaper sleep.

dgsardina, 2020-12-12:
Hello @martin3000 , yes, my display was damaged after playing around with esphome for around 10 days. I believe the damage is permanent, so I suggest avoid using the controller b73 with esphome until the epd is sent to sleep accordingly. 

I am doing more testing now coding directly with Arduino IDE, and I could find that the best way to use the screen and be able to put the ESP to deep sleep is with the gxepd2 library and setting display.hibernate() after refreshing. Let the following code be a reference: (Note, I am still trying to figure out if it is possible to do partial refresh of the epd display just after waking up the esp, but I am starting to believe it is not feasible).
```
void setup() {
  /* Here initialize esp and do something useful  */
 
  if (needToRefreshDisplay()) {
    refreshEpdDisplay();
  }
  espGoToSleep();
}

void refreshEpdDisplay() {
  display.init(115200);
  display.setRotation(1);
  display.setFont(&FreeMonoBold24pt7b);
  display.setTextColor(GxEPD_BLACK);
  display.setFullWindow();
  display.firstPage();
  do  {
    display.fillScreen(GxEPD_WHITE);
    display.setCursor(x, y);
    display.print(message);
  }
  while (display.nextPage());
  display.hibernate();
}

void espGoToSleep() {
  esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
  esp_deep_sleep_start();
  /* Anything after this will never be reached */
}
```

nikito7, 2020-12-12:
Lcd driver cache?

[image]

nikito7, 2020-12-12:
Gif

[image]

dgsardina, 2020-12-12:
I believe the code in esphome is not even powering off the display after updates, which makes the display burn over time. 
On the [GxEPD2 library for B23](https://github.com/ZinggJM/GxEPD2/blob/master/src/epd/GxEPD2_213_B73.cpp) at least a powerOff() would be required after every full update. And probably the code for hibernate() before sending the esp to deep sleep when required by esphome configuration. 

@nikito7 My display was doing just that with esphome. But then some values were burnt. I would suggest to stop using esphome until with this device until this is fixed.

nikito7, 2020-12-12:
But pixels still working

I'm trying "burning" all pixels

I have text in random position now, like screensaver

It was cheap :-)

martin3000, 2020-12-13:
The funny thing is: after 2 weeks, the display area is very dark. But if I put the display in bright light, the display area gets white again.

martin3000, 2020-12-13:
@matikij thank you for creating the custom component. I just had a look at the code and I could not see that the display is powered off after an update, as it is recommended by waveshare. Maybe I just overlooked it?

matikij, 2020-12-13:
Hmm, I would have thought that
https://github.com/matikij/esphome/blob/gxepd2_epaper/esphome/components/gxepd2_epaper/GxEPD2_EPD.cpp#L53
would do the trick. Perhaps I should also call hibernate() ?

matikij, 2020-12-13:
I guess hibernate() is more than just a powerOff() (at least in B73 case)
https://github.com/matikij/esphome/blob/gxepd2_epaper/esphome/components/gxepd2_epaper/GxEPD2_213_B73.cpp#L276
. OTOH, according to original GxEPD2 powerOff() should be enough
https://github.com/ZinggJM/GxEPD2/blob/master/src/GxEPD2_EPD.h#L77

Anyway, I changed the powerOff() call to hibernate() and pushed. Could you
test that ?

niedz., 13 gru 2020 o 11:42 Mateusz Kijowski <mateusz.kijowski@gmail.com>
napisał(a):

martin3000, 2020-12-13:
@matikij Looks good so far. I have no idea what hibernate and poweroff do in the display.
Do you know which of the directories in https://github.com/waveshare/e-Paper/tree/master/Arduino is the b73?
(btw: in your post there is your private email. maybe this was not intentional)

matikij, 2020-12-13:
I don't think that any of these directories matches to what is in GxEPD2, by comparing LUT tables I found that
- https://github.com/ZinggJM/GxEPD2/blob/master/src/epd/GxEPD2_213.cpp#L341 is the same as https://github.com/waveshare/e-Paper/blob/master/Arduino/epd2in13/epd2in13.cpp#L266 (so this is the "B1")
- https://github.com/ZinggJM/GxEPD2/blob/master/src/epd/GxEPD2_213_B72.cpp#L349 is the same as https://github.com/waveshare/e-Paper/blob/master/Arduino/epd2in13_V2/epd2in13_V2.cpp#L34 ("B72")
but I cannot find anything that looks like https://github.com/ZinggJM/GxEPD2/blob/master/src/epd/GxEPD2_213_B73.cpp#L350 in the waveshare repo.
Moreover there seems to be a completely different way of loading the LUT table in https://github.com/waveshare/e-Paper/blob/master/Arduino/epd2in13b_V3/epd2in13b_V3.cpp#L54 , but also the file header says "@filename   :   epd2in9b.cpp" so maybe there's something wrong with the V3 file?
The libs for these displays are in quite a mess :-)

(thanks for the heads-up regarding leaking my email)

nikito7, 2020-12-15:
> @nikito7 My display was doing just that with esphome. But then some values were burnt. I would suggest to stop using esphome until with this device until this is fixed.

Yah. Its dead now. Only two days. :-)
I buyed another, just epaper.
Esp board it's ok

martin3000, 2020-12-21:
@matikij it seems that your version also causes burn in

matikij, 2020-12-21:
@martin3000 can you provide a (redacted, but usable) yaml file that you used for testing?

martin3000, 2020-12-21:
@matikij 
https://github.com/martin3000/ESPhome/blob/master/esph-epaper2.yaml

mh255, 2020-12-26:
Hi, i had the "fading display" problem in esphome version 1.15.3 with a TTGO T5 V2.3 display.
Sending 0xC7 (instead of 0xC4) as first argument for command 0x22 (in WaveshareEPaperTypeA::display()) fixed the problem for me.

initial version:
`...`
` // COMMAND DISPLAY UPDATE CONTROL 2`
` this->command(0x22);`
` this->data(0xC4);`
`...`

working version:
`...`
` // COMMAND DISPLAY UPDATE CONTROL 2`
`  this->command(0x22);`
`  if (this->model_ == TTGO_EPAPER_2_13_IN_B73) {`
`      this->data(0xC7);`
`  }`
`  else {`
`      this->data(0xC4);`
`  }`
`...`

Mediacj, 2021-01-15:
> TTGO T5 V2.3 display.

Thanks for sharing! 

I have changed the code as mentioned in the waveshare_epaper.cpp but that didn't change the problem for me. You have changed it within the Esphome directory, or did you use Arduino for it?

Mediacj, 2021-01-16:
> Hi, i had the "fading display" problem in esphome version 1.15.3 with a TTGO T5 V2.3 display.
> Sending 0xC7 (instead of 0xC4) as first argument for command 0x22 (in WaveshareEPaperTypeA::display()) [...]
Thanks for sharing!

I have changed the code as mentioned in the waveshare_epaper.cpp but that didn't change the problem for me. You have changed it within the Esphome directory? Or did you use Arduino for it to transfer it to the ESP?

mh255, 2021-01-18:
Hi @Mediacj,
> 
> I have changed the code as mentioned in the waveshare_epaper.cpp but that didn't change the problem for me. You have changed it within the Esphome directory? Or did you use Arduino for it to transfer it to the ESP?

i changed it within the esphome directory.

I use anaconda as python distribution. The file resides in my case under "anacondahome/envs/py37_esphome/lib/python3.7/site-packages/esphome/components/waveshare_epaper/waveshare_epaper.cpp". This file is gets compiled whenever i call "esphome mydevice.yaml run" (mydevice references waveshare_epaper). 

Maybe you can test if you edited the right file by inserting a log-statement in waveshare_epaper.cpp.

Something like this:
```
void WaveshareEPaperTypeA::initialize() {
  LOG_DISPLAY("", "################## Mediacj log test ################", this);
  // COMMAND DRIVER OUTPUT CONTROL
  this->command(0x01);
  this->data(this->get_height_internal() - 1);
  this->data((this->get_height_internal() - 1) >> 8);
  this->data(0x00);  // ? GD = 0, SM = 0, TB = 0
``` 
Logging needs to be enabled to see this -> https://esphome.io/components/logger.html

Do you use esphome to flash the firmware?

Best wishes,
Michael :)

Mediacj, 2021-01-18:
Yes I use esphome to flash it.

Thanks for your reply and suggestion!

mh255, 2021-01-18:
> Yes I use esphome to flash it.
> 
> Thanks for your reply and suggestion!

To be honest: i introduced a new Panel type called "TTGO B72B" because i did not want to create a mess within the code. 
Maybe i missed something while summing things up. 

I will attach my device.yaml and the content of the directory esphome/components/waveshare_epaper, and hope this helps.

[waveshare_epaper.zip](https://github.com/esphome/issues/files/5832602/waveshare_epaper.zip)

[lilygoT5_1.zip](https://github.com/esphome/issues/files/5832608/lilygoT5_1.zip)

Mediacj, 2021-01-19:
> To be honest: i introduced a new Panel type called "TTGO B72B" because i did not want to create a mess within the code.
> Maybe i missed something while summing things up.
>
> I will attach my [...]

It think it does work for me now, I had to make the waveshare_epaper.cpp read only after changing it otherwise it would get overwritten. I don't sea artifacts anymore but thought that every full_update of the screen should be with a prior white/black blinking of the screen?

onlysli, 2021-01-22:
Hello,
Can that code be used safely without breaking the panel? Thanks.

JohFriKn, 2021-01-31:
Hello, 

i have the same problem. Are there any news?

Mediacj, 2021-01-31:
I stopped using Esphome for this display and started with MQTT from HA to the display in combination with this repro: https://github.com/G6EJD/ESP32-e-Paper-Weather-Display

matikij, 2021-02-01:
After taking a look at https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/blob/master/examples/Waveshare_2_13_T5/Waveshare_2_13_T5.ino I see that:
a) it is using the GxEPD2 library
b) that it is not using partial updates (https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/blob/master/examples/Waveshare_2_13_T5/Waveshare_2_13_T5.ino#L116)
full updates can be achieved by setting `full_update_every: 1`
perhaps it should be the default ?

nikito7, 2021-02-01:
> full updates can be achieved by setting `full_update_every: 1`
> perhaps it should be the default ?

I will kill epaper....

Just use for testing... short time

Mine died after 2 days

matikij, 2021-02-01:
Ah, true
it's there in https://github.com/esphome/esphome/blob/dev/esphome/components/waveshare_epaper/waveshare_epaper.cpp#L261
in main branch the difference is only the LUT and not the acutal commands.
I guess the only thing I can do is to change the default in my branch in the gxepd2_epaper component

nikito7, 2021-02-14:
[image]

Just to resume...

Someone have this working for more than a week?

nikito7, 2021-02-14:
So many 2.13....
https://github.com/waveshare/e-Paper/tree/master/Arduino

[image]

nikito7, 2021-02-14:
New "raw epaper" back to live with

https://github.com/lewisxhe/TTGO-EPaper-Series/blob/master/bin/T5_V2.3_2.13_GDEH0213B72.bin

Only board is original TTGO T5

matikij, 2021-02-14:
@nikito7 perhaps it's the B1? I have one of those, have a PR with support for it in https://github.com/esphome/esphome/pull/1326 . You could try verifying that it's a B1 by doing an example from GxEPD2 https://github.com/ZinggJM/GxEPD2

nikito7, 2021-02-14:
Funny stuff....
- Back alive with T5_V2.3_2.13_GDEH0213B72.bin
- work with esphome, only on first boot!

nikito7, 2021-02-14:
[image]

Screen off. Dont wake again.

Only work:
1. Flash stock
2. Powell  off
2. Flash esphome
3. Never power it off
4. Ok
5. Power it off
6. Dont wake anymore

nikito7, 2021-02-15:
I have afraid to turn it off

[image]

HINK-E0213-A22-A0 is "2.13in-ttgo"

When arrive a new B73, I will test.

Last posts are about a new "raw epaper", on an old b73 board!

nikito7, 2021-02-15:
Lut 30 waveshare 2.13in ( or b1?)
Lut 70 2.13in-ttgo

B72 0xC0 power on / b73 / b1
B72 0xC3 power off / b73 / b1
B72 0xC4 full update / b1
B72 0x04 part update / b1
B73 0xC4 part update
B73 0xC7 full update

nikito7, 2021-02-18:
ttgo-b73 (original) vs ttgo (waveshare V2 raw)

[image]

nikito7, 2021-02-19:
Testers needed

https://github.com/nikito7/esphome/archive/0_epaper_b73_fix.zip

nikito7, 2021-02-19:
b73 scroll test
[image]

Heiner1967, 2021-02-20:
Hi,

I have the same issue.
Mine is a HINK-E0213-A22-A0  SLH1914

I like to test but am unclear on how exactly.  I use ESPhome and ESPhome Beta from Hassio running on a PI. What is it now i need to do to Test?

nikito7, 2021-02-20:
I don't know
I only use ```pip install *zip```

https://github.com/nikito7/esphome/tree/0-b73-fix

nikito7, 2021-02-21:
2.13in-ttgo-b73
It's ok doing partial updates
Cool image
If you turn off it while updating will fade. It's normal...

2.13in-ttgo (v2)
Darker screen, bad contrast
Partial update not ok or usefull. Lots garbage
Recommend only do full update, 1/60s
May need my fix

Okosotthon, 2021-03-18:
I think mine is also a different version. None of the above methods work for me. 
[image]

[60 more replies]