esphome/issues #1460: esphome - add line to config causes problem with connect to wifi

vobo70, 2020-09-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

hass.io running as VM (hass.os v.4.13), Home Assistant 0.114.4
**ESP (ESP32/ESP8266, Board/Sonoff):**

sonoff basic R1
**ESPHome version (latest production, beta, dev branch)**

ESPHome version 1.14.5
**Affected component:**

**Description of problem:**
When I flash simple sonoff basic r1 esphome code
everything works - meaning sonoff connect to wifi and I can reflash it OTA
but when I add binary_sensor, switch and status_led part I have problem with connection to wifi.
**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: grzalka_zasobnik
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "voboIOT"
  password: "xxxxxxxxxxxxxxxx"

  # Ena

ble fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Grzalka Zasobnik"
    password: "xxxxxxxxxxxxxxxx"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "xxxxxxxxxxxxxxxx"

ota:

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "Sonoff Basic Button"
    on_press:
      - switch.toggle: relay

switch:
  - platform: gpio
    name: "Sonoff Basic Relay"
    pin: GPIO12
    id: relay

status_led:
  pin:
    number: GPIO13
    inverted: yes

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
Using 'COM8' as serial port.
Showing logs:
[23:13:37][D][wifi:319]: Found networks:
[23:13:37][I][wifi:365]: - 'voboIOT' (C4:AD:34:F6:D0:A8) ▂▄▆█
[23:13:37][D][wifi:366]:     Channel: 1
[23:13:37][D][wifi:367]:     RSSI: -66 dB
[23:13:37][I][wifi:365]: - 'voboIOT' (C4:AD:34:F6:E9:E4) ▂▄▆█
[23:13:37][D][wifi:366]:     Channel: 9
[23:13:37][D][wifi:367]:     RSSI: -65 dB
[23:13:37][D][wifi:369]: - 'Grzalka Zasobnik' (6A:C6:3A:81:06:D5) ▂▄▆█
[23:13:37][D][wifi:369]: - 'WLAN1-4JJNNM' (B4:86:55:87:D4:FA) ▂▄▆█
[23:13:37][I][wifi:193]: WiFi Connecting to 'voboIOT'...
[23:13:43][W][wifi_esp8266:391]: Event: Disconnected ssid='voboIOT' reason='Probe Request Unsuccessful'
[23:13:43][W][wifi:451]: Error while connecting to network.
[23:13:43][I][wifi:193]: WiFi Connecting to 'voboIOT'...
[23:13:49][W][wifi_esp8266:391]: Event: Disconnected ssid='voboIOT' reason='Probe Request Unsuccessful'
[23:13:49][W][wifi:451]: Error while connecting to network.
[23:13:54][D][wifi:304]: Starting scan...
[23:14:00][D][wifi:319]: Found networks:
[23:14:00][I][wifi:365]: - 'voboIOT' (C4:AD:34:F6:E9:E4) ▂▄▆█
[23:14:00][D][wifi:366]:     Channel: 9
[23:14:00][D][wifi:367]:     RSSI: -64 dB
[23:14:00][I][wifi:365]: - 'voboIOT' (C4:AD:34:F6:D0:A8) ▂▄▆█
[23:14:00][D][wifi:366]:     Channel: 1
[23:14:00][D][wifi:367]:     RSSI: -65 dB
[23:14:00][D][wifi:369]: - 'Grzalka Zasobnik' (6A:C6:3A:81:06:D5) ▂▄▆█
[23:14:00][D][wifi:369]: - 'WLAN1-4JJNNM' (B4:86:55:87:D4:FA) ▂▄▆█
[23:14:00][I][wifi:193]: WiFi Connecting to 'voboIOT'...
[23:14:06][W][wifi_esp8266:391]: Event: Disconnected ssid='voboIOT' reason='Probe Request Unsuccessful'
[23:14:06][W][wifi:451]: Error while connecting to network.
[23:14:06][I][wifi:193]: WiFi Connecting to 'voboIOT'...
[23:14:12][W][wifi_esp8266:391]: Event: Disconnected ssid='voboIOT' reason='Probe Request Unsuccessful'
[23:14:12][W][wifi:451]: Error while connecting to network.

```

**Additional information and things you've tried:**

drward, 2020-09-14:
I'm seeing the same with Sensor on an ESP32.

sensor:
  - platform: dht
    pin: GPIO23
    temperature:
      name: "Media Center Temperature"
    humidity:
      name: "Media Center Humidity"
    update_interval: 60s
    model: AM2302

vobo70, 2020-09-14:
sorry, I was wrong;
no luck
this part breaks wifi connection:

`binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "pompa recyrkulacja przycisk"
    on_press:
      - switch.toggle: relay_pompa_recyrkulacja`

smoke007, 2020-09-21:
Disabling logger seems to help with wifi issues on Sonoff Basics.  I've fought with wifi breaking over the last many version updates and there have been many long issues reported on this subject.  My Sonoff Basics seem to make the version upgrades best with logger disabled since that seems to be too much for them to handle.

Here's an example of one of my Sonoff Basics with the DHT22 attached and functioning as a thermostat to turn on/off torpedo heater.

`esphome:
  name: pole_barn_heater
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: !secret wifi_name
  password: !secret wifi_password
  reboot_timeout: 1h
  fast_connect: false
  domain: .lan

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "PoleBarn Heater Fallback Hotspot"
    password: !secret ap_password

captive_portal:

# Enable logging
#logger:

# Enable Home Assistant API
api:
  password: !secret api_password
  reboot_timeout: 12h

ota:
  password: !secret ota_password

switch:
  - platform: gpio
    name: "Pole Barn Heater"
    pin: GPIO12
    id: relay

binary_sensor:
  - platform: gpio
    name: "Pole Barn Heater Switch"
    pin:
      number: GPIO14
      mode: INPUT_PULLUP
    on_press:
      - switch.toggle: relay
    on_release:
      - switch.toggle: relay

  - platform: gpio
    name: "Pole Barn Heater Button"
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    on_press:
      - switch.toggle: relay

status_led:
  pin:
    number: GPIO13
    inverted: yes

sensor:
  - platform: dht
    pin:
      number: GPIO03
    model: DHT22
    temperature:
      name: "Pole Barn Heater Temperature"
      filters:
        - lambda: return x * (9.0/5.0) + 32.0;
      unit_of_measurement: "°F"
      on_value_range:
        - below: 57.0
          then:
            - switch.turn_on: relay
        - above: 59.0
          then:
            - switch.turn_off: relay
    humidity:
      name: "Pole Barn Heater Humidity"
    update_interval: 60s`

vobo70, 2020-09-21:
upgrade to 1.15.2 soilves the problem