esphome/issues #1462: ePaper 2.9 inch timeout while displaying image

baerni, 2020-09-12:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP8266 NodeMCU

**ESPHome version (latest production, beta, dev branch)**

latest production

**Affected component:**

[Waveshare E-Paper Display](https://esphome.io/components/display/waveshare_epaper.html)

**Description of problem:**
I have a problem with the e-paper display. I connected everything as in the attached code snippet, but I have this information in the logs: _timeout while displaying image_. Should I change any pin, pin mode or inverted? When I disconnected the optional Busy and Reset pins and commented it in the code, there were no timeout errors, but nothing was displayed on the display.

**Problem-relevant YAML-configuration entries:**
```yaml
...

spi:
  clk_pin: D5  
  mosi_pin: D7

display:
  - platform: waveshare_epaper
    cs_pin: D8
    dc_pin: D6
    busy_pin: D2
    reset_pin: D1
    model: 2.90in
    full_update_every: 10
    update_interval: 5s
    rotation: 90
    lambda: |-
      it.filled_rectangle(50, 60, 30, 42);

```

damanti-me, 2020-09-21:
Hi, for me it helped to define the MOSO pin as well in the config, even tough there is no one on the Displayboard. Then the ESP will use the external SPI. 
Additionaly i used the other SPI pins on the ESP like: 
```yaml
.....
spi:
  clk_pin: GPIO14 #yellow
  mosi_pin: GPIO13 #blue
  miso_pin: GPIO12 #not connected for using external SPI

#Dimensions: 128px x 296px
display:
  - platform: waveshare_epaper
    cs_pin: GPIO15 #orange
    dc_pin: GPIO05 #green
    busy_pin: GPIO00 #purple
    reset_pin: GPIO04 #white
    model: 2.90in
    id: display_update
    update_interval: 30s
    full_update_every: 60
    rotation: 90°
    lambda: |-
      it.print(40, 20, id(my_font), "Temperatur outside");
  ........

```

closed 2021-02-02 by stale[bot]

dougle03, 2021-08-19:
> logs: _timeout while displaying image_.

Hi, did you get anywhere with this? 
I'm getting a flashing screen alternating between noise and the actual display needed...
Pulling my hair out now..

a-marcel, 2021-12-02:
Same here, only noise and the error message `Timeout while displaying image!``
Any news for this topic ?
Thanks

minimag450, 2022-01-27:
I had the same problem with a 1.54inv2. I deleted 

'update_interval: 5s' 

and inserted the 

'full_update_every: 180' 

like this. It works for me.

Can you try it with you and write if it worked?
Greetings, Andreas