esphome/issues #1464: fastled_spi component with P9813 driver board

JonasDoebertin, 2020-09-13:
I'm trying to use [this LED driver board](https://www.ebay.de/itm/272682254000) based on a P9813 with a NodeMCU board and a "dumb" LED strip. However, I can only get the LED strip to flicker while changing the lights state in HA, otherwise it doesn't light up.

The controller is powered by a 12 volt/6 amps power supply.

The NodeMCU is currently powered over USB, as I can't get the VIN output of the driver board to work properly (can measure an output of roughly 0.2 volts, from my understanding it should provide 4.5 - 5.5 volts though). When connected to VIN of the NodeMCU it doesn't power up, but that's not the real issue right now.

The data and clock lines are connected (data = D6 <> DIN, clock = D5 <> CIN).

I also have established a common ground (ground of NodeMCU connected to ground of driver board).

The light get's registered correctly in Home Assistant. If I change its state (color, brightness, on/off) the strip will light up and start flickering (all colors simultaneously) for about 1s (seems to correspond to the transition time of the light). 

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESPHome CLI (pip) on macOS 10.15.6

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 NodeMCU

**ESPHome version (latest production, beta, dev branch)**

1.14.5

**Affected component:**

fastled_spi

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
light:
- platform: fastled_spi
  chipset: P9813
  data_pin: D6
  clock_pin: D5
  num_leds: 1
  rgb_order: RGB
  name: ${friendly_name} LED Strip
```

**Logs (if applicable):**

```
...
[12:29:27][C][fastled:010]: Setting up FastLED light...
[12:29:27][C][light:097]: Setting up light 'Garderobe LED Strip'...
[12:29:27][D][light:264]: 'Garderobe LED Strip' Setting:
[12:29:27][D][light:277]:   Brightness: 100%
[12:29:27][D][light:286]:   Red=100%, Green=100%, Blue=100%
...
[12:29:35][C][fastled:019]: FastLED light:
[12:29:35][C][fastled:020]:   Num LEDs: 1
[12:29:35][C][fastled:021]:   Max refresh rate: 16000
...
[12:31:59][D][light:264]: 'Garderobe LED Strip' Setting:
[12:31:59][D][light:273]:   State: ON
[12:31:59][D][light:303]:   Transition Length: 1.0s
```

**Additional information and things you've tried:**

* Changing to other pins (D1 & D2)
* Switching up data & clock wires

demikl, 2020-09-15:
Hi, I'm using the exact same LED driver board. I already bumped into this issue, but more interestingly, I've also managed to get it working, even in a chain (multiple driver boards on the same data+clock lines, with num_leds > 1).

Let me check how my wiring is right now. As far as I remember, this issue was related to the voltage injected into the VIN pin of the driver board. Maybe 3.3V, but let me check next time I'll be near this setup.

JonasDoebertin, 2020-09-15:
Would be very interested in your solution and wiring @demikl 👍

demikl, 2020-09-17:
The driver board is connected to a 3.3V pin of my ESP8266 nodeMCU board, with DIN (data-in) connected to D6 (GPIO12) and CIN (clock-in) connected to D7 (GPIO13):
[image]

You can find my esphome YAML configuration file here: https://github.com/demikl/Switch433Control/blob/master/ESPHome/ledstrip.yaml#L103

closed 2021-01-29 by stale[bot]