esphome/issues #1465: SHT3X-D needs to report and control heater status

Drizzt321, 2020-09-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io Home-Assistant, 0.114.4 from the QCOW VM image, ESPHome 1.14.4 from HA Integrations.

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266, Wemos D1 mini

**ESPHome version (latest production, beta, dev branch)**

ESPHome 1.14.4 from HA Integrations

**Affected component:**

SHT3X-D temperature/humidity i2c sensor
https://esphome.io/components/sensor/sht3xd.html

**Description of problem:**

So I got some cheap SHT31-D boards from Amazon, and they are reporting temperatures quite a bit above what I would expect from other thermometers I have. I've also compared using a Adafruit SHT31-D board + Arduino with their sample code, with the heater off. Looks like the cheap ones I got start off with the heater on. Since ESPHome code doesn't turn it off to start with, it reports significantly higher temps than it should.

For example, the Adafruit Huzzah with these cheapo SHT31-D is reporting ~76F and ~57% humidity, one of the D1 mini with the Adafruit SHT31-D and ESPHome is reporting ~76F and 57% humidity. A separate D1 mini with the cheapo SHT31-D is reporting ~84F and 45% humidity. This SHT31-D is fresh, hasn't been set to be heater off via the Arduino code, so I presume the default is heater set to on. 

I think the SHT3X-D code should expose the Heater status (On/Off), and provide a control to turn the heater on and off. This lets HA script on/off the heater in a controlled fashion to evaporate/help remove condensation that may build up. 

**Problem-relevant YAML-configuration entries:**
```yaml
i2c:
  sda: 4
  scl: 5
  scan: True
  id: bus_a
sensor:
  - platform: sht3xd
    temperature:
      name: "Kitchen Room Temperature"
    humidity:
      name: "Kitchen Room Humidity"
    address: 0x44
    update_interval: 60s
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

eddyg, 2021-01-12:
I just came here to open an issue and decided to do a search for `sht3x` first and found this issue. (Just in time, too, based on the above comment. Hopefully it won't go stale before getting resolved!)

## Background

I recently set up some ESP8266 D1 Mini clones to act as temperature sensors. I was using [these BME280 sensors modules from Amazon](https://smile.amazon.com/gp/product/B07KR24P6P). **The readings I was getting with ESPHoe were much higher than "reality".**

Thinking maybe the BME280 sensors I received were somehow "bad", I decided to try a [DFRobot SHT31 sensor module](https://www.dfrobot.com/product-2013.html). I also ordered a Sensirion [SHT31 Smart Gadget](https://developer.sensirion.com/platforms/sht31-smart-gadget-development-kit/) development kit that uses the same SHT31 for use as a reference thermometer.

Unfortunately after adding the `sht3x` platform to my .yaml file, **I still didn't get any better temperatures** for the SHT31 sensor. So I'm looking for suggestions on how to improve the sensor readings.

## Sample Temperature Readings

For reference, here are some sample readings:
```
[11:48:15][C][bme280.sensor:142]: BME280:
[11:48:15][C][bme280.sensor:143]:   Address: 0x76
[11:48:15][C][bme280.sensor:156]:   Update Interval: 60.0s
[11:48:15][C][bme280.sensor:158]:   Temperature 'Upstairs Temperature Alt'
[11:48:15][C][bme280.sensor:158]:     Unit of Measurement: '°C'
[11:48:15][C][bme280.sensor:158]:     Accuracy Decimals: 1
[11:48:15][C][bme280.sensor:158]:     Icon: 'mdi:thermometer'
[11:48:15][C][bme280.sensor:160]:   Pressure 'Upstairs Pressure'
[11:48:15][C][bme280.sensor:160]:     Unit of Measurement: 'hPa'
[11:48:15][C][bme280.sensor:160]:     Accuracy Decimals: 1
[11:48:15][C][bme280.sensor:160]:     Icon: 'mdi:gauge'
[11:48:15][C][bme280.sensor:162]:   Humidity 'Upstairs Humidity Alt'
[11:48:15][C][bme280.sensor:162]:     Unit of Measurement: '%'
[11:48:15][C][bme280.sensor:162]:     Accuracy Decimals: 1
[11:48:15][C][bme280.sensor:162]:     Icon: 'mdi:water-percent'
[11:48:15][C][sht3xd:034]: SHT3xD:
[11:48:15][C][sht3xd:035]:   Address: 0x45
[11:48:15][C][sht3xd:039]:   Update Interval: 60.0s
[11:48:15][C][sht3xd:041]:   Temperature 'Upstairs Temperature'
[11:48:15][C][sht3xd:041]:     Unit of Measurement: '°C'
[11:48:15][C][sht3xd:041]:     Accuracy Decimals: 1
[11:48:15][C][sht3xd:041]:     Icon: 'mdi:thermometer'
[11:48:15][C][sht3xd:042]:   Humidity 'Upstairs Humidity'
[11:48:15][C][sht3xd:042]:     Unit of Measurement: '%'
[11:48:15][C][sht3xd:042]:     Accuracy Decimals: 1
[11:48:15][C][sht3xd:042]:     Icon: 'mdi:water-percent'
[11:48:15][C][ota:029]: Over-The-Air Updates:
[11:48:15][C][ota:030]:   Address: tempsensor.local:8266
[11:48:15][C][api:095]: API Server:
[11:48:15][C][api:096]:   Address: tempsensor.local:6053
[11:48:27][D][sht3xd:065]:        Got temperature=25.07°C humidity=34.59%
[11:49:15][D][bme280.sensor:197]: Got temperature=24.2°C pressure=981.2hPa humidity=29.4%
[11:49:27][D][sht3xd:065]:        Got temperature=25.01°C humidity=34.78%
[11:50:15][D][bme280.sensor:197]: Got temperature=24.2°C pressure=981.2hPa humidity=29.5%
[11:50:27][D][sht3xd:065]:        Got temperature=25.03°C humidity=34.69%
[11:51:15][D][bme280.sensor:197]: Got temperature=24.2°C pressure=981.2hPa humidity=29.5%
[11:51:27][D][sht3xd:065]:        Got temperature=25.08°C humidity=34.62%
[11:52:15][D][bme280.sensor:197]: Got temperature=24.2°C pressure=981.2hPa humidity=29.5%
...
[12:16:15][D][bme280.sensor:197]: Got temperature=24.1°C pressure=981.1hPa humidity=29.3%
[12:16:27][D][sht3xd:065]:        Got temperature=25.08°C humidity=34.50%
[12:17:15][D][bme280.sensor:197]: Got temperature=24.1°C pressure=981.1hPa humidity=29.3%
[12:17:27][D][sht3xd:065]:        Got temperature=24.91°C humidity=34.87%
[12:18:15][D][bme280.sensor:197]: Got temperature=24.1°C pressure=981.1hPa humidity=29.5%
[12:18:27][D][sht3xd:065]:        Got temperature=24.91°C humidity=34.79%
[12:19:15][D][bme280.sensor:197]: Got temperature=24.1°C pressure=981.1hPa humidity=29.7%
[12:19:27][D][sht3xd:065]:        Got temperature=25.07°C humidity=34.76%
```

## Reference photos

During this time, I had the SHT31 Smart Gadget _("which demonstrates the outstanding performance of Sensirion's sensors")_ next to the D1 board, as shown here, so I could compare the readings from ESPHome to the :
[image]
**72.1°F == 22.3°C** and as you can see above, ESPHome is reading **25°C** even though the two SHT31 sensors are just centimeters apart!

In this photo I am showing that both the BME280 and SHT31 sensor modules on the D1 Mini board have plenty of clearance around them, and are _not_ near anything that would be causing them to "warm up" (I just realized this image is blurry, focus is on the wrong thing, but good enough to show the sensor module clearances):
[image]

I am looking for suggestions on how to improve the accuracy of either or both of these sensors in ESPHome. Thanks!

Drizzt321, 2021-01-12:
@eddyg I suspect the difference on the sht31x board _may_ be related to the heater being on. I'd look at using the [Adafruit sample/library ](https://learn.adafruit.com/adafruit-sht31-d-temperature-and-humidity-sensor-breakout/wiring-and-test) and use that to turn off the sht3x heater (which is used to burn off moisture/condensation, should not be on always) and then re-try with ESPHome. Might just do the trick. Then again, might not, the heater might already be off.

eddyg, 2021-05-13:
Commenting to keep this issue from going stale.

My temperature readings are still considerably higher than reality.

eddyg, 2021-09-10:
The original issue is still valid; the SHT3X-D code should be updated to expose the heater status.

closed 2021-09-24 by stale[bot]