esphome/issues #1466: Time component ignores TZ on ESP8266

OneArmedBand1t, 2020-09-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266, Wemos D1 mini
**ESPHome version (latest production, beta, dev branch)**

v1.15.0
**Affected component:**

Time component
https://www.esphome.io/components/time.html

**Description of problem:**
On ESP8266 the Timezone is ignored using v1.15.0. It worked fine in v1.14.5
It is likely caused by commit 4c3c14e which attempted to fix bug #1231 (https://github.com/esphome/esphome/pull/1231)

Tested it with time platform "sntp" as well as with platform "homeassistant"

**Board info:**
_Processing d1_mini_03 (board: d1_mini; framework: arduino; platform: espressif8266@2.2.3)_

**Problem-relevant YAML-configuration entries:**

**Example timezone config**
```yaml
time:
  - platform: sntp
    id: esptime
    timezone: Europe/Amsterdam
```

**Logs (if applicable):**

**1.14.5 Behaviour** (Correct)
```
[12:17:15][C][sntp:043]: SNTP Time:
[12:17:15][C][sntp:044]:   Server 1: '0.pool.ntp.org'
[12:17:15][C][sntp:045]:   Server 2: '1.pool.ntp.org'
[12:17:15][C][sntp:046]:   Server 3: '2.pool.ntp.org'
[12:17:15][C][sntp:047]:   Timezone: 'CET-1CEST-2,M3.4.0/2,M10.5.0/3'
...
[12:18:41][D][sntp:059]: Synchronized time: Mon Sep 14 14:18:41 2020
```

**1.15.0 Behaviour** (Incorrect)
```
[12:57:35][C][sntp:043]: SNTP Time:
[12:57:35][C][sntp:044]:   Server 1: '0.pool.ntp.org'
[12:57:35][C][sntp:045]:   Server 2: '1.pool.ntp.org'
[12:57:35][C][sntp:046]:   Server 3: '2.pool.ntp.org'
[12:57:35][C][sntp:047]:   Timezone: 'CET-1CEST-2,M3.4.0/2,M10.5.0/3'
...
[12:59:17][D][sntp:059]: Synchronized time: Mon Sep 14 12:59:17 2020
```

**Additional information and things you've tried:**

eashwarvijay, 2020-09-15:
Having same problem

glmnet, 2020-09-15:
Please try version 1.15.1

The sntp timezone issue should be fixed.

eashwarvijay, 2020-09-15:
THANK YOU 
WORKING AS BEFORE

OneArmedBand1t, 2020-09-15:
Confirmed to be fixed and working in 1.15.1

Thank you!

closed 2020-09-15 by OneArmedBand1t