esphome/issues #1467: PZEM-004T updates multiple times in rapid succession instead of once

Fusseldieb, 2020-09-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant running in Docker on Ubuntu 18.04.4 LTS (irrelevant)

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266

**ESPHome version (latest production, beta, dev branch)**

1.15.0

**Affected component:**

PZEM-004T 10A/100A

**Description of problem:**
I have 3 PZEM sensors hooked up on one ESP and all three get updated on different intervals and the data is sent in rapid succession to Home Assistant. Why can't it wait to "loop" over all sensors first and only **then** pass the data over to Home Assistant? Since I'm adding all three Wattages together in Home Assistant, the Wattage changes 3 times in rapid succession and the log graph gets 3 times bigger than it should.

**Problem-relevant YAML-configuration entries:**
```yaml
sensor:
  - platform: pzemac
    voltage:
      name: "Phase 1 Voltage"
    power:
      name: "Phase 1 Power"
    update_interval: 5s
    address: 1
    
  - platform: pzemac
    voltage:
      name: "Phase 2 Voltage"
    power:
      name: "Phase 2 Power"
    update_interval: 5s
    address: 2
    
  - platform: pzemac
    voltage:
      name: "Phase 3 Voltage"
    power:
      name: "Phase 3 Power"
    update_interval: 5s
    address: 3
```

**Logs (if applicable):**

```
[11:25:55][D][pzemac:049]: PZEM AC: V=218.3 V, I=0.616 A, P=58.6 W, E=1516172.0 Wh, F=60.0 Hz, PF=0.44
[11:25:55][D][sensor:092]: 'Phase 3 Voltage': Sending state 218.30000 V with 1 decimals of accuracy
[11:25:55][D][sensor:092]: 'Phase 3 Power': Sending state 58.60000 W with 1 decimals of accuracy
[11:25:55][D][pzemac:049]: PZEM AC: V=220.7 V, I=2.812 A, P=585.1 W, E=982670.0 Wh, F=60.0 Hz, PF=0.94
[11:25:55][D][sensor:092]: 'Phase 1 Voltage': Sending state 220.70000 V with 1 decimals of accuracy
[11:25:55][D][sensor:092]: 'Phase 1 Power': Sending state 585.09998 W with 1 decimals of accuracy
[11:25:55][D][pzemac:049]: PZEM AC: V=220.8 V, I=2.976 A, P=493.1 W, E=1862496.0 Wh, F=60.0 Hz, PF=0.75
[11:25:55][D][sensor:092]: 'Phase 2 Voltage': Sending state 220.80000 V with 1 decimals of accuracy
[11:25:55][D][sensor:092]: 'Phase 2 Power': Sending state 493.10001 W with 1 decimals of accuracy
[11:26:00][D][pzemac:049]: PZEM AC: V=218.5 V, I=0.602 A, P=55.7 W, E=1516172.0 Wh, F=60.0 Hz, PF=0.42
[11:26:00][D][sensor:092]: 'Phase 3 Voltage': Sending state 218.50000 V with 1 decimals of accuracy
[11:26:00][D][sensor:092]: 'Phase 3 Power': Sending state 55.70000 W with 1 decimals of accuracy
[11:26:00][D][pzemac:049]: PZEM AC: V=220.3 V, I=4.318 A, P=837.4 W, E=982671.0 Wh, F=60.0 Hz, PF=0.88
[11:26:00][D][sensor:092]: 'Phase 1 Voltage': Sending state 220.30000 V with 1 decimals of accuracy
[11:26:00][D][sensor:092]: 'Phase 1 Power': Sending state 837.40002 W with 1 decimals of accuracy
[11:26:00][D][pzemac:049]: PZEM AC: V=220.8 V, I=2.975 A, P=492.7 W, E=1862496.0 Wh, F=60.0 Hz, PF=0.75
[11:26:00][D][sensor:092]: 'Phase 2 Voltage': Sending state 220.80000 V with 1 decimals of accuracy
[11:26:00][D][sensor:092]: 'Phase 2 Power': Sending state 492.70001 W with 1 decimals of accuracy
```

**Additional information and things you've tried:**
\-

Would be nice if this could be implemented in the next version :)

Fusseldieb, 2021-01-12:
Don't stale, issue is still there

Fusseldieb, 2021-05-20:
Well, I settled with a workaround for now:

```
uart:
  rx_pin: D5
  tx_pin: D4
  baud_rate: 9600

sensor:
  - platform: template
    name: "Threephase Energy Sensor"
    lambda: |-
      int all_phases = id(p1).state+id(p2).state+id(p3).state;
      if (all_phases < 100000) {
        return all_phases;
      } else {
        return {};
      }
    unit_of_measurement: "W"
    accuracy_decimals: 0
    update_interval: 3s

  - platform: pzemac
    voltage:
      name: "Phase 1 Voltage"
    power:
      name: "Phase 1 Power"
      id: p1
      internal: true
    update_interval: 1s
    address: 1

  - platform: pzemac
    voltage:
      name: "Phase 2 Voltage"
    power:
      name: "Phase 2 Power"
      id: p2
      internal: true
    update_interval: 1.03s // Slightly out of sync - If all have the same interval, they give read errors more often!
    address: 2
  - platform: pzemac
    voltage:
      name: "Phase 3 Voltage"
    power:
      name: "Phase 3 Power"
      id: p3
      internal: true
    update_interval: 1.05s // Slightly out of sync
    address: 3
```

It basically makes a virtual sensor that combines all three values and sends the value to Home Assistant.

NikonNLG, 2021-08-15:
@Fusseldieb just in case, can you please show me whole config for 3 PZEMs?
I'm trying to setup same configuration, but only 2 of 3 is working at the same time. My config looks like this:

```
uart:
  - tx_pin: D1
    rx_pin: D2
    baud_rate: 9600
    id: uart_phase_1
  - tx_pin: D3
    rx_pin: D4
    baud_rate: 9600
    id: uart_phase_2
  - tx_pin: D5
    rx_pin: D6
    baud_rate: 9600
    id: uart_phase_3

modbus: 
  - id: modbus_phase_1
    uart_id: uart_phase_1
  - id: modbus_phase_2
    uart_id: uart_phase_2
  - id: modbus_phase_3
    uart_id: uart_phase_3

sensor:
  - platform: pzemac
    current:
      name: "Phase 1 Current"
    voltage:
      name: "Phase 1 Voltage"
    power:
      name: "Phase 1 Power"
    update_interval: 10s
    modbus_id: modbus_phase_1
    id: pzemac1

  - platform: pzemac
    current:
      name: "Phase 2 Current"
    voltage:
      name: "Phase 2 Voltage"
    power:
      name: "Phase 2 Power"
    update_interval: 11s
    modbus_id: modbus_phase_2
    id: pzemac2

  - platform: pzemac
    current:
      name: "Phase 3 Current"
    voltage:
      name: "Phase 3 Voltage"
    power:
      name: "Phase 3 Power"
    update_interval: 12s
    modbus_id: modbus_phase_3
    id: pzemac3
```
And getting values only for phase_2 and phase_3, or phase_1 and phase_2, but not all 3 at the same time.