esphome/issues #1470: ruuvitag component doesn't work

marylein, 2020-09-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HassOS 4.13, Supervisor Version 243, ESPHome Addon version: 1.15.1

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32, Generic, board: esp-wrover-kit

**ESPHome version (latest production, beta, dev branch)**

ESPHome Addon version: 1.15.1

**Affected component:**

ruuvitag
https://esphome.io/components/sensor/ruuvitag.html

**Description of problem:**
Since I added the ruuvitag plattform to my yaml I can't get my esp32 to start. I hope the log offers enough information as I'm very new and unexperienced regarding esphome.

**Problem-relevant YAML-configuration entries:**
```yaml
sensor:
  - platform: ruuvitag
    mac_address: 'DA:E5:2F:F7:88:5B'
    humidity:
      name: "ruuvi outside front humidity"
    temperature:
      name: "ruuvi outside front temperature"
    pressure:
      name: "ruuvi outside front pressure"
```

**Logs (if applicable):**

```
[12:15:42]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[12:15:42]Core 1 register dump:
[12:15:42]PC      : 0x400d8231  PS      : 0x00060130  A0      : 0x800d86a1  A1      : 0x3ffce780  
WARNING Decoded 0x400d8231: esphome::ruuvi_ble::parse_ruuvi_data_byte(std::vector<unsigned char, std::allocator<unsigned char> > const&, esphome::ruuvi_ble::RuuviParseResult&) at /config/esphome/esp32_livingroom/src/esphome/core/optional.h:98
[12:15:43]A2      : 0x3fff1158  A3      : 0x3ffce7e8  A4      : 0x00000000  A5      : 0x3ffce8a4  
[12:15:43]A6      : 0x0000000b  A7      : 0x00000000  A8      : 0x00000412  A9      : 0x00000001  
[12:15:43]A10     : 0x00000004  A11     : 0x3ffce8a4  A12     : 0x00000010  A13     : 0x3fff1164  
[12:15:43]A14     : 0x3fff1164  A15     : 0x00000000  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
[12:15:43]EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
[12:15:43]
[12:15:43]Backtrace: 0x400d8231:0x3ffce780 0x400d869e:0x3ffce7e0 0x400d8702:0x3ffce860 0x400d71d7:0x3ffce900 0x401e0bb5:0x3ffce9a0 0x401e0c49:0x3ffce9c0 0x400dc94e:0x3ffce9e0 0x400dea5a:0x3ffcea10 0x400ea419:0x3ffcea30 0x4008e7dd:0x3ffcea50
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x400d8231: esphome::ruuvi_ble::parse_ruuvi_data_byte(std::vector<unsigned char, std::allocator<unsigned char> > const&, esphome::ruuvi_ble::RuuviParseResult&) at /config/esphome/esp32_livingroom/src/esphome/core/optional.h:98
WARNING Decoded 0x400d869e: esphome::ruuvi_ble::parse_ruuvi(esphome::esp32_ble_tracker::ESPBTDevice const&) at /config/esphome/esp32_livingroom/src/esphome/core/optional.h:98
WARNING Decoded 0x400d8702: esphome::ruuvi_ble::RuuviListener::parse_device(esphome::esp32_ble_tracker::ESPBTDevice const&) at /config/esphome/esp32_livingroom/src/esphome/core/optional.h:98
WARNING Decoded 0x400d71d7: esphome::esp32_ble_tracker::ESP32BLETracker::loop() at /config/esphome/esp32_livingroom/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp:241
WARNING Decoded 0x401e0bb5: esphome::Component::call_loop() at /config/esphome/esp32_livingroom/src/esphome/core/component.cpp:156
WARNING Decoded 0x401e0c49: esphome::Component::call() at /config/esphome/esp32_livingroom/src/esphome/core/component.cpp:156
WARNING Decoded 0x400dc94e: esphome::Application::loop() at /config/esphome/esp32_livingroom/src/esphome/core/application.cpp:72
WARNING Decoded 0x400dea5a: loop() at /config/esphome/esp32_livingroom/src/main.cpp:357
WARNING Decoded 0x400ea419: loopTask(void*) at /root/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:19
WARNING Decoded 0x4008e7dd: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
[12:15:43]
[12:15:43]Rebooting...
```

**Additional information and things you've tried:**

glmnet, 2020-09-16:
Seems like a code bug
cc @Alex9779

Alex9779, 2020-09-16:
I am using this component since I wrote it...
But I didn't upgrade to 1.15.x yet, will look into it...

Alex9779, 2020-09-16:
I did some tests with three different ESP32 boards with 1.15.1, but I don't get the error, works perfectly fine for me, I get the readings.
I used a very bare config and one with multiple sensors wifi and a custom component, maybe something interferes? @marylein can you post a full config for you device?

marylein, 2020-09-16:
I'm using [this board](https://www.amazon.de/gp/product/B07Z83MF5W/ ).
I decided to use the "Gerneric ESP32" in the ESPHome Wizard to create the following configuration. Not sure if this was the best decision.

```
esphome:
  name: esp_home
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: "..."
  password: "..."
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.178.37
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.178.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0
    
# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "..."

ota:
  password: "..."
  
esp32_ble_tracker:
  scan_parameters:
    active: false

sensor:
  - platform: xiaomi_hhccjcy01
    mac_address: 'C4:7C:8D:...'
    temperature:
      name: "Findus temperatur"
    moisture:
      name: "Findus moisture"
    illuminance:
      name: "Findus light intensity"
    conductivity:
      name: "Findus conductivity"
  - platform: xiaomi_hhccjcy01
    mac_address: 'C4:7C:8D:...'
    temperature:
      name: "Ben temperatur"
    moisture:
      name: "Ben moisture"
    illuminance:
      name: "Ben light intensity"
    conductivity:
      name: "Ben conductivity"
  - platform: xiaomi_hhccjcy01
    mac_address: 'C4:7C:8D:...'
    temperature:
      name: "Scott temperatur"
    moisture:
      name: "Scott moisture"
    illuminance:
      name: "Scott light intensity"
    conductivity:
      name: "Scott conductivity"
  - platform: ruuvitag
    mac_address: 'DA:E5:2F:...'
    humidity:
      name: "ruuvi outside front humidity"
    temperature:
      name: "ruuvi outside front temperature"
    pressure:
      name: "ruuvi outside front pressure"
```

glmnet, 2020-09-16:
@marylein setup you logger component to log very verbose, then capture logs again, may be there shows up some information that can be used to track the issue.

Alex9779, 2020-09-16:
Just two things different from my config is that I use a different board and the latest Arduino framework:
```
arduino_version: latest
board: esp32dev
```

marylein, 2020-09-16:
I just changed my board to `esp32dev` and added `arduino_version: latest` and now it's working... Thank you so much!

Alex9779, 2020-09-16:
Nice, so I am still curious which config change did the trick, can you change your board back and see if it fails then?
I assume not, IMHO the framework is the cause but would be nice to be sure...

marylein, 2020-09-17:
I tried it again with `board: esp-wrover-kit` and as you suspected, it still works!

Alex9779, 2020-09-17:
Ok, thanks...
@glmnet what now? I had some troubles long ago I can't remeber atm what exactly but that was when I introduced using the lastest arduino version in all of my device configs and apparently the default used by esphome has an issue...

glmnet, 2020-09-17:
Well, if it's working for @marylein and it's not reproducible anymore, we can close this one.

Btw, consider adding yourself to `CODEOWNERS` in this component, like this https://github.com/esphome/esphome/blob/8215a018e97c8506ffe08c913580514b8faf89a8/esphome/components/climate_ir/__init__.py#L9

Thanks!

closed 2020-09-17 by glmnet

Alex9779, 2020-09-17:
> and it's not reproducible anymore

well it still is, if he uses not the latest arduino framework explicitly... but ok...

> Btw, consider adding yourself to `CODEOWNERS` in this component, like this https://github.com/esphome/esphome/blob/8215a018e97c8506ffe08c913580514b8faf89a8/esphome/components/climate_ir/__init__.py#L9

...when I get to it...

glmnet, 2020-09-17:
Ohh, I see..  well, looking at this it seems the `latest` version is the same as the `recommended` which is default, arduino 1.0.4, at least in Esphome v1.15.1 with Platformio 4.3.4

Alex9779, 2020-09-17:
Hmmmm if that is so then why does it work for him by setting it and does not work when omitted?
Didn't try myself, will do when I get home this evening to get maybe a reproducible result from another source 😁