esphome/issues #1471: Xiaomi lywsd03mmc incorrectly reports battery status

AnthonyKNorman, 2020-09-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io Raspberry Pi 4
**ESP (ESP32/ESP8266, Board/Sonoff):**

Generic ESP32
**ESPHome version (latest production, beta, dev branch)**

1.15 release
**Affected component:**

https://esphome.io/components/sensor/xiaomi_ble.html

**Description of problem:**
I have two lywsd03mmc devices. One reports 100% battery and the other reports 10%
The Temperature and humidity reports correctly. I have, until now, been using my own code to read these devices https://github.com/AnthonyKNorman/Xiaomi_LYWSD03MMC_for_HA. This gives sensible readings for battery, but connection is unreliable, so I would prefer to use the excellent ESPHome, which I use extensively already.
This repository identifies the issue https://github.com/JsBergbau/MiTemperature2
"Reading the battery level with the standard Bluetooth Low Energy characteristics doesn't work. It always returns 99 % battery level. Or to be correct, sometimes 10 % when the battery is really empty, see https://github.com/JsBergbau/MiTemperature2/issues/1#issuecomment-588156894 . But often before that device just shuts down before it can report another battery level. With every measurement the Aqara sensor also transmits the battery voltage."

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
'''
esp32_ble_tracker:  

sensor:
  - platform: xiaomi_lywsd03mmc
    mac_address: "A4:C1:38:47:63:36"
    bindkey: "1e32bb306f9933edf8b2b02ccb8a005a"
    temperature:
      name: "LYWSD03MMC-1 Temperature"
    humidity:
      name: "LYWSD03MMC-1 Humidity"
    battery_level:
      name: "LYWSD03MMC-1 Battery Level"
      
  - platform: xiaomi_lywsd03mmc
    mac_address: "A4:C1:38:CF:BC:2F"
    bindkey: "e5f9c049a878dd256a4a5732b66b8670"
    temperature:
      name: "LYWSD03MMC-2 Temperature"
    humidity:
      name: "LYWSD03MMC-2 Humidity"
    battery_level:
      name: "LYWSD03MMC-2 Battery Level"
      '''
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
howing Logs For: xiaomi.yaml
INFO Reading configuration /config/esphome/xiaomi.yaml...
INFO Starting log output from xiaomi.local using esphome API
INFO Connecting to xiaomi.local:6053 (192.168.0.62)
INFO Successfully connected to xiaomi.local
[12:49:42][I][app:105]: ESPHome version 1.15.0 compiled on Sep 14 2020, 19:27:26
[12:49:42][C][wifi:443]: WiFi:
[12:49:42][C][wifi:303]:   SSID: [redacted]
[12:49:42][C][wifi:304]:   IP Address: 192.168.0.62
[12:49:42][C][wifi:306]:   BSSID: [redacted]
[12:49:42][C][wifi:307]:   Hostname: 'xiaomi'
[12:49:42][C][wifi:311]:   Signal strength: -59 dB ▂▄▆█
[12:49:42][C][wifi:315]:   Channel: 6
[12:49:42][C][wifi:316]:   Subnet: 255.255.255.0
[12:49:42][C][wifi:317]:   Gateway: 192.168.0.1
[12:49:42][C][wifi:318]:   DNS1: 192.168.0.1
[12:49:42][C][wifi:319]:   DNS2: 0.0.0.0
[12:49:42][C][logger:185]: Logger:
[12:49:42][C][logger:186]:   Level: DEBUG
[12:49:42][C][logger:187]:   Log Baud Rate: 115200
[12:49:42][C][logger:188]:   Hardware UART: UART0
[12:49:42][C][esp32_ble_tracker:530]: BLE Tracker:
[12:49:42][C][esp32_ble_tracker:531]:   Scan Duration: 300 s
[12:49:42][C][esp32_ble_tracker:532]:   Scan Interval: 320.0 ms
[12:49:42][C][esp32_ble_tracker:533]:   Scan Window: 30.0 ms
[12:49:42][C][esp32_ble_tracker:534]:   Scan Type: ACTIVE
[12:49:42][C][xiaomi_lywsd03mmc:012]: Xiaomi LYWSD03MMC
[12:49:42][C][xiaomi_lywsd03mmc:013]:   Bindkey: 1E.32.BB.30.6F.99.33.ED.F8.B2.B0.2C.CB.8A.00.5A (16)
[12:49:42][C][xiaomi_lywsd03mmc:014]:   Temperature 'LYWSD03MMC-1 Temperature'
[12:49:42][C][xiaomi_lywsd03mmc:014]:     Unit of Measurement: '°C'
[12:49:42][C][xiaomi_lywsd03mmc:014]:     Accuracy Decimals: 1
[12:49:42][C][xiaomi_lywsd03mmc:014]:     Icon: 'mdi:thermometer'
[12:49:42][C][xiaomi_lywsd03mmc:015]:   Humidity 'LYWSD03MMC-1 Humidity'
[12:49:42][C][xiaomi_lywsd03mmc:015]:     Unit of Measurement: '%'
[12:49:42][C][xiaomi_lywsd03mmc:015]:     Accuracy Decimals: 0
[12:49:42][C][xiaomi_lywsd03mmc:015]:     Icon: 'mdi:water-percent'
[12:49:42][C][xiaomi_lywsd03mmc:016]:   Battery Level 'LYWSD03MMC-1 Battery Level'
[12:49:42][C][xiaomi_lywsd03mmc:016]:     Unit of Measurement: '%'
[12:49:42][C][xiaomi_lywsd03mmc:016]:     Accuracy Decimals: 0
[12:49:42][C][xiaomi_lywsd03mmc:016]:     Icon: 'mdi:battery'
[12:49:42][C][xiaomi_lywsd03mmc:012]: Xiaomi LYWSD03MMC
[12:49:42][C][xiaomi_lywsd03mmc:013]:   Bindkey: E5.F9.C0.49.A8.78.DD.25.6A.4A.57.32.B6.6B.86.70 (16)
[12:49:42][C][xiaomi_lywsd03mmc:014]:   Temperature 'LYWSD03MMC-2 Temperature'
[12:49:42][C][xiaomi_lywsd03mmc:014]:     Unit of Measurement: '°C'
[12:49:42][C][xiaomi_lywsd03mmc:014]:     Accuracy Decimals: 1
[12:49:42][C][xiaomi_lywsd03mmc:014]:     Icon: 'mdi:thermometer'
[12:49:42][C][xiaomi_lywsd03mmc:015]:   Humidity 'LYWSD03MMC-2 Humidity'
[12:49:42][C][xiaomi_lywsd03mmc:015]:     Unit of Measurement: '%'
[12:49:42][C][xiaomi_lywsd03mmc:015]:     Accuracy Decimals: 0
[12:49:42][C][xiaomi_lywsd03mmc:015]:     Icon: 'mdi:water-percent'
[12:49:42][C][xiaomi_lywsd03mmc:016]:   Battery Level 'LYWSD03MMC-2 Battery Level'
[12:49:42][C][xiaomi_lywsd03mmc:016]:     Unit of Measurement: '%'
[12:49:42][C][xiaomi_lywsd03mmc:016]:     Accuracy Decimals: 0
[12:49:42][C][xiaomi_lywsd03mmc:016]:     Icon: 'mdi:battery'
[12:49:42][C][captive_portal:169]: Captive Portal:
[12:49:42][C][ota:029]: Over-The-Air Updates:
[12:49:42][C][ota:030]:   Address: xiaomi.local:3232
[12:49:42][C][ota:032]:   Using Password.
[12:49:42][C][api:095]: API Server:
[12:49:42][C][api:096]:   Address: xiaomi.local:6053
[12:50:05][D][esp32_ble_tracker:544]: Found device 75:A7:26:08:36:59 RSSI=-54
[12:50:05][D][esp32_ble_tracker:565]:   Address Type: RANDOM
[12:50:05][D][esp32_ble_tracker:569]:   TX Power: 2
[12:51:13][D][esp32_ble_tracker:148]: Starting scan...
[12:51:14][D][esp32_ble_tracker:544]: Found device 43:17:CE:4D:A6:41 RSSI=-60
[12:51:14][D][esp32_ble_tracker:565]:   Address Type: RANDOM
[12:51:15][D][esp32_ble_tracker:544]: Found device A4:C1:38:CF:BC:2F RSSI=-72
[12:51:15][D][esp32_ble_tracker:565]:   Address Type: PUBLIC
[12:51:15][D][esp32_ble_tracker:567]:   Name: 'LYWSD03MMC'
[12:51:18][D][esp32_ble_tracker:544]: Found device 75:A7:26:08:36:59 RSSI=-62
[12:51:18][D][esp32_ble_tracker:565]:   Address Type: RANDOM
[12:51:18][D][esp32_ble_tracker:569]:   TX Power: 2
[12:51:19][D][esp32_ble_tracker:544]: Found device 50:AC:C0:24:4A:C4 RSSI=-68
[12:51:19][D][esp32_ble_tracker:565]:   Address Type: RANDOM
[12:51:50][D][esp32_ble_tracker:544]: Found device A4:C1:38:47:63:36 RSSI=-66
[12:51:50][D][esp32_ble_tracker:565]:   Address Type: PUBLIC
[12:51:50][D][esp32_ble_tracker:567]:   Name: 'LYWSD03MMC'
[12:52:28][D][xiaomi_ble:273]: Got Xiaomi LYWSD03MMC (A4:C1:38:47:63:36):
[12:52:28][D][xiaomi_ble:279]:   Humidity: 65.0%
[12:52:28][D][sensor:092]: 'LYWSD03MMC-1 Humidity': Sending state 65.00000 % with 0 decimals of accuracy
[12:53:27][D][xiaomi_ble:273]: Got Xiaomi LYWSD03MMC (A4:C1:38:47:63:36):
[12:53:27][D][xiaomi_ble:276]:   Temperature: 23.8°C
[12:53:27][D][sensor:092]: 'LYWSD03MMC-1 Temperature': Sending state 23.80000 °C with 1 decimals of accuracy
```

**Additional information and things you've tried:**

nikito7, 2020-09-24:
Battery values are 100% 50% 10%

10% show batt in lcd

This in MiHome

AnthonyKNorman, 2020-09-24:
Yes, I get that, but if you look at the work JsBergbau has done here https://github.com/JsBergbau/MiTemperature2 you can see it is also capable of reporting actual battery voltage. I have implemented this in my project https://github.com/AnthonyKNorman/Xiaomi_LYWSD03MMC_for_HA. My connection was, however, never stable and I am delighted to find this support in ESPHome, which I use extensively. It's not a showstopper. Just a nice enhancement.

nikito7, 2020-09-24:
It need direct connection. Not supported at esphome.

You can use ATC custom firmware:

https://github.com/atc1441/ATC_MiThermometer

napieraj, 2020-10-01:
esphome is only able to listen for broadcast packets, and not actively connect to a BLE device to get readings

closing this issue for this reason -- if you would like to open a feature_request: [esphome/feature-requests](https://github.com/esphome/feature-requests)

closed 2020-10-01 by napieraj