esphome/issues #1472: ESP32 cannot connect to WiFi when there is an GPIO assigned in the config with esphome version 1.15.1

pengelen, 2020-09-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32
**ESPHome version (latest production, beta, dev branch)**

latest production 1.15.1
**Affected component:**

WiFi Component

**Description of problem:**
ESP32 cannot connect to WiFi when there is a GPIO assigned in the config with esphome version 1.15.1
Can be a switch, light, binary sensor or sensor.
It did work fine with version 1.15.0

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp12
  platform: ESP32
  board: nodemcu-32s

substitutions:
  name: ESP12

wifi:
  ssid: "!secret wlanssid"
  password: "!secret wlanpassword"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp12 Fallback Hotspot"
    password: "!secret wlanfallbackpassword"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

web_server:
  port: 80

time:
  - platform: homeassistant
    id: homeassistant_time

# Text sensors with general information
text_sensor:
  - platform: version
    name: ${name}_version
  - platform: wifi_info
    ip_address:
      name: ${name}_ip

sensor:
  - platform: wifi_signal
    name: "WiFi Signal ${name}"
    update_interval: 60s
  - platform: uptime
    name: ${name}_uptime
  - platform: dht
    pin: GPIO04
    model: DHT22
    temperature:
      name: "${name} Temperature"
    humidity:
      name: "${name} Humidity"
    update_interval: 30s

light:
  - platform: neopixelbus
    variant: SK6812
    pin: GPIO05
    num_leds: 9
    type: GRBW
    name: "${name} SK6812 Light"
    id: light_1
    effects:
      - random:
          name: "My Slow Random Effect"
          transition_length: 30s
          update_interval: 30s
      - random:
          name: "My Fast Random Effect"
          transition_length: 4s
          update_interval: 5s
      - strobe:
      - strobe:
          name: Strobe Effect With Custom Values
          colors:
            - state: True
              brightness: 100%
              red: 100%
              green: 90%
              blue: 0%
              duration: 500ms
            - state: False
              duration: 250ms
            - state: True
              brightness: 100%
              red: 0%
              green: 100%
              blue: 0%
              duration: 500ms
      - flicker:
      - flicker:
          name: Flicker Effect With Custom Values
          alpha: 95%
          intensity: 1.5%
      - addressable_rainbow:
      - addressable_color_wipe:
      - addressable_scan:
      - addressable_scan:
          name: Scan Effect With Custom Values
          move_interval: 100ms
          scan_width: 2

switch:
  - platform: gpio
    pin:
      number: GPIO25
      inverted: true
    name: "${name} Relay #1"
    id: relay1
  - platform: gpio
    pin:
      number: GPIO26
      inverted: true
    name: "${name} Relay #2"
    id: relay2
  - platform: gpio
    pin:
      number: GPIO32
      inverted: true
    name: "${name} Relay #3"
    id: relay3
  - platform: gpio
    pin:
      number: GPIO33
      inverted: true
    name: "${name} Relay #4"
    id: relay4
  - platform: restart
    name: ${name}_Restart

  - platform: template
    id: led_switch_1
    name: "${name} Switch led 1"
    turn_on_action:
        - light.turn_on:
            id: light_1
            brightness: 0.0
            red: 0.0
            green: 0.0
            blue: 0.0
            white: 1.0
            effect: none
    turn_off_action:
        - light.turn_off: light_1
  - platform: template
    id: led_switch_2
    name: "${name} Switch led 2"
    turn_on_action:
        - light.turn_on:
            id: light_1
            brightness: 1.0
            red: 0.0
            green: 0.0
            blue: 1.0
            white: 0.0
            effect: none
    turn_off_action:
        - light.turn_off: light_1
  - platform: template
    id: led_switch_3
    name: "${name} Switch led 3"
    turn_on_action:
        - light.turn_on:
            id: light_1    
            brightness: 100%
            red: 100%
            green: 49%
            blue: 0%
            white: 0%
            effect: 'Scan Effect With Custom Values'
    turn_off_action:
        - light.turn_off: light_1
  - platform: template
    id: led_switch_4
    name: "${name} Switch led 4"
    optimistic: True
    turn_on_action:
        - light.turn_on:
            id: light_1    
            brightness: 100%
            red: 0%
            green: 100%
            blue: 0%
            white: 0%
            effect: none
    turn_off_action:
        - light.turn_off: light_1

```

**Logs (if applicable):**

```
[20:30:29][D][ota:243]: OTA in progress: 91.6%
[20:30:30][D][ota:243]: OTA in progress: 94.8%
[20:30:31][D][ota:243]: OTA in progress: 97.8%
[20:30:32][I][ota:269]: OTA update finished!
[20:30:33][I][app:142]: Rebooting safely...
[20:30:33]ets Jun  8 2016 00:22:57
[20:30:33]
[20:30:33]rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
[20:30:33]configsip: 0, SPIWP:0xee
[20:30:33]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[20:30:33]mode:DIO, clock div:2
[20:30:33]load:0x3fff0018,len:4
[20:30:33]load:0x3fff001c,len:1044
[20:30:33]load:0x40078000,len:8896
[20:30:33]load:0x40080400,len:5828
[20:30:33]entry 0x400806ac
[20:30:33][I][logger:166]: Log initialized
[20:30:33][C][ota:366]: There have been 0 suspected unsuccessful boot attempts.
[20:30:33][I][app:029]: Running through setup()...
[20:30:33][C][switch.gpio:011]: Setting up GPIO Switch 'ESP12 Relay #1'...
[20:30:33][D][switch:025]: 'ESP12 Relay #1' Turning OFF.
[20:30:33][D][switch:045]: 'ESP12 Relay #1': Sending state OFF
[20:30:33][D][switch:025]: 'ESP12 Relay #1' Turning OFF.
[20:30:33][C][switch.gpio:011]: Setting up GPIO Switch 'ESP12 Relay #2'...
[20:30:33][D][switch:025]: 'ESP12 Relay #2' Turning OFF.
[20:30:33][D][switch:045]: 'ESP12 Relay #2': Sending state OFF
[20:30:33][D][switch:025]: 'ESP12 Relay #2' Turning OFF.
[20:30:33][C][switch.gpio:011]: Setting up GPIO Switch 'ESP12 Relay #3'...
[20:30:33][D][switch:025]: 'ESP12 Relay #3' Turning OFF.
[20:30:33][D][switch:045]: 'ESP12 Relay #3': Sending state OFF
[20:30:33][D][switch:025]: 'ESP12 Relay #3' Turning OFF.
[20:30:33][C][switch.gpio:011]: Setting up GPIO Switch 'ESP12 Relay #4'...
[20:30:33][D][switch:025]: 'ESP12 Relay #4' Turning OFF.
[20:30:33][D][switch:045]: 'ESP12 Relay #4': Sending state OFF
[20:30:33][D][switch:025]: 'ESP12 Relay #4' Turning OFF.
[20:30:33][C][light:097]: Setting up light 'ESP12 SK6812 Light'...
[20:30:33][D][light:264]: 'ESP12 SK6812 Light' Setting:
[20:30:33][D][light:277]:   Brightness: 100%
[20:30:33][D][light:286]:   Red=100%, Green=100%, Blue=100%
[20:30:33][D][light:289]:   White Value: 100%
[20:30:33][C][dht:011]: Setting up DHT...
[20:30:33][D][text_sensor:015]: 'ESP12_version': Sending state '1.15.1 Sep 15 2020, 20:29:25'
[20:30:33][C][wifi:033]: Setting up WiFi...
[20:30:33][D][wifi:324]: Starting scan...
[20:30:33][D][dht:048]: Got Temperature=25.7°C Humidity=54.0%
[20:30:33][D][sensor:092]: 'ESP12 Temperature': Sending state 25.70000 °C with 1 decimals of accuracy
[20:30:33][D][sensor:092]: 'ESP12 Humidity': Sending state 54.00000 % with 0 decimals of accuracy
[20:30:33][D][sensor:092]: 'ESP12_uptime': Sending state 0.33700 s with 0 decimals of accuracy
```

**Additional information and things you've tried:**
fast_connect option but keeps on state connect to network <SSID>
Removed all switches, sensors and lights, then it connects to the WiFi network without any problems.
up to version 1.15.0 there where no issues.

pengelen, 2020-10-01:
version 1.15.2 does work with a extra modification, because of the neopixel bug #1481 

esphome:
  name: esp12
  platform: ESP32
  board: nodemcu-32s

  **# a workaround for the latest bug in the neopixel lib bug
  platformio_options:
    lib_deps: NeoPixelBus@2.6.0**

napieraj, 2020-10-01:
with 1.15.2 resolving the wifi issue, and #1481 remaining open for the neopixel dependency issue, I will consider your issue resolved and close this - if you have any further remarks feel free to reopen or make a new issue

closed 2020-10-01 by napieraj