esphome/issues #1473: Output off for a few ms every x min

xsnoopy, 2020-09-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker on Linux Server

**ESP (ESP32/ESP8266, Board/Sonoff):**

Wemos D1 Mini Pro with MCP23017 port expander and Relais modul. 

**ESPHome version (latest production, beta, dev branch)**

V1.15.1

**Affected component:**

https://esphome.io/components/mcp230xx.html

**Description of problem:**
I have a 5V 8 Channel Relais card connected to Port Expander. Every now and then the Relais go off for let say 100ms (guesstimated) I hear the relais going off and going back on right away. This comes without any external event. The node is not integrated in home assistant. It is just sitting on my desk on a dry run. I had a similar issue with an ESP 32 POE board, but thought it was the power supply or something else. Since this problem occurs on an Wemos D1 Mini Pro as well as on the ESP 32 POE I thinkt the cause must be something else. 

How can I help finding the source of this issue? 

I want to use the set up to drive even higher loads and motors and can't get them switching off and on. This would lead to a lot wear on the components.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

esphome:
  name: relais
  platform: ESP8266
  board: esp01_1m

wifi:
  networks:
  - ssid: "asd"
    password: "aaaaa"
  - ssid: "das"
    password: "bbbb"
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "dasdt"
    password: "dasd"

captive_portal:

# Example configuration entry
i2c:
  #- id: bus_a
    sda: 5
    scl: 4
    scan: True

mcp23017:
  - id: 'hub000'
    address: 0x20
  - id: 'hub010'
    address: 0x22
  - id: 'hub100'
    address: 0x24

switch: 
 - platform: gpio
   name: "Hub 000 Pin #A0"
   pin:
     mcp23017: hub000
     # Use pin number 0
     number: 0
     mode: OUTPUT
     inverted: False 

 - platform: gpio
   name: "Hub 000 Pin #A1"
   pin:
     mcp23017: hub000
     # Use pin number 1
     number: 1
     mode: OUTPUT
     inverted: False 
 - platform: gpio
   name: "Hub 000 Pin #A2"
   pin:
     mcp23017: hub000
     # Use pin number 2
     number: 2
     mode: OUTPUT
     inverted: False 
 - platform: gpio
   name: "Hub 000 Pin #A3"
   pin:
     mcp23017: hub000
     # Use pin number 3
     number: 3
     mode: OUTPUT
     inverted: False 
 - platform: gpio
   name: "Hub 000 Pin #A4"
   pin:
     mcp23017: hub000
     # Use pin number 4
     number: 4
     mode: OUTPUT
     inverted: False 
 - platform: gpio
   name: "Hub 000 Pin #A5"
   pin:
     mcp23017: hub000
     # Use pin number 5
     number: 5
     mode: OUTPUT
     inverted: False 
     
 - platform: gpio
   name: "Hub 000 Pin #A6"
   pin:
     mcp23017: hub000
     # Use pin number 6
     number: 6
     mode: OUTPUT
     inverted: False 
 - platform: gpio
   name: "Hub 000 Pin #A7"
   pin:
     mcp23017: hub000
     # Use pin number 7
     number: 7
     mode: OUTPUT
     inverted: False 
 - platform: gpio
   name: "Hub 000 Pin #B0"
   pin:
     mcp23017: hub000
     # Use pin number 8
     number: 8
     mode: OUTPUT
     inverted: False 
 - platform: gpio
   name: "Hub 000 Pin #B1"
   pin:
     mcp23017: hub000
      # Use pin number 9
     number: 9
     mode: OUTPUT
     inverted: False 

 - platform: gpio
   name: "Hub 000 Pin #B2"
   pin:
     mcp23017: hub000
     # Use pin number 10
     number: 10
     mode: OUTPUT
     inverted: False 

 - platform: gpio
   name: "Hub 000 Pin #B3"
   pin:
     mcp23017: hub000
     # Use pin number 11
     number: 11
     mode: OUTPUT
     inverted: False 

 - platform: gpio
   name: "Hub 000 Pin #B4"
   pin:
     mcp23017: hub000
     # Use pin number 12
     number: 12
     mode: OUTPUT
     inverted: False 

 - platform: gpio
   name: "Hub 000 Pin #B5"
   pin:
     mcp23017: hub000
     # Use pin number 13
     number: 13
     mode: OUTPUT
     inverted: False 

 - platform: gpio
   name: "Hub 000 Pin #B6"
   pin:
     mcp23017: hub000
     # Use pin number 14
     number: 14
     mode: OUTPUT
     inverted: False 

 - platform: gpio
   name: "Hub 000 Pin #B7"
   pin:
     mcp23017: hub000
     # Use pin number 15
     number: 15
     mode: OUTPUT
     inverted: False 

web_server:
  port: 80

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

glmnet, 2020-09-16:
Most likely your issue is a reboot, the automatic reboot specifically... as you are using `api:` and you are also stating that you are not connecting it to home-assistant 

You should look at serial logs in this case, to be completely sure.
Check docs and [faq  ](https://esphome.io/guides/faq.html#my-node-keeps-reconnecting-randomly)

I'm closing this, will re open if you add logs and it shows something different.

closed 2020-09-16 by glmnet

xsnoopy, 2020-09-16:
Here is the serial log when the relais switch off. It seems to be a reboot. How do I stop that?

[21:29:26][D][sensor:092]: 'WiFi Signal abc': Sending state -64.00000 dB with 0 decimals of accuracy
[21:30:04][E][api:084]: No client connected to API. Rebooting...
[21:30:04][I][app:132]: Forcing a reboot...

glmnet, 2020-09-16:
See faq link in my previous post