esphome/issues #1474: (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 2.25.1 internal error, aborting at /builds/idf/crosstool-NG/.build/src/binutils-2.25.1/bfd/merge.c line 876 in _bfd_merged_section_offset  /root/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: Please report this bug.

dereitz, 2020-09-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io on Ubuntu 18.04 via install script and using esphome repositories

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32  Specific board here:
https://www.amazon.com/gp/product/B07Q576VWZ/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

**ESPHome version (latest production, beta, dev branch)**
1.15.1

**Affected component:**
N/A

**Description of problem:**
I tried upgrading from 0.14.x to 1.15.1 today and receive the following log which indicates to report a bug...  Please help!

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: closet
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pw

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Closet Fallback Hotspot"
    password: "password"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

esp32_ble_tracker:

binary_sensor:
  - platform: ble_presence
    mac_address: ABC
    device_class: presence
    name: "Name's Harness"
  - platform: ble_presence
    mac_address: XYZ
    device_class: presence
    name: "Name"

```

**Logs (if applicable):**

```
Updating /config/esphome/closet.yaml
------------------------------------------------------------

INFO Running:  esphome --dashboard /config/esphome/closet.yaml run --no-logs --upload-port OTA
INFO Reading configuration /config/esphome/closet.yaml...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Deleting /data/closet/.pioenvs
INFO Deleting /data/closet/.piolibdeps
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/closet
[... 124 lines cut ...]
Compiling /data/closet/.pioenvs/closet/FrameworkArduino/esp32-hal-time.c.o
Compiling /data/closet/.pioenvs/closet/FrameworkArduino/esp32-hal-timer.c.o
Compiling /data/closet/.pioenvs/closet/FrameworkArduino/esp32-hal-touch.c.o
Compiling /data/closet/.pioenvs/closet/FrameworkArduino/esp32-hal-uart.c.o
Compiling /data/closet/.pioenvs/closet/FrameworkArduino/libb64/cdecode.c.o
Compiling /data/closet/.pioenvs/closet/FrameworkArduino/libb64/cencode.c.o
Compiling /data/closet/.pioenvs/closet/FrameworkArduino/main.cpp.o
Compiling /data/closet/.pioenvs/closet/FrameworkArduino/stdlib_noniso.c.o
Compiling /data/closet/.pioenvs/closet/FrameworkArduino/wiring_pulse.c.o
Compiling /data/closet/.pioenvs/closet/FrameworkArduino/wiring_shift.c.o
Archiving /data/closet/.pioenvs/closet/libFrameworkArduino.a
Linking /data/closet/.pioenvs/closet/firmware.elf
/root/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: BFD (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 2.25.1 internal error, aborting at /builds/idf/crosstool-NG/.build/src/binutils-2.25.1/bfd/merge.c line 876 in _bfd_merged_section_offset

/root/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: Please report this bug.

collect2: error: ld returned 1 exit status
*** [/data/closet/.pioenvs/closet/firmware.elf] Error 1
========================= [FAILED] Took 48.11 seconds =========================
=========== [ERROR] /config/esphome/closet.yaml ===========

```

**Additional information and things you've tried:**
N/A

dereitz, 2020-09-15:
Perhaps this was a one time thing?  I added an i2c sensor, did a compile, etc, and it appears my ESP32 is now running 1.15.1:

```
INFO Reading configuration /config/esphome/closet.yaml...
INFO Starting log output from closet.local using esphome API
INFO Connecting to closet.local:6053 (192.168.0.158)
INFO Successfully connected to closet.local
[19:34:48][I][app:105]: ESPHome version **1.15.1** compiled on Sep 15 2020, 19:27:13
```

dominikheinz, 2020-09-16:
I have this same error everytime i try to compile a binary for ESP32/8266. See #1476 

Did you already figure out what causing this?

dereitz, 2020-09-16:
I stumbled into fixing it I guess.  I did a clean build files and then went to update my yaml configuration file and then pushed the updates via upload.  That completed without issue, so I guess this is "fixed".

dominikheinz, 2020-09-16:
I tried that too, but without any success unfortunately.  😔 
I am really hoping this issue will get addressed soon, as it hinders me from using EspHome.

closed 2021-01-28 by stale[bot]