esphome/issues #1475: Control of RGBW Bulb wonky in HA

tduffy83, 2020-09-16:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HassOS VMDX VM
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266
**ESPHome version (latest production, beta, dev branch)**

1.15
**Affected component:**

Light

**Description of problem:**
I should preface this with the fact that I've turned on the new color_interlock feature.  The white value slider in HA behaves more like a binary switch.  All the way to the left the bulb displays white in RGB mode (all colors at once) and anywhere else the bulb is in white mode (only white led, no rgb.  Also, the white center of the color picker puts the bulb in white led mode, no rgb white mode.  This is contrary to the way all my other lights in HA behave and is not intuitive at all.  Typically with a slider you expect it to vary things, not be a binary choice.  This implementation is hacky at best, and confusing at worst for people used to how a light in HA behaves.

Expected behavior:  Change the white value slider, and the light changes to white mode with no color leds at the brightness of the slider.  The brightness slider is tied to the white slider in this case.  Pick the white value in the color picker, bulb uses the RGB leds to approximate white (after all this is the color picker).  

**Problem-relevant YAML-configuration entries:**
```yaml
N/A

```

**Logs (if applicable):**

```
N/A

```

**Additional information and things you've tried:**

N/A

sparkydave1981, 2020-09-18:
I believe this is the same issue I have just encountered after updating my RGBW lights from ESPhome 1.14.x to 1.15.  Previously my lights would need a brightness value of at least 1 for the white LED to turn on, with the white value slider then adjusting the brightness of the white value.  Now that no longer works at all.  Now the brightness adjusts the RGB LED's however the W LED can only be on at the same time as the RGB's since the brightness slider also affects the W LED.

Can we please have this fixed such that the W LED is controlled purely with the white value slider and the RGB LED's purely by the brightness slider?

tduffy83, 2020-09-18:
set the color_interlock variable to true to prevent the white led being on at the same time as the RGB led's.

sparkydave1981, 2020-09-18:
That worked perfectly, thank you

asasi83, 2020-09-21:
I can confirm the same behavior as tduffy83 stated in the first post.

napieraj, 2020-10-03:
@tduffy83 I understand your issue is resolved and you managed to achieve the desired behavior using the `color_interlock` variable?

tduffy83, 2020-10-03:
> @tduffy83 I understand your issue is resolved and you managed to achieve the desired behavior using the `color_interlock` variable?

No my issue is not resolved.  Color interlock does function, but the control of the bulb is not correct in home assistant as I described.

asasi83, 2020-10-03:
@napieraj in 1.14 you could control the brightness of the white value and/or brightness off RGB (you had full control of RGB and white values and you could mix the two however you want). Now at least in my case, you cant control the brightness of the white value and basically, the RGB is always on.

nofuturekid, 2020-10-08:
I second @asasi83 - I have the same issues. Turning on warm white automatically turns off the RGB and before version 1.15 I was able to run the RGB part ie. 75%/75%/75% and and the white LEDs at 100%. This is not working anymore.
Neither with "color_interlock = true" nor with "disabled".

Please revert the old behavior. Fixing things for a small group and breaking it for many other's it not the way it should go.

yllwfsh, 2020-10-08:
Same here. I preferred the old behaviour too. I actually used to mix white and colour values quite often to create some gentle colors

nofuturekid, 2020-10-08:
Right now I'm looking for a way to rollback to something prior 1.15 but as I'm using HomeAssistant with the HA extensions, it's not that easy.

tduffy83, 2020-10-08:
Ok folks your issues with color interlock are separate from the original issue and this isn't a support thread.  With that said, try setting color_interlock to false.  You mentioned you set it to disabled, which is not correct.

nofuturekid, 2020-10-09:
As far as it's me, I can tell, it makes no difference setting this "color_interlock" to "false" or "true" - In both cases same behaviour.
Ie. RGB was set to 75%/50%/25% and as soon as I turn on the warm white the RGB are reported back as 100% but a are "off" and only the warm white is lit up.

margau, 2020-10-09:
Hi, same behavior here
color interlock false, same config as with 1.14. 
The behavior was probably introduced with https://github.com/esphome/esphome/pull/967. It seems, that https://github.com/esphome/esphome/pull/1042/ is also affected.

I see this as urgent, as it breaks the default "straightforward" behavior with HASS. A fix with the next dot release would seem good.

Some problems seem to be within https://github.com/esphome/esphome/blob/e35d56defe09eeaebd57b9c15d138a21dc72d53f/esphome/components/light/light_state.cpp#L398 ff:
It seems that this messes with rgb colors when they are not updated (which is usally the case, when you adjust both channels separately in HASS), but without checking for interlock_true before.

The same here:
https://github.com/esphome/esphome/blob/e35d56defe09eeaebd57b9c15d138a21dc72d53f/esphome/components/light/light_state.cpp#L417 (white gets set to zero when an RGB value changes to other than 100%, without checking the interlock before).

If I find some time, I'll try to prepare an PR.

sparkydave1981, 2020-10-09:
> As far as it's me, I can tell, it makes no difference setting this "color_interlock" to "false" or "true" - In both cases same behaviour.
> Ie. RGB was set to 75%/50%/25% and as soon as I turn on the warm white the RGB are reported back as 100% but a are "off" and only the warm white is lit up.

Color_interlock: true should give that behaviour, but not when set false

tduffy83, 2020-10-09:
Can you guys open an issue for your expected behavior when color_interlock is set to false?  This issue is for when it's set to true.

nofuturekid, 2020-10-10:
@tduffy83 So I maybe misunderstood your issue here only because #1531 (which described exactly my problem) was closed  as duplicate of this one here. So I'm sorry for hijacking.

tduffy83, 2020-10-10:
Yes whoever closed that issue as a duplicate did not understand that the issues are different.

kwirk, 2021-04-01:
This is still an issue when `color_Interlock: true`. My work around for now is to override the bulb entity's `supported_features` in HA customisation to remove the white slider for now, as it causes too much confusion.

kwirk, 2021-05-05:
This change in Home Assistant 2021.5 may be relevant (I assume ESPHome will need updating to support this): https://www.home-assistant.io/blog/2021/05/05/release-20215/#color-modes

sparkydave1981, 2021-06-05:
This is definitely still an issue.  My RGBW lights are ignoring the interlock and still outputting to the white channel.  Has anyone managed to get it sorted locally?

oxan, 2021-07-10:
This should be fixed with the merge of esphome/esphome#1895, which will be available in v1.20.

closed 2021-07-10 by oxan