esphome/issues #1478: ESP8266 NodeMCU - Waveshare 7.50inV2 - Could not allocate buffer for display/Bootloop

maikkoe, 2020-09-16:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 NodeMCU
**ESPHome version (latest production, beta, dev branch)**

Current version: 1.15.1
**Affected component:**

https://esphome.io/components/display/waveshare_epaper.html

**Description of problem:**
If I try using the 7.50V2 with a ESP8266 NodeMCU i get the error "Could not allocate buffer for display!" and the ESP is in a boot loop.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: wsi75
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: !secret
  password: !secret

# Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Waveshare Info Fallback Hotspot"
    password: !secret

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret

ota:
  password: !secret

spi:
  clk_pin: D5
  mosi_pin: D7

display:
  - platform: waveshare_epaper
    id: epaper
    cs_pin: D8
    dc_pin: D3
    busy_pin: D2
    reset_pin: D4
    model: 7.50inV2

```

**Logs (if applicable):**

```
[14:07:24][E][display:017]: Could not allocate buffer for display!
[14:07:24][C][wifi:033]: Setting up WiFi...
[14:07:24][D][wifi:324]: Starting scan...
[14:07:24]
[14:07:24]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[14:07:24]
[14:07:24]Exception (29):
[14:07:25]epc1=0x40206902 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
[14:07:25]
[14:07:25]>>>stack>>>
[14:07:25]
[14:07:25]ctx: cont
[14:07:25]sp: 3ffffc50 end: 3fffffc0 offset: 0190
[14:07:25]3ffffde0:  000000ff 3fff1ef4 00000000 4020690d  
[14:07:25]3ffffdf0:  3fff200c 3fff1fd4 3fff1ef4 40206cd4  
[14:07:25]3ffffe00:  00000000 4bc6a7f0 00000348 4020517d  
[14:07:25]3ffffe10:  00000000 00000000 4bc6a7f0 402051d4  
[14:07:25]3ffffe20:  3fff1f04 00000003 40100716 cf5c28f5  
[14:07:25]3ffffe30:  00000000 00000000 00000100 4020666b  
[14:07:25]3ffffe40:  3fff1b94 00000005 3fff0660 4021d494  
[14:07:25]3ffffe50:  3fff1b94 00000005 3fff0660 40100370  
[14:07:25]3ffffe60:  3fff1b94 00000005 00000020 40100e8f  
[14:07:25]3ffffe70:  00000348 00000005 00000006 40209fc9  
[14:07:25]3ffffe80:  3fff10c4 00000006 3fff0660 00000100  
[14:07:25]3ffffe90:  3fff1b94 00000005 3fff0660 4020af72  
[14:07:25]3ffffea0:  3fff10c4 00000006 00000020 40100ec6  
[14:07:25]3ffffeb0:  00000100 3fff1e14 3fff1e2c 4020a310  
[14:07:25]3ffffec0:  3fff1ef4 3fff0660 00000010 3fff1ef4  
[14:07:25]3ffffed0:  3fff1fa4 3fff0660 00000000 4020c5e5  
[14:07:25]3ffffee0:  3fff1d50 00000000 00000000 3fff1d80  
[14:07:25]3ffffef0:  00000000 00000000 00000000 4021ec68  
[14:07:25]3fffff00:  00000000 4021ec68 00000000 4021ec68  
[14:07:25]3fffff10:  00000000 4021ec68 00000000 4021ec68  
[14:07:25]3fffff20:  00000000 00000000 3fff1cb8 00000000  
[14:07:25]3fffff30:  00000000 3fff1ce0 00000000 00000000  
[14:07:25]3fffff40:  00000000 4021ec68 00000000 4021ec68  
[14:07:25]3fffff50:  00000000 4021ec68 00000000 4021ec68  
[14:07:25]3fffff60:  00000000 4021ec68 00000000 00000000  
[14:07:25]3fffff70:  3fff1eb8 feefeffe feefeffe feefeffe  
[14:07:25]3fffff80:  3fff1ecc 3fff1ef4 feefeffe feefeffe  
[14:07:25]3fffff90:  feefeffe feefeffe feefeffe 3fff099c  
[14:07:25]3fffffa0:  3fffdad0 00000000 3fff095c 4021b0d0  
[14:07:25]3fffffb0:  feefeffe feefeffe 3ffe8520 4010058d  
[14:07:25]<<<stack<<<
[14:07:25]
[14:07:25]last failed alloc call: 4020519C(48000)
[14:07:25]
[14:07:25]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[14:07:25]
[14:07:25] ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[14:07:25]
[14:07:25]load 0x4010f000, len 3584, room 16 
[14:07:25]tail 0
[14:07:25]chksum 0xb0
[14:07:25]csum 0xb0
[14:07:25]v2843a5ac
[14:07:25]~ld
[14:07:25][I][logger:166]: Log initialized
[14:07:25][C][ota:366]: There have been 1 suspected unsuccessful boot attempts.
[14:07:25][I][app:029]: Running through setup()...
[14:07:25][C][spi:022]: Setting up SPI bus...
[14:07:25][E][display:017]: Could not allocate buffer for display!

```

**Additional information and things you've tried:**

I've used "model: 7.50in" in the configutation, no boot loop, but also no working Display.

glmnet, 2020-09-16:
Seems like a memory limitation on the esp8266, the display is 800 width x 480 height, ESPHomes uses a 2 bit [framebuffer ](https://en.wikipedia.org/wiki/Framebuffer) which is 800x480x2 / 8 = 48000 bytes ~ 47k RAM, the ESP8266 has a heap size you about that size.
You might want try disabling other parts of the code, like `captive_portal`, `ota`, etc. at least to try out.

maikkoe, 2020-09-16:
OTA, captive portal and ap commented out. Same behavior :(

maikkoe, 2020-09-18:
I've tested a few more constellations. I can't get the 7.50inV2 display to work. Currently I'm using a Lolin32 lite and get the following message in Verbose Logging:

[16:53:11][V][app:081]: A component took a long time in a loop() cycle (0.75 s).
[16:53:11][V][app:082]: Components should block for at most 20-30ms in loop().

and nothing is shown on the display. In order to eliminate the boards and the code as a source of error, I ordered the 4.20in display and it works with both, the ESP8266 NodeMCU and the Lolin32 Lite.

glmnet, 2020-09-18:
Cv @PaulAntonDeen

PaulAntonDeen, 2020-09-19:
I'm not sure what Cv means but I have no idea what this problem is or how to solve it.

maikkoe, 2020-10-01:
I have an ESP32 driverboard from Waveshare, so everything works now.

closed 2020-10-01 by maikkoe

eecue, 2021-03-26:
This also fixed my Rev 2.1 HAT issue.

eecue, 2021-03-26:
Strangely, this causes my epaper display to invert twice and the flash full black on every update.

eecue, 2021-03-26:
I have a 7.5" v2