esphome/issues #1481: NeoPixel LEDs does not work on ESP32

nkitanov, 2020-09-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP32**

**ESPHome version1.15.1**

**Affected component:**
NepixelBus LEDs

**Description of problem:**
After updating to the latest esphome 1.15.1 nepixelbus LEDs stopped working and ESP32 becomes unresponsive, even don't connect to wifi. 
On clean build I see this warning, probably caused by this bug in the latest neopixel library: https://github.com/Makuna/NeoPixelBus/issues/365
```
/data/iaq_livingroom_esp32/.piolibdeps/iaq_livingroom_esp32/NeoPixelBus-esphome/src/internal/Esp32_i2s.c: In function 'i2sSetPins':
/data/iaq_livingroom_esp32/.piolibdeps/iaq_livingroom_esp32/NeoPixelBus-esphome/src/internal/Esp32_i2s.c:264:13: warning: 'i2sSignal' may be used uninitialized in this function [-Wmaybe-uninitialized]
             gpio_matrix_out(out, i2sSignal, invert, false);
             ^
/data/iaq_livingroom_esp32/.piolibdeps/iaq_livingroom_esp32/NeoPixelBus-esphome/src/internal/Esp32_i2s.c: In function 'i2sInit':
/data/iaq_livingroom_esp32/.piolibdeps/iaq_livingroom_esp32/NeoPixelBus-esphome/src/internal/Esp32_i2s.c:383:5: warning: 'i2sIntSource' may be used uninitialized in this function [-Wmaybe-uninitialized]
     esp_intr_alloc(i2sIntSource, ESP_INTR_FLAG_IRAM | ESP_INTR_FLAG_LEVEL1, &i2sDmaISR, &I2S[bus_num], &I2S[bus_num].isr_handle);
     ^
```

**Problem-relevant YAML-configuration entries:**
```light:
  - platform: partition
    name: led1
    id: led1
    default_transition_length: 0s
    segments:
      - id: rgb_led
        from: 0
        to: 0
  - platform: partition
    name: led2
    id: led2
    default_transition_length: 0s
    segments:
    - id: rgb_led
      from: 1
      to: 1
  - platform: partition
    name: led3
    id: led3
    default_transition_length: 0s
    segments:
    - id: rgb_led
      from: 2
      to: 2
  - platform: neopixelbus
    num_leds: 3
    pin: GPIO16
    name: "RGB strip"
    id: rgb_led
    default_transition_length: 0s

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
Changed method of the lights to BIT_BANG and it works (still getting compile warning on a clean build). Then changed the method to ESP32_I2S_0 and the ESP works but LEDs are not possible to control. 

The issue is probably related to https://github.com/Makuna/NeoPixelBus/issues/365

d3sbo, 2020-09-19:
I get the same

squirtbrnr, 2020-09-19:
https://github.com/esphome/issues/issues/1450

use platformio_options and the lib_deps header to specify you want to build against the neopixelbus 2.5.2 library.  This is a workaround, but ultimately there's a problem with the 2.5.7 library.

nkitanov, 2020-09-19:
Thank you, I was looking for the option to override the library. Now it works with esphome 1.15.1 and neopixelbus 2.5.2 library.

nkitanov, 2020-09-21:
It is fixed in the upstream library now. How can use neopixel 2.6.0 in esphome? Or it needs to be released in esphome also...

idofr, 2020-09-27:
I confirm having the same problem and solving it with
```yaml
esphome:
  name: test_node
  platform: ESP32
  board: esp-wrover-kit
  
  # a workaround for the latest bug in the neopixel lib bug
  platformio_options:
    lib_deps: NeoPixelBus@2.6.0
```

Fusseldieb, 2020-09-29:
> I confirm having the same problem and solving it with
> 
> ```yaml
> esphome:
>   name: test_node
>   platform: ESP32
>   board: esp-wrover-kit
>   
>   # a workaround for the latest bug in the neopixel lib bug
>   platformio_options:
>     lib_deps: NeoPixelBus@2.6.0
> ```
> 
> Notice that this is currently the latest release of the NeoPixelBus which includes a bug fix for this exact problem as reported in [Makuna/NeoPixelBus#365](https://github.com/Makuna/NeoPixelBus/issues/365)

Just tried, unfortunately it couldn't compile:

```
INFO Reading configuration /config/esphome/cozinha_led_teto.yaml...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Deleting /data/cozinha_led_teto/.pioenvs
INFO Deleting /data/cozinha_led_teto/.piolibdeps
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/cozinha_led_teto
Processing cozinha_led_teto (board: esp32cam; framework: arduino; platform: espressif32@1.12.4)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif32 3.10004.200129 (1.0.4) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
Looking for NeoPixelBus library in registry
LibraryManager: Installing id=547 @ 2.6.0
NeoPixelBus @ 2.6.0 has been successfully installed!
Dependency Graph
|-- <NeoPixelBus> 2.6.0
|   |-- <SPI> 1.0
|-- <Update> 1.0
|-- <WiFi> 1.0
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
Compiling /data/cozinha_led_teto/.pioenvs/cozinha_led_teto/src/esphome/components/api/api_connection.cpp.o
Compiling /data/cozinha_led_teto/.pioenvs/cozinha_led_teto/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/cozinha_led_teto/.pioenvs/cozinha_led_teto/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/cozinha_led_teto/.pioenvs/cozinha_led_teto/src/esphome/components/api/api_server.cpp.o
In file included from src/esphome/components/api/api_connection.h:7:0,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/components/api/api_server.h:16:22: fatal error: AsyncTCP.h: No such file or directory

******************************************************************
* Looking for AsyncTCP.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:AsyncTCP.h"
* Web  > https://platformio.org/lib/search?query=header:AsyncTCP.h
*
******************************************************************

compilation terminated.
*** [/data/cozinha_led_teto/.pioenvs/cozinha_led_teto/src/esphome/components/api/api_connection.cpp.o] Error 1
In file included from src/esphome/components/api/api_server.cpp:1:0:
src/esphome/components/api/api_server.h:16:22: fatal error: AsyncTCP.h: No such file or directory

******************************************************************
* Looking for AsyncTCP.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:AsyncTCP.h"
* Web  > https://platformio.org/lib/search?query=header:AsyncTCP.h
*
******************************************************************

compilation terminated.
*** [/data/cozinha_led_teto/.pioenvs/cozinha_led_teto/src/esphome/components/api/api_server.cpp.o] Error 1
========================== [FAILED] Took 7.00 seconds ==========================
```

squirtbrnr, 2020-09-29:
Your error has nothing to do with NeoPixelBus. The compile error is pointing at something missing for the API.

I confirmed with my own builds that specifying `NeoPixelBus@2.6.0` does force the node to compile with the new library version and the node/LEDs work again.

Fusseldieb, 2020-09-29:
> Your error has nothing to do with NeoPixelBus. The compile error is pointing at something missing for the API.
> 
> I confirmed with my own builds that specifying `NeoPixelBus@2.6.0` does force the node to compile with the new library version and the node/LEDs work again.

Well, the error wasn't there before.

How can i force-include AsyncTCP?

Thanks in advance.

edoardo-conti, 2020-09-30:
I solved by adding two more dependencies (I have to include `FastLED` to be able to compile successful):

```
# a workaround for the latest bug in the neopixel lib bug
  platformio_options:
    lib_deps: 
      - NeoPixelBus@2.6.0
      - AsyncTCP
      - FastLED
      - ESPAsyncWebServer-esphome
```

Fusseldieb, 2021-01-29:
Was this fixed at some point?

nkitanov, 2021-02-06:
No, I just checked with esphome 1.16.0 and it still uses neopixel 2.5.7 not 2.6.0 (or what is latest). I still use `lib_deps: NeoPixelBus@2.6.0` and it's ok.

benneq, 2021-02-09:
Got the same issue. My config only includes neopixelbus and a button, and it hangs while booting.

`method: BIT_BANG` kinda works with my ESP32 NodeMCU, but a single LED in the middle of the LED stripe won't turn off.

But the manual version upgrade did the trick for me 👍 
```
esphome:
  name: test32
  platform: ESP32
  board: nodemcu-32s
  
  platformio_options:
    lib_deps: NeoPixelBus@2.6.1
```

Now everything works as expected!

angelnu, 2021-04-08:
@OttoWinter - is there a reason for using a fork (neopixelbus-esphome) instead of the original repo?

https://github.com/esphome/esphome/pull/1519 does not work because the fork is outdated - it could be updated by maybe is not needed?

OttoWinter, 2021-04-08:
> @OttoWinter - is there a reason for using a fork (neopixelbus-esphome) instead of the original repo?
> 
> [esphome/esphome#1519](https://github.com/esphome/esphome/pull/1519) does not work because the fork is outdated - it could be updated by maybe is not needed?

There is, the code in neopixelbus was breaking our CI, and the PR fixing that in their repo was closed (which I kind of understand, because it's not really their problem): https://github.com/Makuna/NeoPixelBus/pull/277 - but we still needed a functioning CI, hence the fork.

schmurtzm, 2021-04-08:
Same problem here, so I update to last version of NeoPixelBus : 
  platformio_options:
    lib_deps: NeoPixelBus@2.6.2
    
It works but I observe that it can impact I2C performances (when I switch my leds my apds9960 I2C connection is often lost). fastled_clocklessdoesn't impact I2C in the same way.

angelnu, 2021-04-09:
> There is, the code in neopixelbus was breaking our CI, and the PR fixing that in their repo was closed (which I kind of understand, because it's a weird downstream issue that's only showing with specific compiler configurations) [Makuna/NeoPixelBus#277](https://github.com/Makuna/NeoPixelBus/pull/277) - but we still needed a functioning CI, hence the fork.

Ok, Makuna did not remember the reason so I asked in https://discordapp.com/channels/789177382221119519/789177382221119524/830147966748196906 if we could try again with a PR.

schmurtzm, 2021-05-12:
Just for information, quick test on ESPHome 1.7.2 because I saw that there was some update on NeoPixelBus Light but it still crashing.

nkitanov, 2021-09-09:
This is fixed on the recent esphome version.

closed 2021-09-09 by nkitanov